FreeCAD/FreeCAD #9212: IMAGE WORKBENCH MISSING

Jasonv310, 2023-04-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
I now have the image workbench missing since I upgraded to .21
```

### Subproject(s) affected?

None

### Issue description

I now have the image workbench missing since I upgraded to .21

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-04-07:
The image workbench functionality has been integrated into the core of FreeCAD. To load an image, use File->Open or File-> Import. To control its placement, double-click on it in the tree and use the Task panel.

closed 2023-04-07 by chennes

xtemp09, 2023-04-07:
@chennes, the wikipedia does not mention the removal. [^1]
 [^1]: https://wiki.freecad.org/Image_Workbench

Jasonv310, 2023-04-07:
What is the easiest way to scale my pic or item? I get to the scale part but it wont let me go past 0 on the settings and my project is already to big. 
Jason "Cueball" VernonPresidentPunishers LEMC Hellbranch Ohio ***@***.*** 

 
 

Footnotes
   
   -    
https://wiki.freecad.org/Image_Workbench  ↩