FreeCAD/FreeCAD #9223: SIGSEGV in TechDraw

xtemp09, 2023-04-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32728 (Git)
Build type: Debug
Branch: master
Hash: 2e23fcf99e86ee2efea085e002392fe63c1cc83d
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.0.dev
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Techdraw

### Problem description

1. Insert something, for example, a cube from Part
2. Insert a default page
3. Insert a projection group
4. Select an edge
5. Choose _Calculate the area of selected faces_
6. Observe segmentation fault

<details><summary>Screenshot</summary>

[image]

</details>

The crash takes place within the function `CmdTechDrawExtensionAreaAnnotation::activated(int iMsg)` at
 https://github.com/FreeCAD/FreeCAD/blob/f43a041dc882816da75118355e46dae775ea5e9b/src/Mod/TechDraw/Gui/CommandExtensionPack.cpp#L1808-L1809

---

P.S. If I select a face, choose _Calculate the area of selected faces_,  set _Scale type_ to _Custom_, it does not change anything.

<details><summary>Screenshot</summary>

[image]

</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-04-10:
Should be fixed by PR #9229.

xtemp09, 2023-04-10:
Does scale work?

WandererFan, 2023-04-11:
As far as I know scale works.  I get the same answer for various scales.

WandererFan, 2023-04-11:
I think the Scale (and ScaleType) property is supposed to be hidden.  It doesn't make sense for Balloons to have a scale property. Balloons are based on Dimensions and Dimensions hide the Scale property. @edi271, any thoughts?

The area is calculated from the projected shape.   If the projection is orthogonal to a face, then the area will be 100 mm2.  If the project is not orthogonal, then it will give you the area of a parallelogram.

xtemp09, 2023-04-11:
WandererFan, the font is too small.

Perhaps, the tool should be called _Calculate the area of selected region_.

WandererFan, 2023-04-11:
"the font is too small."

FontSize property on the View tab.