FreeCAD/FreeCAD #9230: [Issue] non-continuous lines in TechDraw do not follow ISO128

WandererFan, 2023-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77469

### Version

0.21 (Development)

### Full version info

```shell
any
```

### Subproject(s) affected?

None

### Issue description

ISO128 provides standards for the representation of non-continuous lines.  The standard specifies the lengths of marks and 
spaces as a function of the line width.

There are 6 graphical elements for line representation in ISO128:
Element/Line Width Multiplier
Dot: 1
Gap: 3
ShortDash: 6
Dash: 12
LongDash: 24
Space: 18

There are 15 non-continuous line representations:

- ISO01Continuous: n/a
- ISO02Dashed: Gap Dash
- ISO03DashedSpaced: Dash Space
- ISO04LongDashedDotted: LongDash Gap Dot Gap
- ISO05LongDashedDoubleDotted: LongDash Gap Dot Gap Dot Gap
- ISO06LongDashedTripleDotted: LongDash Gap Dot Gap Dot Gap Dot Gap
- ISO07Dotted: Dot Gap
- ISO08LongDashShortDash: LongDash Gap ShortDash Gap
- ISO09LongDashDoubleShortDash: LongDash Gap ShortDash Gap
- ISO10DashedDotted: Dash Gap Dot Gap
- ISO11DoubleDashedDotted: Gap Dash Gap Dot Gap Dash
- ISO12DashedDoubleDotted: Dash Gap Dot Gap Dot Gap
- ISO13DoubleDashedDoubleDotted: Dash Gap Dash Gap Dot Gap Dot Gap
- ISO14DashedTripleDotted: Dash Gap Dot Gap Dot Gap Dot Gap
- ISO15DoubleDashedTripleDotted: Dash Gap Dash Gap Dot Gap Dot Gap Dot Gap

Currently, TechDraw provides 5 options for line styles:
- continuous (ISO 01)
- dashed (similar to ISO 02)
- dotted (similar to ISO 07)
- dashDot (similar to ISO 10)
- dashDotDot (similar to ISO 12)

The mark and space lengths are provided by Qt according to Qt::PenStyle.

In order to support the ISO line representations, we need to draw lines using the Qt::CustomDashLine style and QPen::setDashPattern(const QList<qreal> &pattern) or QPainterPathStroker::setDashPattern(const QList<qreal> &dashPattern) methods.

Affected files will include:
    QGIViewPart.cpp
    QGIEdge.cpp
    TaskLineDecor.cpp
    and the related preference code.

When to offer line styles will need to be reviewed - for example visible lines are always continuous, but
there is more than 1 convention for hidden lines.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-04-16:
Note that void QGICenterLine::setTools() is an example of making a line with custom dashes.

WandererFan, 2023-04-17:
related: https://forum.freecad.org/viewtopic.php?p=675972#p675972

benj5378, 2023-06-15:
Currently looking at this

Roy-043, 2023-10-08:
Related: #10976.

WandererFan, 2023-10-09:
Not sure that #10976 is related.  That looks like a rendering issue, where this is a defect in the definition of the dash pattern.

maxwxyz, 2023-12-04:
This should be closed with the merge of #11594

WandererFan, 2023-12-06:
Completed by PR #11594

closed 2023-12-06 by WandererFan