FreeCAD/FreeCAD #9234: [Issue] SIGSEGV in BRepTools::EvalAndUpdateTol() when using PartDesign_Draft

mpictor, 2023-04-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Gentoo Linux (KDE//usr/share/xsessions/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27319 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 67706b1)
Hash: 67706b14df32b91b948952ea2c9350948ba4d4db
Python version: 3.9.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Issue description

I'm trying to use `draft` to create a transition between one piece and another. 
In the screenshot, I want to angle the green face like the yellow one (~30 degrees)
[image]

the cad file:
[solderironholder.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/11207005/solderironholder.FCStd.gz)
When you open this file, the face should be in view.

### Anything else?

The freecad I'm running is `FreeCAD_weekly-builds-27319-Linux-Conda_glibc2.12-x86_64.AppImage`

```
FreeCAD 0.20, Libs: 0.20R27319 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
```
backtrace
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x37c30) [0x7f00a785ac30]
#1  0x7f002f3a05f0 in BRepTools::EvalAndUpdateTol(TopoDS_Edge const&, opencascade::handle<Geom_Curve> const&, opencascade::handle<Geom2d_Curve>, opencascade::handle<Geom_Surface> const&, double, double) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKBRep.so.7+0xb0
#2  0x7f0037157ded in Draft_Modification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKOffset.so.7+0x54d
#3  0x7f002f3aebdf in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKBRep.so.7+0x1c6f
#4  0x7f002f3ad586 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKBRep.so.7+0x616
#5  0x7f002f3ad586 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKBRep.so.7+0x616
#6  0x7f002f3b01ac in BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKBRep.so.7+0x3bc
#7  0x7f002fc5c719 in BRepBuilderAPI_ModifyShape::DoModif() from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKTopAlgo.so.7+0x69
#8  0x7f0037129c70 in BRepOffsetAPI_DraftAngle::Build() from /tmp/.mount_freeca3mLcsQ/usr/lib/./libTKOffset.so.7+0xf0
#9  0x7eff9c168359 in PartDesign::Draft::execute() from /tmp/.mount_freeca3mLcsQ/usr/lib/_PartDesign.so+0xbd9
#10  0x7f00a8d8fb10 in App::DocumentObject::recompute() from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADApp.so+0x4a0
#11  0x7f0037586cda in Part::Feature::recompute() from /tmp/.mount_freeca3mLcsQ/usr/lib/Part.so+0xa
#12  0x7f00a8d236dd in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADApp.so+0x2fd
#13  0x7f00a8d7c4d8 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADApp.so+0x78
#14  0x7eff843ba447 in PartDesignGui::TaskDraftParameters::onAngleChanged(double) from /tmp/.mount_freeca3mLcsQ/usr/lib/PartDesignGui.so+0xc7
#15  0x7f00a7d97a18 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0x6c0
#16  0x7f00a9d35f44 in Gui::QuantitySpinBox::valueChanged(double) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x44
#17  0x7f00a9d372a5 in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0xd5
#18  0x7f00a9d379f7 in Gui::QuantitySpinBox::updateText(Base::Quantity const&) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x77
#19  0x7f00a9d38183 in Gui::QuantitySpinBox::stepBy(int) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x83
#20  0x7f00a86ae7d5 in QAbstractSpinBox::wheelEvent(QWheelEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0x79
#21  0x7f00a85d4515 in QWidget::event(QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0x15b
#22  0x7f00a86b257f in QAbstractSpinBox::event(QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0xbd
#23  0x7f00a85a97f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#24  0x7f00a85ae377 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0xb23
#25  0x7f00a9911829 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x89
#26  0x7f00a7d81606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0x86
#27  /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5(+0x1af3b7) [0x7f00a85e43b7]
#28  /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5(+0x1affcc) [0x7f00a85e4fcc]
#29  0x7f00a85a97f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#30  0x7f00a85af31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#31  0x7f00a9911829 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x89
#32  0x7f00a7d81606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0x86
#33  0x7f00a8123dc8 in QGuiApplicationPrivate::processWheelEvent(QWindowSystemInterfacePrivate::WheelEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Gui.so.5+0x166
#34  0x7f00a812692c in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Gui.so.5+0x6c
#35  0x7f00a8113c65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Gui.so.5+0x63
#36  /tmp/.mount_freeca3mLcsQ/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5b010) [0x7f00a3333010]
#37  /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f00a447676d]
#38  /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/./libglib-2.0.so.0(+0x56a01) [0x7f00a4476a01]
#39  /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f00a4476aa1]
#40  0x7f00a7db00e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0x5e
#41  0x7f00a7d7e019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0x187
#42  0x7f00a7d81e8f in QCoreApplication::exec() from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libQt5Core.so.5+0xfd
#43  0x7f00a9899b0d in Gui::Application::runApplication() from /tmp/.mount_freeca3mLcsQ/usr/bin/../lib/libFreeCADGui.so+0x188d
#44  /tmp/.mount_freeca3mLcsQ/usr/bin/freecad(+0x451f) [0x5595995b251f]
#45  /lib64/libc.so.6(+0x2334a) [0x7f00a784634a]
#46  /lib64/libc.so.6(__libc_start_main+0x7c) [0x7f00a78463fc]
#47  /tmp/.mount_freeca3mLcsQ/usr/bin/freecad(+0x4841) [0x5595995b2841]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-12:
Can you please try to reproduce on development branch? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

mpictor, 2023-04-13:
In the process of putting it in an archive so github would allow the upload, I seem to have drug a copy of the file into itself.
this should be a valid file: [solderironholder.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/11217555/solderironholder.FCStd.gz)

`FreeCAD_weekly-builds-32785-2023-04-12-conda-Linux-x86_64-py310.AppImage` crashes, though I have to take the draft up to 45 degrees for that to happen.

Around 30 degrees, I sometimes get `failed to create compose table`. I repeated several times, trying to get a screenshot, and discovered that this message doesn't always show up.

Output of the new version:
```
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
failed to create compose table
```
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x37c30) [0x7fd459760c30]
#1  0x7fd3d2c9f26c in BRepTools::EvalAndUpdateTol(TopoDS_Edge const&, opencascade::handle<Geom_Curve> const&, opencascade::handle<Geom2d_Curve>, opencascade::handle<Geom_Surface> const&, double, double) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKBRep.so.7+0xac
#2  0x7fd3e1280995 in Draft_Modification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKOffset.so.7+0x525
#3  0x7fd3d2cac3c6 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKBRep.so.7+0x1c56
#4  0x7fd3d2caabe9 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKBRep.so.7+0x479
#5  0x7fd3d2caabe9 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKBRep.so.7+0x479
#6  0x7fd3d2cad97c in BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKBRep.so.7+0x3bc
#7  0x7fd3d356f0c9 in BRepBuilderAPI_ModifyShape::DoModif() from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKTopAlgo.so.7+0x69
#8  0x7fd3e12510f0 in BRepOffsetAPI_DraftAngle::Build(Message_ProgressRange const&) from /tmp/.mount_FreeCANevyt7/usr/lib/./libTKOffset.so.7+0x100
#9  0x7fd2c6e0e67c in PartDesign::Draft::execute() from /tmp/.mount_FreeCANevyt7/usr/lib/_PartDesign.so+0xb9c
#10  0x7fd45ad69862 in App::DocumentObject::recompute() from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADApp.so+0xc2
#11  0x7fd3e1afebaa in Part::Feature::recompute() from /tmp/.mount_FreeCANevyt7/usr/lib/Part.so+0xa
#12  0x7fd45acfd2cd in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADApp.so+0x41d
#13  0x7fd45ad4be8f in App::Document::recomputeFeature(App::DocumentObject*, bool) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADApp.so+0xef
#14  0x7fd2c6835a48 in PartDesignGui::TaskDraftParameters::onAngleChanged(double) from /tmp/.mount_FreeCANevyt7/usr/lib/PartDesignGui.so+0xb8
#15  /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Core.so.5(+0x1dff9b) [0x7fd459cc6f9b]
#16  0x7fd45be12c34 in Gui::QuantitySpinBox::valueChanged(double) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x44
#17  0x7fd45be13d98 in Gui::QuantitySpinBox::updateFromCache(bool, bool) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0xe8
#18  0x7fd45be13efe in Gui::QuantitySpinBox::updateText(Base::Quantity const&) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x6e
#19  0x7fd45be14a0c in Gui::QuantitySpinBox::stepBy(int) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x7c
#20  0x7fd45a601559 in QAbstractSpinBox::wheelEvent(QWheelEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5+0x79
#21  0x7fd45a52522a in QWidget::event(QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5+0x15c
#22  0x7fd45a4fb087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#23  0x7fd45a4ff8fe in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5+0xbf4
#24  0x7fd45b971bd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x99
#25  0x7fd459ca9a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Core.so.5+0x8e
#26  /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5(+0x1a38a6) [0x7fd45a5348a6]
#27  /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5(+0x1a4555) [0x7fd45a535555]
#28  0x7fd45a4fb087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#29  0x7fd45b971bd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x99
#30  0x7fd459ca9a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Core.so.5+0x8e
[... cut ...]
#39  0x7fd459cab7af in QCoreApplication::exec() from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libQt5Core.so.5+0x107
#40  0x7fd45b8e5006 in Gui::Application::runApplication() from /tmp/.mount_FreeCANevyt7/usr/bin/../lib/libFreeCADGui.so+0x22c6
#41  /tmp/.mount_FreeCANevyt7/usr/bin/freecad(+0x8a92) [0x564feab2ca92]
#42  /lib64/libc.so.6(+0x2334a) [0x7fd45974c34a]
#43  /lib64/libc.so.6(__libc_start_main+0x7c) [0x7fd45974c3fc]
#44  /tmp/.mount_FreeCANevyt7/usr/bin/freecad(+0x8de1) [0x564feab2cde1]
```

luzpaz, 2024-04-27:
bu,ping for visibility

luzpaz, 2024-08-27:
@mpictor can you still reproduce on latest dev ? (please paste the full [About](https://wiki.freecad.org/About) info)

maxwxyz, 2024-10-07:
@mpictor is this still an issue in the latest dev version / RC? If there is no feedback I'd propose to close this issue.

mpictor, 2024-10-09:
Sorry, had not seen the notifications. I'll try to find time to check in the next couple days.

mpictor, 2024-10-12:
Unfortunately this is still an issue in `1.0.0RC2`

I updated the file as prompted when opening, but ignored a number of warnings and errors, mostly regarding constraints. I'm assuming those can't (or shouldn't) lead to a crash.

The rendering starts to look weird over 30 degrees, and it crashes at 44.

```
The Sketch has partially redundant constraints!
<Sketch> SketchObject.cpp(345): Edge too small: Edge10
Sketcher constraint number 7 is malformed!
Sketcher constraint number 7 is malformed!
The Sketch has malformed constraints!
Sketch011: Sketch with redundant constraints
Please remove the following redundant constraint:
11

Sketch015: Sketch with redundant constraints
Please remove the following redundant constraint:
22

Sketch040: Over-constrained sketch
Please remove at least one of the following conflicting constraints:
1
, 2, 5, 6

Sketch049: Sketch with redundant constraints
Please remove the following redundant constraints:12
, 15, 21, 24

Sketch034: Sketch with malformed constraints
Please remove the following malformed constraint:
7

Sketch054: AttachEngine3D: subshape not found Pad029.Face13
Sketcher constraint number 7 is malformed!
Sketcher constraint number 7 is malformed!
The Sketch has malformed constraints!
Sketch011: Sketch with redundant constraints
Please remove the following redundant constraint:
11

Sketch015: Sketch with redundant constraints
Please remove the following redundant constraint:
22

Sketch040: Over-constrained sketch
Please remove at least one of the following conflicting constraints:
1
, 2, 5, 6

Sketch049: Sketch with redundant constraints
Please remove the following redundant constraints:12
, 15, 21, 24

Sketch034: Sketch with malformed constraints
Please remove the following malformed constraint:
7

Sketch054: AttachEngine3D: subshape not found Pad029.Face13
```
...and the actual crash
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x3c480) [0x7fd0f210b480]
#1  0x7fd0638cecfa in BRepTools::EvalAndUpdateTol(TopoDS_Edge const&, opencascade::handle<Geom_Curve> const&, opencascade::handle<Geom2d_Curve> const&, opencascade::handle<Geom_Surface> const&, double, double) from /tmp/.mount_FreeCA14sYDv/usr/lib/./libTKBRep.so.7.7+0xaa
#2  0x7fd060fd8935 in Draft_Modification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) from /tmp/.mount_FreeCA14sYDv/usr/lib/././libTKOffset.so.7.7+0x645
#3  0x7fd0638dcba7 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCA14sYDv/usr/lib/./libTKBRep.so.7.7+0x1c27
#4  0x7fd0638db68d in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCA14sYDv/usr/lib/./libTKBRep.so.7.7+0x70d
#5  0x7fd0638db68d in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /tmp/.mount_FreeCA14sYDv/usr/lib/./libTKBRep.so.7.7+0x70d
#6  0x7fd0638de018 in BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) from /tmp/.mount_FreeCA14sYDv/usr/lib/./libTKBRep.so.7.7+0x368
[... 14 lines cut ...]
#21  0x7fd0f32e3c08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5+0x178
#22  0x7fd0f32a53ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#23  0x7fd0f32adb16 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5+0x1c56
#24  0x7fd0f4be0079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libFreeCADGui.so+0x99
#25  0x7fd0f277ab28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Core.so.5+0x108
#26  /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5(+0x1c8ad5) [0x7fd0f32fdad5]
#27  /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca1f8) [0x7fd0f32ff1f8]
#28  0x7fd0f32a53ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7fd0f4be0079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libFreeCADGui.so+0x99
#30  0x7fd0f277ab28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Core.so.5+0x108
#31  0x7fd0f2c03031 in QGuiApplicationPrivate::processWheelEvent(QWindowSystemInterfacePrivate::WheelEvent*) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Gui.so.5+0xd1
#32  0x7fd0f2bdd0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Gui.so.5+0xac
#33  /tmp/.mount_FreeCA14sYDv/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7fd0ec98dbea]
#34  /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fd0ee7b8ad7]
#35  /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fd0ee7bbac7]
#36  /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fd0ee7bc0ce]
[... cut ...]
#39  0x7fd0f2781212 in QCoreApplication::exec() from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7fd0f4b3f4e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCA14sYDv/usr/bin/../lib/libFreeCADGui.so+0xa41
#41  /tmp/.mount_FreeCA14sYDv/usr/bin/freecad(+0x789f) [0x5567d47e589f]
#42  /usr/lib64/libc.so.6(+0x26320) [0x7fd0f20f5320]
#43  /usr/lib64/libc.so.6(__libc_start_main+0x89) [0x7fd0f20f53d9]
#44  /tmp/.mount_FreeCA14sYDv/usr/bin/freecad(+0x7c12) [0x5567d47e5c12]
```

luzpaz, 2024-10-13:
Can you add the full [About](https://wiki.freecad.org/About) info of the RC2 version you tested on?

luzpaz, 2024-10-13:
I open the file and recompute. This is what is reported:  
```shell
10:51:41  solderironholder#Sketch029: The Sketch has partially redundant constraints!
10:51:41  solderironholder#Sketch031: The Sketch has partially redundant constraints!
10:51:41  <Sketch> SketchObject.cpp(345): Edge too small: Edge10
10:51:41  Sketcher constraint number 7 is malformed!
10:51:41  Sketcher constraint number 7 is malformed!
10:51:41  solderironholder#Sketch034: The Sketch has malformed constraints!
10:51:41  Sketch011: Sketch with redundant constraints
Please remove the following redundant constraint:
11

10:51:41  Sketch015: Sketch with redundant constraints
Please remove the following redundant constraint:
22

10:51:41  Sketch040: Over-constrained sketch
Please remove at least one of the following conflicting constraints:
1
, 2, 5, 6

10:51:41  Sketch049: Sketch with redundant constraints
Please remove the following redundant constraints:12
, 15, 21, 24

10:51:41  Sketch034: Sketch with malformed constraints
Please remove the following malformed constraint:
7

10:51:41  Sketch054: AttachEngine3D: subshape not found Pad029.Face13
```

But I cannpt seem to reproduce the crash.

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git) AppImage
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

mpictor, 2024-11-03:
I need to make a point of checking for replies more frequently :/
I'm using the AppImage `FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage`, sha1sum `50be5e4d2530599e3c3ad296d572ece81827ef8c`

```
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Lithophane
  * OpticsWorkbench 1.0.9
```

closed 2025-06-26 by github-actions[bot]