FreeCAD/FreeCAD #9247: techdraw on linux (debian bookworm): no handles (points) to select points for adding measures

woecj, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Austria (de_AT)
Installed mods: 
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

None

### Issue description

In techdraw I can not select points to add measures - no selector dots can be switched on.
So, I cannot add dimensions to drawings at all.

This effect is not there if I use the appimage downloaded from FreeCad site.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-13:
Where did you install FreeCAD from ?

woecj, 2023-04-13:
I installed it from the debian distribution repository.

luzpaz, 2023-04-13:
Yea, Debian and Ubuntu repos are not maintained (third party dependencies lag way behind). We have a PPA but it's not being maintained AFAIK right now. We're looking for a maintainer https://forum.freecad.org/viewtopic.php?t=69055 

AppImage, Flatpak, Snap can be used. 

Closing

closed 2023-04-13 by luzpaz

WandererFan, 2023-04-13:
duplicate of #6698?