FreeCAD/FreeCAD #9251: [Issue] Copy Paste of simple element is dauntingly complex.

PaddleStroke, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=65899

### Version

0.21 (Development)

### Full version info

```shell
0.21 dev
```

### Subproject(s) affected?

Core

### Issue description

For the new user copy pasting in the Tree is needlesly complex.
Most of the time the user just wants to copy paste a sketch, or a body. 
But when trying to copy, a dialog appears to ask what the user wants to copy.. Let just see this very simple example where I try to copy sketch001: 
[image]

This just overwhelms the user and is the open door to many bugs anyway (see https://github.com/FreeCAD/FreeCAD/issues/6985)

Bahviour suggestion for body: 
- Just copy paste the body...

Behaviour suggestion for sketches :
- right click the sketch and copy (or ctrlC)
- Nothing happens. (No dependency window shows up)
- Right click on one body or another, paste (or ctrlV) :
- Copy the sketch inside that body at the end of the body's elements.
___A - If the sketch was on XY YZ ZX planes, use the body's plane automatically.
___B - If the sketch was on one face of the selected body, use that face.
___C - If the sketch was on one face of another body: Open a dialog to offer user 3 possibilities :
______a - Paste the sketch on a newly created plane with same equation as where it was in its original body.
______b - Paste the sketch on XY plane (and user will re-map the sketch where he wants it.)
______c - Select dependencies : Open the 'Select object' dialog which currently opens at step 2.

If you really want to keep this 'dependency selector' create a special 'Copy advanced' command that is separate.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-25:
related: https://github.com/FreeCAD/FreeCAD/issues/6278

pierreporte, 2024-12-05:
> Case where the object is pasted in the same container (or in a subcontainer): do nothing...

I would expect an actual copy of the object. `Sketch` object is copied → `Sketch001` is created when pasted.

> The most classic example is a sketch attached to its body origin plane: For this replace it with the new container's. If it's pasted as root, then dettach and set placement correctly.

It depends on what you want to do. Most of the time, I would expect the pasted object to be at the same position as the original object, even if the target container has a different origin, so that it would need to compute an offset. Even in an assembly: say I copy a datum line stored in a part (the active document is the assembly), when pasted in another part (in the same active document), it is coincident with the original sketch.

It would be important to provide special pasting methods. In Catia, you can have up to 3 choices: paste with the feature tree, paste as a result but linked to the original object (say a sketch, the resulting object will change as the original sketches is modified), or paste as a result (in the datum line example above, the two lines would be completely independent, and the second one will be positioned with coordinates even though the original may be attached to something). Pasting as a result can change the nature of objects. For a body, the result behaves like a step file. A sketch line would be a 3D line (this is not implemented yet, see #14010).

zackwhit, 2024-12-06:
> I would expect an actual copy of the object. `Sketch` object is copied → `Sketch001` is created when pasted.

I second this.

> I would expect the pasted object to be at the same position as the original object, even if the target container has a different origin

This is less important to me; actually often times I want new objects to be at 0,0,0 after they're pasted into a new group or root.

Architecturally speaking, I think allowing the existence of a dialog box is important. There are already a myriad of different copy/paste contextually desirable outcomes. For simple things, like copy-pasting a sketch into another body, I don't think that should require a dialog box. However, if you had something like a reference object (plane, axis, etc), and it's not included in the selection, might be nice to prompt either to include it, reference the original, or select new reference. The assembly workbench also has some potentially different desirable behavior here too. If I were writing this feature I would be inclined to have a context-conscious system, we can fill out the edge cases over time past simple "paste-into-selection". I dunno, just my 2 cents.

yorikvanhavre, 2024-12-20:
I think the fine-tuning copy dialog should not disappear. It might be klunky but it is still very useful. What about keeping it as a kind of "copy special..." command? Many apps have a "paste" and a "paste special" command with more options. Indeed the dialog could also be transferred to the paste (or better, paste special) operation, but it's pretty logical as it is now.

What about allowing each object to define what should be copied by the (simple) "copy" command? There would be defaults, but the defaults would then be overridable by the developers

PaddleStroke, 2024-12-20:
My idea was along those lines:
- Define the behavior when we can. (ie the objects decide what to do)
- If no special behavior can be found, then default to current mode with the selection dialog.
- We can also have a 'special paste' command.

pierreporte, 2024-12-20:
À “special copy” tool would be useful when the défaut pasting is not the needed one. It exists in other programs for this reason.

leoheck, 2025-03-09:
Any progress on this?

maxwxyz, 2026-11-07:
Part of this is in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/a6bb5ecb76e4e93afbd25f24c958bd5faee74402 and https://github.com/AstoCAD/FreeCAD/commit/fa819f78ea6c57e82ae77fc6f67634d5ea797841

Common sketch and body copies skip the dependency dialog, and Copy Special keeps it available. Ambiguous dependencies still ask, it does not implement all the proposed face-attachment choices.