FreeCAD/FreeCAD #9253: [Issue] Many tooltips are not helpful.

PaddleStroke, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76836

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

None

### Issue description

Many toolstips are not very informative. Qt gives us the possibilities to make complete tooltips with pictures and links.

I suggest we improve the tooltip by adding a picture and a link to the wiki.
[image]

Also it has been asked to have a way to disable tooltips for veteran users that don't want them to get in the way.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-04-13:
I offer to make a PR to : 
- Make proper tooltips for PartDesign.
- Add the 'disable tooltip' setting.
- Add to the StyleBook guidelines to how tooltips should look like and what they should contain.

Then we try to get people involved to improve the tooltips accross the application.

What do you think about that?

PaddleStroke, 2023-04-14:
Ok this makes sense.
Then each card in the project, should it be individual issues or just unlinked cards? I fear that would put many issues on the tracker. But if each action is not an issue we're not following the defined process I guess.

Btw is it acceptable to put HTML in QT_TR_NOOP?
```
    sToolTipText    = QT_TR_NOOP("Create a <b>line in</b> the sketch \n" 
        "<img src=':/icons/dialogs/Sketcher_Settings.svg'>"); 
```

adrianinsaval, 2023-04-14:
> Besides we could put a URL to the tool wiki page.

The problem is the wiki will keep evolving, instead we should point to the versioned clone of the wiki that is used by the help module.

chennes, 2023-04-15:
> Btw is it acceptable to put HTML in QT_TR_NOOP?

It's risky, because it's very easy for translations to accidentally break it. 

```
 sToolTipText    = QT_TR_NOOP("Create a line in the sketch <a href='https://wiki.freecad.org/'>Wiki</a> <br>"
```
might be better written as
```
 sToolTipText = QT_TR_NOOP("Create a line in the sketch") 
    + " - <a href='https://wiki.freecad.org/'>"
    + QT_TR_NOOP("Wiki") 
    + "</a> <br>";
```
(or something along those lines -- you could use `format` if you preferred).

luzpaz, 2023-04-15:
BTW, @realthunder added this (images in tooltips) option to Link branch IIRC.

PaddleStroke, 2023-04-18:
[image]
Hmm the picture is somehow badly resized and looks very ugly. I tried to vectorize it into a svg but the result is the same. That would need further investigation

chennes, 2023-04-18:
I'm surprised that image didn't vectorize well -- did you use Inkscape, or something else? I can do it in Illustrator if you like.

PaddleStroke, 2023-04-18:
It did vectorize well. It just rendered poorly in the app. Probably when Qt turns it back into a png to display.

Mannshoch, 2024-08-30:
I propose to work together with the Wiki:
https://forum.freecad.org/viewtopic.php?t=76836&start=20#p684011

Reqrefusion, 2024-12-05:
Actually, I see this as part of offline documentation. [Here](https://github.com/Reqrefusion/FreeCAD-Documentation-html)'s a wiki rendering of the relevant work, which is always up to date. There is an order on the wiki for the relevant images, they are generally named by adding a number at the end of the tool, but not every tool has them, there are more than one or they are not descriptive enough.