FreeCAD/FreeCAD #9255: [Issue] Pref General: Useless General groupbox.

PaddleStroke, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

None

### Issue description

There is a General groupbox holding all the items of the General tab. This is redundant and cluttering the view.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-04-13:
Yeah, axe it.

adrianinsaval, 2023-04-15:
fixed by #9260

closed 2023-04-15 by adrianinsaval