FreeCAD/FreeCAD #9256: [Issue] Pref: 'Stylesheet' is called 'Theme' by most people out there.

PaddleStroke, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

None

### Issue description

In preference, our setting 'Stylesheet' is not the most intuitive name. We call that like this because it's using Qt stylesheet. But general users and most applications everywhere call that 'Theme'.

So I suggest changing the name to 'Theme'.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-04-13:
I agree with you theme should also include other color preferences.
The fact that actual 'theme' is achieve by a combination of Stylesheet + prefpack is quite unintuitive.

We should deal with that internally and offer to the user on the preference page a single 'theme' setting.

 Then in 'Display' we could have a 'Theme' tab that would go into more details about the theme. ie stylesheet and other. I was going to say 'and preference pack' but the issue is that preference pack does not only do theme...

The

chennes, 2023-04-13:
Right: preference packs are very general (by design). But we could use metadata to let Pack authors indicate that their Pack is a "theme", and then display a list of those in the UI. I think it is achievable.

PaddleStroke, 2023-04-13:
Also, I think it would be good to have built-in official themes that don't require the use of the addon manager.

chennes, 2023-04-13:
Agreed: I think that UltraProDark and UltraProLight (or whatever their real names are!) would be great candidates for inclusion in our distribution.

luzpaz, 2023-04-16:
CC @MisterMakerNL :point_up:

MisterMakerNL, 2023-06-13:
I came across this and the middle road here was to call the preference packs that contain "theme" stuff to name them Themes:
https://github.com/FreeCAD/FreeCAD/pull/9715/commits/9be0f548bb19bf577425cf504eba5080c5facd11
We can easily combine preference packs with theme's it's probably wiser to hide stylesheets link them to a preference pack theme and call it highlight colors.

closed 2023-07-27 by luzpaz