FreeCAD/FreeCAD #9262: [Issue] Navigation cube font size strange behaviour

Jolbas, 2023-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=674532#p674532

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * kugghjul.py
  * dxf-library
  * fcgear 1.0.0
  * __pycache__
  * sheetmetal 0.2.61
  * tida.py
  * Curves 0.6.8
  * Assembly4 0.12.7
[/code]
```

### Subproject(s) affected?

Core

### Issue description

• The font size setting of the navigation cube does not match the size of the navigation cube. Setting the font size to 100 gives a 4 times smaller font. Image showing cube size 250 and font size 250
[image]
• Setting a too big font size makes the text beed over the edges of the face it should lie on.
Image showing cube size 250 and font size 900
[image]
• If tuning the font size to match the cube size and then change the cube size you have to tune the font size again. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-04-13:
I have an almost done fix for this. The font size is instead set as percentage of cube size and labels that doesn't fit is individually downsized until they fits the face.

chennes, 2023-04-13:
How does your fix interact with https://github.com/FreeCAD/FreeCAD/pull/9244?

ronnystandtke, 2023-04-13:
Great, I absolutely love the idea of autoscaling the font size on the navicube.

I see the following variants:
- individual autoscaling on every face
  - most probably resulting in different font sizes on every face
  - some people don't seem to like it (see forum discussion)
- synchronized autoscaling
  - the face with the longest text determines the font size
  - same font size on every face
  - a bit of wasted space
  - maybe visually more pleasing
- manual font size

I would probably default to the second option but offer the other variants in the preferences.

adrianinsaval, 2023-04-14:
I think the second option makes the most sense, we can then easily use 100 as the default value and it would look good regardless of platform or selected font. Offering 3 options seems overkill.

Jolbas, 2023-04-14:
I would like to keep the option to make labels as big as possible (100%)
[image]
I think I prefer something like this to make it more readable without having a massive TOP label (25%)
[image]
With my suggestion it wouldn't be hard to make all labels have the same font size (18%)
[image]
I don't really get why you want to ditch the flexible font size in favour for the option to undersize the font (5%)
[image]
How about removing the numbers from the setting and instead choose between the following?
- individual autoscaling on every face
- individual but set maximum size so that the median length word fits
- synchronized autoscaling
Those options would result in the three first images.

Note: Changed REAR to BOTTOM to get longest and shortest word in the same picture

adrianinsaval, 2023-04-14:
Ok, I think I got a little confused, the 25% option looks pretty good actually. And since the number is not very meaningful when it is dynamically ignored I think you are right that it might not be worth showing.
And that leads me to the question... Do we even need an option here? The font size option now exists because users need it to adapt to their cube sizes and font of choice, with an automated way of getting sane font sizes I think we can just get rid of the option altogether. What usecase is there for changing the font size separately?

So I now propose, we just set it to 25% and get rid of the GUI to change this.

Jolbas, 2023-04-15:
Sure. I like your proposal.

Jolbas, 2023-04-20:
I have now tested a non GUI parameter called FontZoom.
 1.0 = The labels are individually maximized
 0.0 = The longest label is maximized. All labels have the same font size 
Values in between controls the maximum font size
A negative value is possible and reduces the font size further.
Default value 0.3 reduces the size of "BOTTOM" and with some fonts also "FRONT" in the default label set
I think this works very well.

PR #9356 (2023-04-24): Refactor Navigation Cube
This PR does not conform to the one issue recommendation. Hopefully you can bear reviewing it anyway.
Fixes #9262 
Fixes #6935 
Fixed #6818 
But also a non registered issue from the forum:
[https://forum.freecad.org/viewtopic.php?t=77768](url)
Will hopefully also fix a problem on Raspberry Pi
[https://forum.freecad.org/viewtopic.php?t=76438](url)

Here is a list of changes
- Uses std commands in menu coming with icons
- Cube only draggable if activated in menu 
- [image]
- Ortographic and Perspective reflected in Navigation Cube
- [image]
- [image]
- Remove textures ex [... cut]
closed 2023-04-24 by chennes