FreeCAD/FreeCAD #9263: [Feature Request] New Unit, inch + fractional inch

ergodicbreak, 2023-04-13:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=674705#p674705

### Subproject(s) affected?

Core

### Idea description

### Background

Woodworking (i.e. furniture, small-scale wooden projects, etc.) plans in the USA typically use inch + fractional inch measurements (e.g. 37 and 1/16th inches).

Currently the closest Unit in FreeCAD is Building US, which prints as feet + inches + fractional inches, which is more typical in USA architecture (houses, landscaping, etc.). 

### Proposal

A new Unit, perhaps called Woodworking US, would let FreeCAD users display dimensions in inches + fractional inch. 

### Prior work

Preliminary work on this has already been done, at https://github.com/chadrosenquist/FreeCAD/commit/4708f9d5e899a1886e6391d1d1d1c4d32a157c33

(see associated forum post at https://forum.freecad.org/viewtopic.php?t=57274). 

However it seems from my searching this request didn't go anywhere. I reached out to that contributor via forum PM a few days ago, but haven't heard anything back yet. I'm guessing that unmerged code has the same license as FC by default?

I'd welcome feedback whether that code would be a suitable base for a PR. Thanks!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wt, 2023-12-08:
I would love this. I have been drawing up designs for car parts that were originally designed in inches. I would love to have this feature. It's not only useful for woodworkers.

basspenguin, 2024-01-08:
This would be really useful for me too. I do carpentry professionally and we only use inches and fractions. Same with woodworking. Feet and yards are not useful for any of my projects.

dekay, 2024-12-11:
I would like to add another vote for this feature. It is common in North American carpentry circles to work in inches + fractional inches only. Right now I am modeling a pinball machine cabinet and going off another plan with dimensions like 51 1/2". I can't get Freecad to natively display this. Closest it gets is 4'3" + 1/2" which is very awkward to go back and forth between, not to mention that the double quote for inches gets repeated twice. Any chance this feature could be added? And can the plus sign be removed so less space is taken up?

Jah-On, 2025-01-13:
Would be useful for beekeeping equipment in the U.S. of A!

markbreedlove, 2025-09-03:
I would find this very useful, as well.

tim-rohrer, 2025-09-14:
This is valid for most woodworking in the US. I actually don't mind feet, inches, fractions for longer measurements (i.e., > 5 feet, but that is arbitrary and very opinionated), but would definitely prefer just inches and the fraction for shorter (perhaps less that 5 feet as that is about as long as anyone can measure between their hands?) pieces.

The bigger annoyance I have though is the use of `+`. We just don't write it that way. Most plans and drawings I've used would express 2 feet and 3 and one half inches as either `2' 3-1/2"` (it is a hyphen, not a minus sign), `27-1/2"` or possibly `2' 3 1/2"`.

MisterCrispyFL, 2026-08-27:
I'd like to add another vote for this. For general woodworking in the US, the lack of fractional, inch-only measurements is frustrating. The general consensus on forums seems to be "lol just use metric" but EVERYTHING here is in fractional inches. If I'm making a piece of furniture, I don't want to have to mentally convert from yards. Let me measure 47 1/4 inches and call it a day. 

Also, I completely agree with @tim-rohrer earlier, the "3in + 3in/4" way of having to type everything is equally frustrating and has led to many typos.

tritao, 2026-08-27:
#30139 will bring some changes that will make this easier, I can look into after that PR lands.