FreeCAD/FreeCAD #9271: FreeCAD-0.20.2-WIN-x64-portable-3.zip

jangirke, 2023-04-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
See above
Zip file issue
```

### Subproject(s) affected?

None

### Issue description

Windows 10 errors
File SHA512 5345c24de791f646dddd860e993673fc5a313a55ceed459ca4606d4dc7d

Several errors while extracting with Windows Explorer:
100+ Errors
"Error 0x80004005: Unspecified error"

### Anything else?

Please fix

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jangirke, 2023-04-14:
Workaround:
Seems like 7zip isn't throwing errors.
It's still an unnecessary error.
Plz fix

xtemp09, 2023-04-15:
I can't reproduce on Windows 10. Quick search shows that this [Windows 10 error](https://support.microsoft.com/en-us/windows/fix-error-0x80004005-9acfca89-b5e4-b976-6fa1-ef358450f3ac) has already been fixed.

Please, close the issue.

closed 2023-04-15 by luzpaz

jangirke, 2023-04-17:
It doesn't have anything to do with the Macromedia / Adobe Flash Player.
It is the error while unpacking:
https://www.makeuseof.com/fix-unspecified-error-code-0x80004005-in-windows/

https://learn.microsoft.com/en-us/troubleshoot/windows-server/windows-security/user-account-control-and-remote-restriction

xtemp09, 2023-04-17:
@jangirke, it's still a Windows problem. It does not come from invalid archive.

luzpaz, 2023-04-17:
@jangirke where did you download this from ?