FreeCAD/FreeCAD #9273: [Problem] Freecad crash when environment variable XDG_SESSION_TYPE=wayland

ramast, 2023-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
FreeCAD 0.20.2 but it affected earlier versions too.
```

### Subproject(s) affected?

None

### Problem description

Pretty much the title. My Desktop is using wayland instead of X11 and Freecad kept crashing any time I try to open any file or start new design.

After running `unset XDG_SESSION_TYPE`, FreeCad worked very well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-15:
please post _full_ version information as is asked in the issue template:

> Please use the About FreeCAD dialog to copy your full version information and paste it here.

We need to know exactly what you are running (appimage, flatpak, snap, self compiled, package manager version?)

luzpaz, 2023-04-15:
Was there any output with the crash ?  
Can you also by chance recreate it on development build ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

ramast, 2023-04-16:
Sure thing.

version information:

[code]
OS: Gentoo Linux (KDE//usr/share/wayland-sessions/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: RelWithDebInfo
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

Freecad error

[code]
freecad: /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/glue/gl.cpp:2354: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
Aborted (core dumped)
[/code]

I've tested this Freecad Development

[code]
OS: Gentoo Linux (KDE//usr/share/wayland-sessions/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32821 (Git) AppImage
Build type: Release
Branch: master
Hash: eaf9d43f37238c33dd1c47b859a6fdf5b7f629fc
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

It worked fine though it gave a warning

[code]
Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
[/code]

adrianinsaval, 2023-04-21:
Seems like two different problems, for the first one: are you able to run other opengl apps in your system? what's the output of glxinfo?

for the second: conda packages (and therefore the appimages) do not support wayland, it may be possible to run using xwayland, I'm not familiar with XDG_SESSION_TYPE env variable, what is setting this?

ramast, 2023-04-21:
> Seems like two different problems, for the first one: are you able to run other opengl apps in your system?

Yes, opengl works perfectly fine.

> what's the output of glxinfo?

[glxinfo.txt](https://github.com/FreeCAD/FreeCAD/files/11298925/glxinfo.txt)

> the appimages do not support wayland

I think this is the reason why it doesn't crash.
Seems FreeCAD crashes only when using wayland (rather than xwayland).

I could be wrong but my understanding is that FreeCAD uses Qt which has native support for wayland but that crashes FreeCAD for some reasons. The prebuilt image which doesn't come with wayland support would run under xwayland which doesn't cause this crash.

> I'm not familiar with XDG_SESSION_TYPE env variable, what is setting this?

I can tell you for sure that I didn't set it manually. I don't know what set it but I think its automatically set when I start KDE in wayland mode

adrianinsaval, 2023-04-21:
did you try this recommendation from the error message:

> (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)

adrianinsaval, 2023-04-21:
found an old similar report of this on fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1795410 they blame it on coin3, supposedly fixed in coin4, but your info says coin4 so it might not be related.

ramast, 2023-04-22:
> did you try this recommendation from the error message:

Yes, it still crashes but different error.

I can see libCoin trying to use libX11 which it shouldn't?

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x37c10) [0x7f298eb09c10]
#1  /usr/lib64/libX11.so.6(XDefaultScreenOfDisplay+0) [0x7f298ce92090]
#2  /usr/lib64/libCoin.so.80(+0x437e6e) [0x7f298e5eee6e]
#3  /usr/lib64/libCoin.so.80(glxglue_init+0x3c) [0x7f298e5ef7dc]
#4  /usr/lib64/libCoin.so.80(cc_glglue_instance+0x15b) [0x7f298e5e7d3b]
#5  0x7f298e4893b8 in SoGLRenderActionP::isDirectRendering(SoState const*) const from /usr/lib64/libCoin.so.80+0x48
#6  0x7f298e48b943 in SoGLRenderActionP::render(SoNode*) from /usr/lib64/libCoin.so.80+0xc3
#7  0x7f298e484462 in SoAction::apply(SoNode*) from /usr/lib64/libCoin.so.80+0x3e2
#8  0x7f299149eca9 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x29
#9  0x7f2991554a83 in Gui::View3DInventorViewer::renderScene() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x143
#10  0x7f2991515735 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x295
#11  0x7f299151c5e2 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x42
#12  0x7f298fb8b1fe in QWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x20e
#13  0x7f298fc398ae in QFrame::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x1e
#14  0x7f298f16c075 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0xb5
#15  0x7f298fb47a5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x6e
#16  0x7f29911fd528 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x88
#17  0x7f298f16c2c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
#18  0x7f298fb83006 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib64/libQt5Widgets.so.5+0x36
#19  0x7f298fbabff9 in QOpenGLWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x129
#20  0x7f298fb47a6f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x7f
#21  0x7f29911fd528 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x88
#22  0x7f298f16c2c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
#23  0x7f298fb87b73 in QWidgetPrivate::show_helper() from /usr/lib64/libQt5Widgets.so.5+0x73
#24  0x7f298fb8acf3 in QWidgetPrivate::setVisible(bool) from /usr/lib64/libQt5Widgets.so.5+0x193
#25  0x7f298fb87ab9 in QWidgetPrivate::showChildren(bool) from /usr/lib64/libQt5Widgets.so.5+0x149
#26  0x7f298fb87b4f in QWidgetPrivate::show_helper() from /usr/lib64/libQt5Widgets.so.5+0x4f
#27  0x7f298fb87ad9 in QWidgetPrivate::showChildren(bool) from /usr/lib64/libQt5Widgets.so.5+0x169
#28  0x7f298fb87b4f in QWidgetPrivate::show_helper() from /usr/lib64/libQt5Widgets.so.5+0x4f
#29  0x7f298fb8acf3 in QWidgetPrivate::setVisible(bool) from /usr/lib64/libQt5Widgets.so.5+0x193
#30  0x7f298fb87ab9 in QWidgetPrivate::showChildren(bool) from /usr/lib64/libQt5Widgets.so.5+0x149
#31  0x7f298fb87b4f in QWidgetPrivate::show_helper() from /usr/lib64/libQt5Widgets.so.5+0x4f
#32  0x7f298fb87ad9 in QWidgetPrivate::showChildren(bool) from /usr/lib64/libQt5Widgets.so.5+0x169
#33  0x7f298fb87b4f in QWidgetPrivate::show_helper() from /usr/lib64/libQt5Widgets.so.5+0x4f
#34  0x7f298fb8acf3 in QWidgetPrivate::setVisible(bool) from /usr/lib64/libQt5Widgets.so.5+0x193
#35  0x7f298fccc4cc in QMdiSubWindow::eventFilter(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x24c
#36  0x7f298f16c075 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0xb5
#37  0x7f298fb47a5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x6e
#38  0x7f29911fd528 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x88
#39  0x7f298f16c2c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
#40  0x7f298fb8adf5 in QWidgetPrivate::setVisible(bool) from /usr/lib64/libQt5Widgets.so.5+0x295
#41  0x7f29911b0779 in Gui::Document::createView(Base::Type const&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x469
#42  0x7f2991140f13 in Gui::Application::slotNewDocument(App::Document const&, bool) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xbd3
#43  0x7f29908f0cb3 in boost::signals2::detail::signal_impl<void (App::Document const&, bool), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::Document const&, bool)>, boost::function<void (boost::signals2::connection const&, App::Document const&, bool)>, boost::signals2::mutex>::operator()(App::Document const&, bool) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x383
#44  0x7f2990a9d82b in App::Application::newDocument(char const*, char const*, bool, bool) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x167b
#45  0x7f2990ab7953 in App::Application::sNewDocument(_object*, _object*, _object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0xc3
#46  /usr/lib64/libpython3.10.so.1.0(+0xf8363) [0x7f299019d363]
[... cut ...]
#83  0x7f298f16acab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x12b
#84  0x7f298f17330d in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x8d
#85  0x7f299114671b in Gui::Application::runApplication() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x1b1b
#86  freecad(+0x5886) [0x5645add8e886]
#87  /lib64/libc.so.6(+0x2334a) [0x7f298eaf534a]
#88  /lib64/libc.so.6(__libc_start_main+0x7c) [0x7f298eaf53fc]
#89  freecad(+0x5b51) [0x5645add8eb51]
```

LittleFox94, 2023-08-06:
> did you try this recommendation from the error message:
> 
> > (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)

fwiw, for me it then works - but at least the Sketcher tool seems really slow, like it's not hardware accelerated anymore (didn't really try any other tool, but can do if that helps). At least I think it was faster when I ran it under X11, but since it's slow when moving things (and not e.g. when scrolling around or zoomin), it might be the solver after all.

Running 0.20.2 on Debian Trixie with current updates installed

luzpaz, 2023-08-06:
@LittleFox94 can you reproduce on latest stable v0.21 ?

ramast, 2023-08-07:
Yes

```
FreeCAD 0.22.0, Libs: 0.22.0devR14555 (Git shallow)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

QOpenGLFunctions created with non-current context
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
connect failed: No such file or directory
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
FreeCAD: /var/tmp/portage/media-libs/coin-4.0.0/work/coin/src/glue/gl.cpp:2354: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
```

khimaros, 2023-08-22:
running into this problem with the version of FreeCAD (0.20.2) in debian trixie

it probably warrants a opening a bug on https://bugs.debian.org

the `COIN_GL_NO_CURRENT_CONTEXT_CHECK=1` workaround is sufficient.

adrianinsaval, 2023-09-21:
what version of mesa are the people that have this problem running? Is there a chance of updating it for you?

ramast, 2023-09-21:
My mesa version is 23.0.3

adrianinsaval, 2023-09-21:
No clue but you could try

mikaelhug, 2023-10-02:
Any solution to this? im running into the same issue on Debian Trixie. FreeCAD novice here so unfortunately no clue what to do.

ramast, 2023-10-02:
Two temporary solutions has been suggested so far:

1. `export COIN_GL_NO_CURRENT_CONTEXT_CHECK=1` 
2.  `unset XDG_SESSION_TYPE`

Try running either one of these commands in a terminal before running FreeCad in same terminal.

deltalejo, 2023-10-04:
I'm having the same issue on Debian Trixie. Although the `COIN_GL_NO_CURRENT_CONTEXT_CHECK` workaround allows FreeCAD to run, the 3D rendering is a bit laggish.
By now I'm running FreeCAD with `-platform xcb` option, I think it has the same effect as `unset XDG_SESSION_TYPE`.
e.g. in the desktop entry:
`Exec=/usr/bin/freecad - --single-instance %F -- -platform xcb`

euphi, 2023-10-17:
I have same issue with a freshly compiled Freecad 0.20.2 on Gentoo.
With `COIN_GL_NO_CURRENT_CONTEXT_CHECK` set, FreeCAD segfaults.

Avoiding wayland by starting FreeCAD with `XDG_SESSION_TYPE=X freecad` works fine for me.

mosfet80, 2023-12-20:
FreeCAD 0.22.0, Libs: 0.22.0devR35381 (Git)

bindir:                      bin
BLAS:                        -undefined-
CMAKE_VERSION:               3.27.9
Compiler:                    /usr/bin/c++ (13.2.0)
datadir:                     share
docdir:                      share/doc/FreeCAD
includedir:                  include
libdir:                      lib
prefix:                      /usr/local
Python:                      3.11.7 [/usr/bin/python3] Suffix: [.cpython-311-x86_64-linux-gnu]

   ==============
     Config
   ==============

BUILD_FEM:                   ON
BUILD_GUI:                   ON
BUILD_SMESH:                 ON
BUILD_START:                 ON
BUILD_TECHDRAW:              ON
BUILD_VR:                    OFF 
BUILD_WEB:                   ON
CMAKE_BUILD_TYPE:            OFF 
CMAKE_CXX_FLAGS:             -Wall -Wextra -Wno-write-strings 
CMAKE_CXX_STANDARD:          17
CMAKE_INSTALL_PREFIX:        /usr/local
CMAKE_PREFIX_PATH:           OFF 
ENABLE_DEVELOPER_TESTS:      ON
FREECAD_CREATE_MAC_APP:      -undefined-
FREECAD_LIBPACK_USE:         OFF 
FREECAD_QT_VERSION:          Auto
FREECAD_USE_EXTERNAL_KDL:    OFF 
FREECAD_USE_EXTERNAL_SMESH:  OFF 
FREECAD_USE_FREETYPE:        ON
FREECAD_USE_PYBIND11:        OFF 
PYTHON_EXECUTABLE:           /usr/bin/python3
PYTHON_LIBRARY:              /usr/lib/x86_64-linux-gnu/libpython3.11.so
USE_CUDA:                    -undefined-
USE_OPENCV:                  -undefined-

   ==============
     Libraries
   ==============

Boost:                       1.74.0
Coin3D:                      4.0.2 [/usr/lib/x86_64-linux-gnu/libCoin.so] [/usr/include]
Coin3D_DOC:                  not found
DesignerPlugin:              not built (BUILD_DESIGNER_PLUGIN is OFF)
Doxygen:                     1.9.4 Language: English
Eigen3:                      3.4.0
fmt:                         9.1.0
Freetype:                    2.13.2
HDF5:                        1.10.10
Matplotlib:                  3.6.3 PathDirs: /usr/lib/python3/dist-packages/matplotlib
MEDFile:                     4.1.0 [/usr/lib/x86_64-linux-gnu/libmedC.so;/usr/lib/x86_64-linux-gnu/libmed.so] [/usr/include]
NETGEN:                      not enabled
OCC:                         7.6.3 [/usr/lib/x86_64-linux-gnu] [/usr/include/opencascade]
OCC_Libs:                    [TKFillet;TKMesh;TKernel;TKG2d;TKG3d;TKMath;TKIGES;TKSTL;TKShHealing;TKXSBase;TKBool;TKBO;TKBRep;TKTopAlgo;TKGeomAlgo;TKGeomBase;TKOffset;TKPrim;TKSTEPBase;TKSTEPAttr;TKSTEP209;TKSTEP;TKHLR;TKFeat]
OpenGL_Lib:                  [/usr/lib/x86_64-linux-gnu/libGL.so]
OpenGLU_Incl:                [/usr/include]
OpenGLU_Lib:                 [/usr/lib/x86_64-linux-gnu/libGLU.so]
pivy:                        0.6.8
pybind11:                    not enabled
PYCXX:                       7.1.7 Incl: /home/xx/Software/FreeCAD/src Src:/home/reale/Software/FreeCAD/src/CXX
PySide:                      5.15.11 [/usr/lib/../include/PySide2]
PySideTools:                 v: 2  uic: [/usr/lib/qt5/bin/uic]  rcc: [/usr/lib/qt5/bin/rcc]
QtConcurrent:                5.15.10
QtCore:                      5.15.10
QtNetwork:                   5.15.10
QtOpenGL:                    5.15.10
QtPrintSupport:              5.15.10
QtSvg:                       5.15.10
QtTest:                      5.15.10
QtUiTools:                   5.15.10
QtWebEngineWidgets:          5.15.15
QtWidgets:                   5.15.10
QtXml:                       5.15.10
Rift:                        not enabled (BUILD_VR)
Shiboken:                    5.15.11 [/usr/lib/../include/shiboken2;/usr/include/python3.11]
SMESH:                       7.7.1.0 build internal
SPNAV:                       [/usr/lib/libspnav.so] [/usr/include]
SWIG:                        4.1.0
VTK:                         9.1.0
XercesC:                     3.2.4 [/usr/lib/x86_64-linux-gnu/libxerces-c.so] [/usr/include]
ZLIB:                        1.2.13

QSocketNotifier: Can only be used with threads started with QThread
QOpenGLFunctions created with non-current context
Loading Assembly workbench...
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
FreeCAD: ./src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
Aborted

khumarahn, 2024-02-05:
I have this problem on too. The workarounds above do not help
```
$ XDG_SESSION_TYPE=X freecad
FreeCAD 0.22.0, Libs: 0.22.0devRUnknown
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

User path has changed to /home/khu/.local/share/FreeCAD/. Please move user modules and macros
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x396a0) [0x7f989e0796a0]
#1  0x7f989b65a290 in SoCallbackAction::~SoCallbackAction() from /usr/lib64/libCoin.so.80+0x90
#2  0x7f989b65a3e9 in SoCallbackAction::~SoCallbackAction() from /usr/lib64/libCoin.so.80+0x9
#3  0x7f989b6ecf84 in SoDragger::~SoDragger() from /usr/lib64/libCoin.so.80+0xd4
#4  0x7f989f9f4799 in Gui::TDragger::~TDragger() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x9
#5  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#6  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#7  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#8  0x7f989ba07c58 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x178
#9  0x7f989ba07ca9 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x9
#10  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#11  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#12  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#13  0x7f989ba0f551 in SoSwitch::~SoSwitch() from /usr/lib64/libCoin.so.80+0x51
#14  0x7f989ba0f579 in SoSwitch::~SoSwitch() from /usr/lib64/libCoin.so.80+0x9
#15  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#16  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#17  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#18  0x7f989ba07c58 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x178
#19  0x7f989b9b0529 in SoAnnotation::~SoAnnotation() from /usr/lib64/libCoin.so.80+0x9
#20  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#21  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#22  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#23  0x7f989ba07c58 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x178
#24  0x7f989ba07ca9 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x9
#25  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#26  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#27  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#28  0x7f989ba07c58 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x178
#29  0x7f989ba07ca9 in SoSeparator::~SoSeparator() from /usr/lib64/libCoin.so.80+0x9
#30  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#31  0x7f989b8b2029 in SoChildList::truncate(int) from /usr/lib64/libCoin.so.80+0x179
#32  0x7f989b8b1e1f in SoChildList::~SoChildList() from /usr/lib64/libCoin.so.80+0xf
#33  0x7f989b987077 in SoBaseKit::~SoBaseKit() from /usr/lib64/libCoin.so.80+0x27
#34  0x7f989fa015e9 in Gui::SoFCCSysDragger::~SoFCCSysDragger() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x9
#35  0x7f989b8abb19 in SoBase::destroy() from /usr/lib64/libCoin.so.80+0x139
#36  0x7f989f829311 in Gui::TaskCSysDragger::~TaskCSysDragger() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x21
#37  0x7f989f8293b9 in Gui::TaskCSysDragger::~TaskCSysDragger() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x9
#38  0x7f989fa5ad71 in Gui::TaskView::TaskView::removeDialog() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x181
#39  0x7f989fa5b673 in Gui::TaskView::TaskView::accept() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xc3
#40  /usr/lib64/libQt5Core.so.5(+0x2beafd) [0x7f989c4beafd]
#41  /usr/lib64/libQt5Widgets.so.5(+0x306f30) [0x7f989d106f30]
#42  /usr/lib64/libQt5Core.so.5(+0x2bec0c) [0x7f989c4bec0c]
#43  0x7f989d05c142 in QAbstractButton::clicked(bool) from /usr/lib64/libQt5Widgets.so.5+0x42
#44  /usr/lib64/libQt5Widgets.so.5(+0x25c3aa) [0x7f989d05c3aa]
#45  /usr/lib64/libQt5Widgets.so.5(+0x25ded7) [0x7f989d05ded7]
#46  0x7f989d05e0f3 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib64/libQt5Widgets.so.5+0x103
#47  0x7f989cfaa788 in QWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x268
#48  0x7f989cf67eee in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x7e
#49  0x7f989cf703d4 in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x11c4
#50  0x7f989f729331 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x61
#51  0x7f989c48caf8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
[... cut ...]
#65  0x7f989c48b5bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x12b
#66  0x7f989c4935cd in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x8d
#67  0x7f989f66bedc in Gui::Application::runApplication() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x3bac
#68  freecad(+0x7d13) [0x555bf0a10d13]
#69  /usr/lib64/libc.so.6(+0x23f3a) [0x7f989e063f3a]
#70  /usr/lib64/libc.so.6(__libc_start_main+0x85) [0x7f989e063ff5]
#71  freecad(+0x6fd1) [0x555bf0a0ffd1]
```

Amr-Ibra, 2024-03-16:
* Debian testing
* GNOME on Wayland
* mesa 23.3.5
* freecad 0.21.2 → Qt5 on Wayland

Open new file → crash

`freecad: ./src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Zusicherung »current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"« nicht erfüllt.
Abgebrochen (Speicherabzug geschrieben)`

xundeenergie, 2024-03-27:
Gnome on debian bookworm with wayland, freecad installed by flatpak:

```
$ flatpak run  org.freecadweb.FreeCAD
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

QSocketNotifier: Can only be used with threads started with QThread
(qt.qpa.qgnomeplatform) Could not find color scheme  ""
QOpenGLFunctions created with non-current context
[2:32:0327/144109.084908:ERROR:address_tracker_linux.cc(214)] Could not bind NETLINK socket: Die Adresse wird bereits verwendet (98)
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x7f7b17051e80]
#1  /usr/lib/x86_64-linux-gnu/libX11.so.6(XDefaultScreenOfDisplay+0x4) [0x7f7b130efa64]
#2  /app/lib/libCoin.so.80(+0x4e3abe) [0x7f7b16ae3abe]
#3  /app/lib/libCoin.so.80(glxglue_init+0x42) [0x7f7b16ae45b2]
#4  /app/lib/libCoin.so.80(cc_glglue_instance+0x23f) [0x7f7b16ae248f]
#5  0x7f7b1693f2f8 in SoGLRenderActionP::isDirectRendering(SoState const*) const from /app/lib/libCoin.so.80+0x48
#6  0x7f7b1694169b in SoGLRenderActionP::render(SoNode*) from /app/lib/libCoin.so.80+0xcb
#7  0x7f7b1693ad82 in SoAction::apply(SoNode*) from /app/lib/libCoin.so.80+0x422
#8  0x7f7b1a7892af in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /app/freecad/lib/libFreeCADGui.so+0x2f
#9  0x7f7b1a87d967 in Gui::View3DInventorViewer::renderScene() from /app/freecad/lib/libFreeCADGui.so+0x167
#10  0x7f7b1a831645 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /app/freecad/lib/libFreeCADGui.so+0x285
#11  0x7f7b1a83bb07 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /app/freecad/lib/libFreeCADGui.so+0x47
#12  0x7f7b185bcc98 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x268
#13  0x7f7b18667ece in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#14  0x7f7b178dfa2b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x8b
#15  0x7f7b1857ad5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x6e
#16  0x7f7b1a455180 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#17  0x7f7b178dfca8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#18  0x7f7b185b4e85 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x35
#19  0x7f7b185dcb50 in QOpenGLWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x230
#20  0x7f7b1857ad6e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#21  0x7f7b1a455180 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#22  0x7f7b178dfca8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#23  0x7f7b185b97f3 in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x73
#24  0x7f7b185bc743 in QWidgetPrivate::setVisible(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x193
#25  0x7f7b185b9739 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x149
#26  0x7f7b185b97cf in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4f
#27  0x7f7b185b9759 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x169
#28  0x7f7b185b97cf in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4f
#29  0x7f7b185bc743 in QWidgetPrivate::setVisible(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x193
#30  0x7f7b185b9739 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x149
#31  0x7f7b185b97cf in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4f
#32  0x7f7b185b9759 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x169
#33  0x7f7b185b97cf in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4f
#34  0x7f7b185bc743 in QWidgetPrivate::setVisible(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x193
#35  0x7f7b186f7399 in QMdiSubWindow::eventFilter(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x239
#36  0x7f7b178dfa2b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x8b
#37  0x7f7b1857ad5e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x6e
#38  0x7f7b1a455180 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
[... cut ...]
#82  0x7f7b178f0226 in QMetaObject::invokeMethod(QObject*, char const*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x816
#83  0x7f7ab97beaca in WebGui::WebEngineUrlRequestInterceptor::interceptRequest(QWebEngineUrlRequestInfo&) from /app/freecad/lib/WebGui.so+0x1ea
#84  /app/lib/libQt5WebEngineCore.so.5(+0xa2bb31) [0x7f7aaca2bb31]
```

bgbsww, 2024-07-01:
Brand new Ubuntu 24.04 installation shows this bug after first compile of the source:

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Debug
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.13
  * Assembly3 0.12.2
  * fcgear 1.0.0
  * sheetmetal 0.4.18
  * DynamicData 2.62.0
  * BIM021 2021.12.0
  * parts_library

COIN_GL_NO_CURRENT_CONTEXT_CHECK  does work around it effectively.

Console messages:

20:34:43  QOpenGLFunctions created with non-current context
20:34:54  (qt.qpa.wayland) Wayland does not support QWindow::requestActivate()

adrianinsaval, 2024-07-05:
> Brand new Ubuntu 24.04 installation shows this bug after first compile of the source:
> 
> OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu) Word size of FreeCAD: 64-bit Version: 0.22.0dev.37928 (Git) Build type: Debug Branch: main Hash: [e7699b5](https://github.com/FreeCAD/FreeCAD/commit/e7699b531eba4bee262f7214699f07b96f66ba7b) Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3 Locale: English/United States (en_US) Installed mods:
> 
>     * FreeCAD_BearingsWB 0.0.2
> 
>     * Assembly4 0.50.13
> 
>     * Assembly3 0.12.2
> 
>     * fcgear 1.0.0
> 
>     * sheetmetal 0.4.18
> 
>     * DynamicData 2.62.0
> 
>     * BIM021 2021.12.0
> 
>     * parts_library
> 
> 
> COIN_GL_NO_CURRENT_CONTEXT_CHECK does work around it effectively.
> 
> Console messages:
> 
> 20:34:43 QOpenGLFunctions created with non-current context 20:34:54 (qt.qpa.wayland) Wayland does not support QWindow::requestActivate()

have you tried a qt6 build?

bgbsww, 2024-07-05:
Not yet, still exploring the out-of-the-box versions of things on 24.04.  I'll note it when I do.  The workaround is fine though.

sgeulette, 2024-08-17:
Also have the bug with a default brand new installation of ubuntu 24.04. 
Have tried with last stable 0.21.2 or daily version.
Started correctly with `env COIN_GL_NO_CURRENT_CONTEXT_CHECK=1 /usr/bin/freecad-daily` but yet errors after opening existing project:
```
QSocketNotifier: Can only be used with threads started with QThread
QOpenGLFunctions created with non-current context
Migrating Start Workbench to Start command... done.
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Body
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Fillet
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Fillet
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch002
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad002
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad002
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch003
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad003
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad003
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch004
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Hole
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Hole
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch005
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pocket
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pocket
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch006
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad004
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad004
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch007
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad005
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pad005
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Sketch008
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pocket001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette2#Pocket001
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Body
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Sketch
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Pad
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Pad
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Sketch001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Pad001
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Pad001
[... cut ...]
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Pad003
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Sketch006
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Hole
<App> PropertyTopoShape.cpp(427): Pending recompute for generating element map: Couverture_sonnette1#Hole
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
```

coldtobi, 2025-01-04:
(https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1051296:1.0 still crashes with wayland)

I can reproduce it here, setting COIN_GL_NO_CURRENT_CONTEXT_CHECK works for me.

euphi, 2025-01-05:
FreeCAD 1.0 

Gentoo: `Installed versions:  1.0.0-r2^t(08:20:25 12/29/24)(X addonmgr bim cam fem gui mesh points reverse smesh techdraw -cloud -debug -designer -idf -inspection -netgen -openscad -pcl -robot -spacenav -surface -test PYTHON_SINGLE_TARGET="python3_12 -python3_10 -python3_11")`

```
FreeCAD: /var/tmp/portage/media-libs/coin-4.0.3/work/coin/src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
[1]    73471 IOT instruction (core dumped)  /usr/lib64/freecad/bin/FreeCAD

```

ohthehugemanatee, 2025-04-06:
`COIN_GL_NO_CURRENT_CONTEXT_CHECK=1` does not solve it for me. I can only open a previously saved file if I unset `unset XDG_SESSION_TYPE` (which is normally set to `wayland` as I'm on swaywm) . 
```
OS: Ubuntu 22.04.5 LTS (sway)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)
```

Installed freecad: 

```
freecad-common/jammy,jammy,now 0.19.2+dfsg1-3ubuntu1 all [installed,auto-removable]
freecad-python3/jammy,now 0.19.2+dfsg1-3ubuntu1 amd64 [installed,auto-removable]
libfreecad-python3-0.19/jammy,now 0.19.2+dfsg1-3ubuntu1 amd64 [installed,auto-removable]
```

maurerle, 2025-04-27:
I have the same problem running on `wayland` on debian testing.
Starting the problem with `COIN_GL_NO_CURRENT_CONTEXT_CHECK=1 freecad` works just fine for me.

```
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.0.
Build type: Release
Python 3.13.2, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

I changed the exec directive to
`Exec=env COIN_GL_NO_CURRENT_CONTEXT_CHECK=1 /usr/bin/freecad - --single-instance %F` in the desktop file `/usr/share/applications/org.freecad.FreeCAD.desktop` as well to have it fixed there.

Anekto, 2025-06-04:
I seem to have the same issue on kwin, kde desktop, ubuntu studio 25.04. freecad is installed via apt and ubuntu repository.

FreeCAD 1.0.0, Libs: 1.0.0R

```
freecad
FreeCAD 1.0.0, Libs: 1.0.0R
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

QOpenGLFunctions created with non-current context
Migrating Start Workbench to Start command... done.
freecad: ./src/glue/gl.cpp:2249: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
Aborted (core dumped)
anekto@anekto:~$ export COIN_GL_NO_CURRENT_CONTEXT_CHECK=1
anekto@anekto:~$ freecad
FreeCAD 1.0.0, Libs: 1.0.0R
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

QOpenGLFunctions created with non-current context
Migrating Start Workbench to Start command... done.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
```

with the export it works otherwise it crashes on trying to open a file. previously on pop os it did not need the workaround but it also might have used xwayland?

luzpaz, 2025-06-04:
Ubuntu repository AFAIR is unreliable, right ?

Anekto, 2025-06-12:
> Ubuntu repository AFAIR is unreliable, right ?

I dont know but pop-os shipped lots of outdated versions from 22.04 which made some stuff very difficult.

25.04 has its own issues but at least it gets recent patches instead of waiting for the 24.04 patch being outdated on release?

luzpaz, 2025-06-12:
We can't expect in-house Ubuntu/Debian packaging  to work if they aren't staying up to date with 3rd party dependencies.

BigCojones, 2025-06-23:
Having the same error on Kubuntu 25.04

Coin workaround at least give me usability in that it doesn't instantly crash, but it is constantly complaining about the error:
`Coin warning in glxglue_isdirect(): Couldn't get current GLX context.`

luzpaz, 2025-08-05:
Heads up: Coin 4.0.5 released (https://github.com/coin3d/coin/releases/tag/v4.0.5)

maurerle, 2025-10-05:
On Debian trixie i now get:
```
Wayland detected. Forcing Qt to use X11 backend (xcb) to avoid Coin3D EGL issue.
FreeCAD 1.0.0, Libs: 1.0.0R
```

So everything is fine - I don't know about the current situation without the [debian specific patch](https://udd.debian.org/patches.cgi?src=freecad&version=1.0.0%2Bdfsg-8)..?

neclimdul, 2025-10-05:
From the discussion [[1](https://github.com/FreeCAD/FreeCAD/issues/9273#issuecomment-1518369657)] [[2](https://github.com/FreeCAD/FreeCAD/issues/9273#issuecomment-2995878127)] I believe it was narrowed down to a problem in [coin3d](https://github.com/coin3d/coin/issues/581) which the developer addressed in their latest release https://github.com/FreeCAD/FreeCAD/issues/9273#issuecomment-3156069611

On arch where I could test FreeCAD with coin3d 4.0.5 easily I wasn't able to recreate this.

So I'm not sure what more FreeCAD can do to fix this at this point. I think its probably distributions responsibility to either update coin3d or provide the appropriate patches to work around the problem like Debian is doing.