FreeCAD/FreeCAD #9277: [Issue] Folder may not be hidden by spacebar

PeterDFC, 2023-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=613178#p613178

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32821 (Git) AppImage
Build type: Release
Branch: master
Hash: eaf9d43f37238c33dd1c47b859a6fdf5b7f629fc
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * dodo 1.0.0
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Originally reported on Jul 26, 2022 with an older FC version at that time:

OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 161
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-15:
@yorikvanhavre have you got any clue what is happening here?

yorikvanhavre, 2023-04-19:
I can confirm the problem too, but no idea what is happening... There is a lot more to it, for ex. switching the "Visibility" property of any object between True and False has no effect anymore. Note that if you remove the group and the wall from the BuildingPart, then everything works normally again.

I suspect this has to do with the Group Extension, that is used by the BuildingPart object, that has bugs since the App::Link merge that have never been really fixed. Somehow it mimmicks the shapes of the objects inside it, so things appear shown or hidden when they in fact aren't

In the NativeIFC addon I used a dirty workaround trick that works for now, which is to set a dumb, minuscule 1mmx1mmx1mm shape to the container, but we should solve this for good. Only I can't find how :/

bgbsww, 2023-12-11:
May I be assigned, please?

PR #11763 (2023-12-18): Gui: fix group visibility toggle
Fixes #9277
commit 74a824cc (2023-12-18): Gui: fix group visibility toggle

Fixes #9277
closed 2023-12-18 by yorikvanhavre