FreeCAD/FreeCAD #9278: Problem: Different Assembly Workbenches are not compatible

sliptonic, 2023-04-15:
One of the key findings of the Ondsel Assembly workbench blog series was the need for a common data format for the interchange of assembly information.

The creation of a data standard should be an early priority to allow the various addon assembly workbenches time to adapt and/or build migration scripts.

This issue is for continued discussion about the requirements for a common data standard.

Zolko-123, 2023-04-15:
FreeCAD has defined in the past the [Std_Part](https://wiki.freecad.org/Std_Part) as assembly container:

[image]

> The Std Part element was developed to be the basic building block to create mechanical [assemblies](https://wiki.freecad.org/Assembly)

> In general, the Std Part object was designed to serve as the basic building block to create assemblies

So I would propose to confirm this and decide that a FreeCAD assembly should use the _Std_Part_ as basic building bloc of assemblies. In this way, an assembly built with 1 assembly workbench can be re-used as sub-assembly in a higher-level assembly by another assembly workbench. I don't know how important it is that an assembly made by  one assembly WB can be **edited** by another assembly WB, but what is certainly sure is that they need to be **re-usable**.

Additionally, I suggest to add some other requirements to make this assembly structure both compatible with many assembly workbenches in FreeCAD, and also compatible with what other CAD software have converged-to as common workflow, by using the existing Properties :

> Notably properties that help it manage information in the context of an assembly, for example, **Data** _Type_ , _Id_ , _License_ , _LicenseURL_ and _Group_ . 

For visual clue, to differentiate between a regular Std_Part and a Std_Part used for an assembly, the _Type_ property can be set to "Assembly" which gives another icon in the model tree:

[image]

This icon is already present in the FreeCAD repository, is visually similar to the SolidWorks and Siemsn NX assembly icons, and thus will be easily identified : 

[image]

[image]

As a summary, a common data format for assemblies could be created inside a new document by :

```
assembly = App.ActiveDocument.addObject( 'App::Part', 'Assembly' )
assembly.Type = 'Assembly'
```

This would conserve and extend existing FreeCAD documentation without the need to re-write everything, allow any FreeCAD assembly WB to identify and re-use assemblies, and would follow industry good practices.

aiksiongkoh, 2023-04-24:
Zolko's suggestion sounds good to me.

adrianinsaval, 2023-05-29:
Is there any reason why we need to differentiate a regular Part container from an assembly container? why not make them one and the same, (maybe even rename the container to Assembly)

Zolko-123, 2023-05-29:
well, I see 2 use-cases, but I'm not sure how relevant they are:

- parts using booleans, where you design 2 shapes and subtract (or add or intersect) them together: in this case, you can use a _Std_Part_ container for the final result, which is not an assembly but a complex yet standalone part
- for surface functions, where you design 3D surfaces and cut and intersect them, producing a _Std_Part_ which is a general purpose 3D container

In both of these cases, the end product is not really an assembly, so that would be a reason to differentiate. What do you think ?

adrianinsaval, 2023-05-29:
The first one seems legit, for the second one it's not a Part either so maybe you could just use a compound or group.

My real motivation for bringing this up is that in Part Design the term Body does not fit right and carries some unfortunate expectations from some users (single solid rule) that were needlessly added to the software as limitations, but we can't easily use the Part name because it's already used here... So I'm thinking perhaps we can call the current Part container just Container, that's very generic but still fits IMO. Or maybe we adopt the Part container in PD but set the type to `CumulativeSolid` or `PartDesign`, I should probably move this to it's own issue though.

PaddleStroke, 2023-08-24:
I agree with @adrianinsaval, we should rename the Std_Part to Std_Assembly as the name Part is confusing every one.

Regarding the usecases you mentioned @Zolko-123 it seems that Std_Group would be a good candidate. Besides I can't understand your first usecase. When you make a boolean with 2 parts, you end up with a single entity no? At least in Part and PartDesign.

[image]

PaddleStroke, 2023-08-24:
Then there is the question about the name of part Design Body, but that's another topic.

PaddleStroke, 2023-08-24:
However now that we are planing to make an integrated assembly workbench, the question is : why would the Std_Assembly (Std_Part) be part of core and not part of the assembly wb?

wwmayer, 2023-08-24:
> However now that we are planing to make an integrated assembly workbench, the question is : why would the Std_Assembly (Std_Part) be part of core and not part of the assembly wb?

This isn't a good idea because several other modules already use Part containers (Part Design, import, ...). So, moving this functionality to another application module will result into higher dependencies among application modules and this way the modular architecture more and more turns into a monolithic architecture. The result is that we don't only reduce flexibility but also increase build time.

PaddleStroke, 2023-08-24:
Thanks for the explanation @wwmayer .

grd, 2023-08-27:
> This issue is for continued discussion about the requirements for a common data standard.

Where are the requirements / specs? I can't find them and I think that the specs document should be here or linked to somewhere else.

yorikvanhavre, 2023-08-28:
So basically the difference between part and assembly is that an assembly would be a part container which contains constraints? If that's the case, there is no fundamental difference between a current std_part and what would be a std_assembly? Is there any issue in calling the current std_part an assembly? I kind of like that idea. After all, in the common language, any stuff put together one way or another is an assembly... That would be one of the issues with the"Part" name solved.

But the same problem might then rise again: Why is there an "assembly" object outside the assembly WB...

PaddleStroke, 2023-08-28:
@yorikvanhavre from what I understood : 
- Currently there are no differences between App::Part and 'Assembly'. It's just the icon that changes when you change the string property 'Type' to 'Assembly'.
- App::Part is a container that is designed to be an assembly.
- The name 'Part' exist because apparantly in the industry the 'Part' concept can be made of several solid bodies as long as they are all fixed together. Basically a 'Part' is an assembly of solids but not moving.

So 'Part' is an assembly of solids but not moving. So for example a toothbrush is a Part.
I think it's a confusing way of seing things as a toothbrush can be considered an assembly of several things. Anyway I read that this is the terminology that SolidWorks is using so it is perhaps indeed industry standard. Though I note that OnShape for instance is not using this 'Part' concept. Instead they use 'Part' to describe solid body. Which imo makes more sense.

WandererFan, 2023-08-28:
App::Part is neither a part nor an assembly.  It is a grouping of DocumentObjects with a position and orientation (a Placement or LCS).

An Assembly would include an aggregation of such groups (as well as simple Part::Feature objects) plus a bunch of data about the Assembly (BOM, Constraints, etc).

In the Open Assembly Model, a part is something that can not be subdivided.  An assembly is a collection of (sub-) assemblies and/or parts, plus the necessary relationships between them.  The general term to include both parts and (sub-) assemblies is a "component".

An assembly is also an OAM Artifact, so it has Specification, Form, Function and Behaviour information.

Zolko-123, 2023-08-29:
> Is there any issue in calling the current std_part an assembly?

yes, documentation. Also: why do you want to replace _Std_Part_ and not simply add a new command _Std_Assembly_ ?

> But the same problem might then rise again: Why is there an "assembly" object outside the assembly WB...

yes, this is the fundamental problem : that an assembly OBJECT – and not WB ! – be part of FreeCAD and usable by all assembly tools. So the solution to the problem becomes simple: add a new command Std_Assembly to the structure toolbar, that creates the assembly object as mentioned here :

```
assembly = App.ActiveDocument.addObject( 'App::Part', 'Assembly' )
assembly.Type = 'Assembly'
```

The problem is that it must be checked first that there is no other object called "Assembly" in the document. So the code for the new _Std_Assembly_ command would be:

```
assy = App.ActiveDocument.getObject('Assembly')
if assy is None:
    assembly = App.ActiveDocument.addObject( 'App::Part', 'Assembly' )
    assembly.Type = 'Assembly'
else:
    ...
```

yorikvanhavre, 2023-08-29:
That's getting clearer...

So a "part" is just a "piece" of an assembly. It can then be just anything. In that sense, std_part is misleading because it is in fact rather a container, not *necessarily* a part. It would only be a part when it's part of something. In fact, we should not call anything a part by default. Only the user should know and define when something is a part or not

And then an assembly is just a collection of parts that make up that assembly. There is no further formal requirement for it.

Then then way our std_part works, where you can "switch" it between "Part" and "Assembly", seems to me actually very good...

grd, 2023-08-29:
IMO naming in FreeCAD can improve **_a lo_**t. I know it's history but release 1.0 is about to come and most of the time that's when you rename things (that is why you have releases to begin with). This one should be on the list... if there was a list. Yes, I agree completely with what Zolko said and I also realize the consequences of it all. **_But confused naming is hurting anyone who dives into FreeCAD_** and my autistic mind has problems with it ;-)

sliptonic, 2023-08-29:
> The problem is that it must be checked first that there is no other object called "Assembly" in the document.

Can you elaborate more on why you think this is necessary?  Users will likely feel that it is awkward if they are forced to use multiple documents for simple things.  It also complicates sharing of moderately complex assemblies.

I think A3 allows for the user to pull in all external references to build a single document assembly regardless of complexity but it builds a nested structure.

Zolko-123, 2023-08-31:
> Can you elaborate more on why you think this is necessary? Users will likely feel that it is awkward if they are forced to use multiple documents for simple things. It also complicates sharing of [...]

you're right, this is fundamental: it's not a necessity, it's a self-imposed limitation to maximize robustness. 

The point of assemblies is to parallelize work by distributing it amongst several designers. This is usually done by cutting large assemblies into smaller ones, and then inserting these sub-assemblies (with App::Link-s) into the larger one. So we want to ensure that all these designers will in effect be able to easily share their "part". If each designer could name the assembly as he wished, we will run into trouble with exotic glyphs (Assembly = Összeállítás → __ssze__ll__t__s).

Another aspect is industry standards: in all other CAD software that I have used, there was only 1 assembly per file. Actually, there was only 1 part per file, of any type. So by calling the default assembly container "_Assembly_" we ensure this uniqueness. It's easy to document, easy to understand, easy to use, easy to distribute. 

In real life, for real mechanical designers, I don't think that this a limitation because 1 assembly by file/document is what they do anyway .

> I think A3 allows for the user to pull in all external references to build a single document assembly regardless of complexity but it builds a nested structure.

yes, this is what App::Link does. Also in Asm4. The difference is that the A3 container is not an App::Part but a custom Part::FeaturePython (or similar, custom to that workbench). A2+ also uses a custom container and uses a pre-App::Link type of linking. I think that @realthunder has said that it wouldn't be too difficult to convert the A3 custom assembly container to an App::Part container. For A2+ it might be more complicated.

WandererFan, 2023-09-01:
[OAMOntology.pdf](https://github.com/FreeCAD/FreeCAD/files/12498950/OAMOntology.pdf)
[OpenAssemblyModel.pdf](https://github.com/FreeCAD/FreeCAD/files/12498951/OpenAssemblyModel.pdf)
[OpenAssemblyModel2.pdf](https://github.com/FreeCAD/FreeCAD/files/12498954/OpenAssemblyModel2.pdf)

pierreporte, 2023-10-28:
As a mechanical engineer, I agree with @Zolko-123. In my opinion, the ideal situations for me would be:

* 4 kinds of containers: assembly, part, body, geometrical set.
* A part can contain several bodies and geometrical sets, sharing a common origin (like Catia and unlike FreeCAD today).
* Geometrical sets can contain datum elements (planes, axis, etc.), sketches, surfaces, and operations on them (like intersections). Its the same concept as Catia.
* An assembly can contain geometrical sets whose content can be used by its parts and sub-assemblies (like Solidworks, unlike Catia). Now FreeCAD automatically creates a body when creating a sketch directly in a part.

For the file structure, there should be only one root (a part or an assembly). However, it would be useful if the file is able to fully contain sub-assemblies and sub-parts. This is not intended to allow users to make one-file projects (it would be possible but it is not the intended workflow) but rather to allow to make quasi-parts easier to produce and manage. “Quasi-parts” (not a real term) is an assembly with one main part and several accessories like pins and helicoils. In the industry, these assemblies are basically treated as parts (hence the term). This quasi-part feature is not a requirement for me but it must be common enough because Solidworks does it.

czinehuba, 2023-11-08:
Yes. Same opinion here. And other software work similarly with subtle differences but the same principle.  
Assembly can have parts or sub-assemblies (or nested sub-assemblies); whereas parts can have multiple bodies. 
Difference between parts and bodies from assembly stand point is the a part as a default when inserted has 6DOF so it can freely move until constraints are applied. Whereas bodies have 0DOF and do not move within the part container. And can't have assembly constraint applied to them. Take for example a bearing. That can be modelled as a multi body part, but then you wouldn't have to worry about constraining it's individual bodies when inserted in the assembly.

sliptonic, 2024-03-13:
I'm tagging the issue as a Google Summer of Code candidate project.
Having an integrated tool or migration script to migrate assemblies between workbenches would be an extremely valuable thing.  It's a well-bounded unit of work so it's a good one for a student to attack.

Zolko-123, 2024-09-13:
This problem "Different Assembly Workbenches are not compatible" #9278 still persists in the latest 1.0.0-rc-38643 build.

When creating an Assembly in the Assembly workbench, it doesn't create an `App::Part` called "Assembly" as discussed here, but creates an `Assembly::AssemblyObject` called "Assembly". This new assembly _TypeId_ isn't compatible with any of the existing assembly workbenches A2+, Asm3 and Asm4. It creates yet another new incompatible workbench. 

And while all previous assembly containers were compatible with previous FreeCAD versions, this new assembly object cannot be used with previous versions : this is an unfortunate regressions, what is the reason for it ? What prevented the use of the Std_Part (App::Part)  container to be used, as has been agreed and has been documented for years ?

> The [Std Part](https://wiki.freecad.org/Std_Part) object was designed to serve as the basic building block to create assemblies.

https://wiki.freecad.org/Assembly