FreeCAD/FreeCAD #9282: [Issue] Translation updates take too long

chennes, 2023-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a - process issue
```

### Subproject(s) affected?

None

### Issue description

In PR #9275 @ronnystandtke points out that our current translation workflow introduces a very long lag between when a string is made translatable in the source code, and when that string is actually available in translated form.

I think there are two things at work here:
1. The roundtrip from source code to CrowdIn, and vice versa, is entirely manual right now, and is not done one a regular schedule This means that developers and translators never know when to expect their work to appear.
2. There is not good documentation on how developers can immediately test that their newly-translatable strings can in fact be translated. I believe this is done by putting translation files in $USER_APP_DATA_DIR/languages.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-15:
How is the upload to crowdin made? Is there mechanism to have a token to authenticate and automate the process? If so it could be done on a schedule via github actions.

It might also be possible to make an action to generate a PR for updating the translations in the repo, but that might clutter the repo with useless branches.

d1nuc0m, 2023-04-30:
> Is there mechanism to have a token to authenticate and automate the process?

These may be interesting:

- [GitHub Crowdin Action [GitHub]](https://github.com/crowdin/github-action)
- [GitHub Actions [Crowdin]](https://store.crowdin.com/github-actions-webhook)

chennes, 2023-05-01:
I'm ultimately going to start automating this with CrowdIn's GitHub actions, but before I do I want to integrate the generation of the QM and QRC files into the build process. I think we can get that working in a couple of weeks here.

yorikvanhavre, 2023-05-01:
We definitely need to remove qm files from the source before ramping up translations frequency, otherwise it will just inflate the repo size dramatically.

Also, there is the issue that when we use another version of lupdate, it sometimes rewrite all the ts files (changes the order somehow). This is less critical of course, but might be a good idea to define a version to be used and allow no other...