FreeCAD/FreeCAD #9283: [Problem] Sketcher: solver problems while dragging trigger absurd amount of warning and error messages

adrianinsaval, 2023-04-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32862 (Git)
Build type: Release
Branch: master
Hash: 6ca182565ffc6191b7183a9905b6cdd8f714e6f7
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
Installed mods: 
  * A2plus 0.4.54b
  * AirPlaneDesign
  * Assembly3 0.11.4
  * Assembly4 0.12.7
  * Autoload
  * CfdOF 1.21.7
  * CubeMenu
  * CurvedShapes 1.0.0
  * Curves 0.6.8
  * DynamicData 2.34.0
  * fasteners 0.3.38
  * FCGear 1.0.0
  * FC_SU
  * Glass
  * IconThemes
  * Plot
  * SelectorToolbar
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

[image]

To reproduce, create a rounded rectangle, drag the corner point to make the rectangle smaller until the errors are triggered.

I understand that the dragging led to an invalid state, but it doesn't seem right to flood the user with obscure error messages for something that does not really cause any serious problem, the dragging action is just reverted.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-15:
@AjinkyaDahale this might be of your interest

AjinkyaDahale, 2023-04-26:
Sorry for the late reply @adrianinsaval. I'm not sure why there are so many error messages, specifically are they one for each zero-radius or because of multiple solves. There should definitely be an error report but not so many. The error UI could also be unified, so that reports should only be at one or two places.

closed 2025-01-20 by github-actions[bot]