FreeCAD/FreeCAD #9288: [Issue] Shapestring change does not trigger recompute in PartDesign

Miq1, 2023-04-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77632

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32552 (Git)
Build type: Release
Branch: master
Hash: e2b3a5d2f791fb78a61daac9bdab374058f682f2
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.46.0
  * fasteners 0.4.56
  * kicadStepUpMod 10.18.5
  * Manipulator 1.5.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

See attached simple ShapeString.zip (aka ShapeString.FCStd).

The Placement of the ShapeString has expressions for X and Y to center the ShapeString based on its BoundBox. While the expressions seem to be evaluated upon a change of the ShapeString, the Placement remains unchanged unless you will click on 'Apply' in the Placement dialog.

I would expect the recompute to affect the ShapeString position as well.

### Anything else?

[ShapeString.zip](https://github.com/FreeCAD/FreeCAD/files/11241499/ShapeString.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-21:
this has little or nothing to do with Part Design, shapestring is a draft feature

Roy-043, 2023-04-25:
This issue is more generic. It seems occurs for all scripted objects.

See the attached file with a Part_Tube and a Draft_Rectangle.
[placement_expressions_scripted_objects.zip](https://github.com/FreeCAD/FreeCAD/files/11320349/placement_expressions_scripted_objects.zip)

An object created with this code:
https://wiki.freecad.org/Create_a_FeaturePython_object_part_II#Complete_code
Also shows the issue.

Roy-043, 2025-01-02:
This is still an issue.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```