FreeCAD/FreeCAD #9296: [Issue] Tooltip content is rendered in status line without filtering of paragraph and line-break tags

neildarlow, 2023-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * Assembly4 0.50.1
[/code]
```

### Subproject(s) affected?

None

### Issue description

When hovering over tooltips the entire content is rendered in the status line including HTML tags such as paragraph and line-break. Some tooltips overwhelm the status line and it might be better to just display the first line of the tooltip.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-04-17:
I think we should just drop the tooltip display from the status line entirely.

luzpaz, 2023-04-17:
Can you reproduce on dev version? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds? )

chennes, 2023-04-17:
Forum discussion here: https://forum.freecad.org/viewtopic.php?t=77657

adrianinsaval, 2023-04-18:
agreed with dropping the tooltips on the status bar

maxwxyz, 2024-03-03:
@PaddleStroke FYI, as discussed with the Assembly WB tooltips

PaddleStroke, 2024-03-04:
+1 to drop tooltips in status bar.

yorikvanhavre, 2024-03-04:
-1 - I like the tooltip in the status bar, it's immediate, while the tooltip takes a delay to appear... When browsing menus it's super cool. Removing the tooltips in the status bar is IMHO removing a feature because of a bug we introduced ourselves. We should either fix that (clean the html tags out) or we make use of the StatusTip (currently == Tooltip), which is apparently designed for that...

neildarlow, 2024-03-18:
Didn't Yorik suggest using a different object to present the status line content. There was a suggestion it would handle the HTML stripping automatically.

Regards,
Neil

chennes, 2024-03-22:
It's not that it would be handled automatically, it's that we'd write the status tips specifically for the status bar, and thus omit HTML, line breaks, etc. That is the best long-term solution, but in the short term we probably should just add some simple filtering to the value before we set it. Someday perhaps we can remove that filter, but it won't harm anything in the meantime. This sounds like "low-hanging fruit" to me :).

neildarlow, 2024-03-22:
Why not fix it properly now while it's fresh in the minds of people. Also, bear in mind that criticisms are made of FreeCAD's UI when comparing with commercial offerings and this would stick-out like a sore thumb if not fixed. Worse case is that Ondsel fix it and FreeCAD doesn't.

Roy-043, 2026-08-16:
Several Assembly commands have this issue.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/eb43f3bdb0bb7b1902b3e4784e8eacb2f599f44a

Rich tooltips are converted to plain text for the status bar, with the first non-empty paragraph kept on one line.