FreeCAD/FreeCAD #9297: [Problem] Task panels are left open when the document is closed

sliptonic, 2023-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32854 +4 (Git)
Build type: Unknown
Branch: (HEAD detached from upstream/pr/8459)
Hash: 08aa5d2f6ab8c2d8806c486d7b78d5d77ded66fc
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.46.0
  * Assembly4 0.12.5
  * fasteners 0.4.55
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * Render 2023.2.2
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.59
  * QuickMeasure 2022.10.28
  * Nodes 0.1.36
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.11.4
  * ExtremeProDark 2.6.4
  * FeedsAndSpeeds.bak
  * Curves 0.6.5
  * A2plus 0.4.60k
  * Dracula 0.0.2
[/code]
```

### Subproject(s) affected?

None

### Problem description

If the user has an active task open with the task panel visible in the comboview and then closes the document without first ending the task, the panel is left open.

In most cases this doesn't cause any significant problem.  It will, however, usually result in many errors and warnings in the console when the user clicks on 'cancel', 'ok', or 'apply'  

Regardless, leaving the task panel open makes the application feel buggy and unstable.

Whenever a document is closed, the system should check if there is any open and dependent task panel and close that as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-04-20:
I can't reproduce. To my scarce knowledge, [this piece](https://github.com/FreeCAD/FreeCAD/blob/7d2107be23e76c33541beb664daa3f59987c2f1d/src/Gui/Document.cpp#L2017-L2032) of code closes taskview when document is being closed.

luzpaz, 2023-04-20:
@xtemp09 do you mind adding the full About info on the build you tested on? TIA!

sliptonic, 2023-04-20:
There may be others but this is the one that prompted the issue:
Steps to reproduce:  

Create a new document.
Drag an image into the 3D window.  This creates an image plane.
Double click the tree node to open the task panel
Close the document by clicking the 'x' on the MDI tab.

xtemp09, 2023-04-20:
```ruby
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

I can reproduce it with Python workbenches. The other workbenches seem to work.

https://user-images.githubusercontent.com/9303235/233453110-742445d0-9fd9-420b-a7da-0916d99de915.mp4

xtemp09, 2023-05-21:
This issue is somehow related to #9626. Python workbenches have issues.

1. Go to _Draft_, do something, keeping the task active.
2. Close the project.
3. Start a new project.
4. Go to _Arch_.

You will have buttons of both _Arch_ and _Draft_ at the toolbars.

[image]

<details><summary>Version info</summary>

</details>

0penBrain, 2023-05-21:
> You will have buttons of both _Arch_ and _Draft_ at the toolbar.

I ever saw Arch also importing the Draft tools.

xtemp09, 2023-05-22:
Alright, false alarm about the toolbars. I've never used Arch.

maxwxyz, 2024-07-01:
Is this still relevant? The mentioned issue was closed.

PaddleStroke, 2024-07-02:
I can reproduce the 2 bugs mentioned here : 
- The 'Image plane setting' taskbox does not close if document is closed.
- Draft taskboxes does not close when the document is closed.

@wwmayer do you perhaps know what is going on? I'm not sure to understand [the snippet that xtemp09 posted](https://github.com/FreeCAD/FreeCAD/blob/7d2107be23e76c33541beb664daa3f59987c2f1d/src/Gui/Document.cpp#L2017-L2032).
If I understand correctly, the document being closed trigger an unset edit of the document object being edited. Said unset edit should close any open task. But is there necessarily an object in edit if an task is opened? The draft line tool for example does not seem to have any document object being in edit no?

Jookia, 2024-07-03:
Is #15130 a case of this? @theosib

theosib, 2024-07-03:
> Is #15130 a case of this? @theosib

I don't think it is. In my case, the document isn't being closed. There's just a window that should remain on top, but it gets hidden, blocking up other elements of the UI.

PaddleStroke, 2024-07-15:
@yorikvanhavre will you tackle the draft taskboxes ? Werner expanded the task dialog to make it possible : 
https://github.com/FreeCAD/FreeCAD/pull/15116

PaddleStroke, 2024-07-15:
The PR https://github.com/FreeCAD/FreeCAD/pull/15441 fix this bug for the Task Image Dialog