FreeCAD/FreeCAD #9312: [Problem] MDIViews not accessible correctly

xraf32, 2023-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32867 (Git) AppImage
Build type: Release
Branch: master
Hash: 89ebbfebcd2574aa59c72795add4c9aa793a4212
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * dodo 1.0.0
  * fasteners 0.4.56
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

Disclaimer: I am not a software developer and English is not my native language.
This bug was reported in 2019, so it is not here on GitHub but in another site. I put the links in the "anything else?" section. This bug was closed but only half of it was fixed. Steps to reproduce:
1. Open New Document
2. Open Spreadsheet Workbench, add Spreadsheet
3. Double click spreadsheet to open it in the UI
4. Paste into python console: Gui.ActiveDocument.mdiViewsOfType("SpreadSheet::SheetView")
--> Type not known, but should be (according to the report of 2019)
5. Paste into python console: Gui.ActiveDocument.mdiViewsOfType("Gui::MDIView")
--> spreadSheet MDI view is correctly listed (in 2019 it was not. This is the half which was fixed)
6. Go into TechDraw Workbench, add Page
7. Paste into python console: Gui.ActiveDocument.mdiViewsOfType("TechDraw::PageView")
--> Type not known, but should be (according to the report of 2019)
7. Paste into python console: Gui.ActiveDocument.mdiViewsOfType("Gui::MDIView")
--> TechDraw MDI Page is listed as "None" (in 2019 it was not. This is the half which was fixed)

### Anything else?

Report of 2019: https://tracker.freecad.org/view.php?id=4125
Forum discussion that lead to the bug report: https://forum.freecad.org/viewtopic.php?style=10&t=39249
It is marked as "solved" in the forum. I could not find any more info about the valid arguments of the function Gui.ActiveDocument.mdiViewsOfType(), so I don't know if they changed. If they did not, then the bug is still there. It is also present in the latest stable release (0.20.2).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-04-19:
I don't know the history and have had no use for this functionality but from the code I conclude that the following lines in your testing should be changed:

`Gui.ActiveDocument.mdiViewsOfType("SpreadSheet::SheetView")`

should actually be:

`Gui.ActiveDocument.mdiViewsOfType("SpreadsheetGui::SheetView")`

and

`Gui.ActiveDocument.mdiViewsOfType("TechDraw::PageView")`

should actually be:

`Gui.ActiveDocument.mdiViewsOfType("TechDrawGui::MDIViewPage")`

Syres916, 2023-04-19:
So I'm struggling to conclude it's a regression because I don't know where the incorrect types came from.

xraf32, 2023-04-19:
Is there a way to get a list of all accepted strings for a function like this? Searching the internet for the function all I can find are that old bug report, where I don't see mentions that these strings are wrong, and this rather recent discussion about a Macro:
https://forum.freecad.org/viewtopic.php?style=1&t=76990
So now I know of 3 valid string arguments for `mdiViewsOfType()`
`"SpreadsheetGui::SheetView"`
`"TechDrawGui::MDIViewPage"`
`"Gui::View3DInventor"`

I have the same difficulty with any function that takes the name of a type as input. For example `getActiveObject()`:
`Gui.ActiveDocument.mdiViewsOfType("Gui::View3DInventor")[0].cast_to_base().getActiveObject(???)`

Syres916, 2023-04-19:
On Linux Mint I used the following in a Terminal to find all relevant lines in the source code:

`grep -r '/home/username/freecad-daily-source/src' -e 'TYPESYSTEM_SOURCE_ABSTRACT(' | grep -r '/home/username/freecad-daily-source/src' -e 'Gui::MDIView)'`

which gave me the 7 below:

```/home/username/freecad-daily-source/src/Gui/View3DInventor.cpp:TYPESYSTEM_SOURCE_ABSTRACT(Gui::View3DInventor,Gui::MDIView)
/home/username/freecad-daily-source/src/Gui/SplitView3DInventor.cpp:TYPESYSTEM_SOURCE_ABSTRACT(Gui::AbstractSplitView,Gui::MDIView)
/home/username/freecad-daily-source/src/Gui/TextDocumentEditorView.cpp:TYPESYSTEM_SOURCE_ABSTRACT(Gui::TextDocumentEditorView, Gui::MDIView)
/home/username/freecad-daily-source/src/Gui/EditorView.cpp:TYPESYSTEM_SOURCE_ABSTRACT(Gui::EditorView, Gui::MDIView)
/home/username/freecad-daily-source/src/Gui/MDIViewPyWrap.cpp:TYPESYSTEM_SOURCE_ABSTRACT(Gui::MDIViewPyWrap,Gui::MDIView)
/home/username/freecad-daily-source/src/Mod/Spreadsheet/Gui/SpreadsheetView.cpp:TYPESYSTEM_SOURCE_ABSTRACT(SpreadsheetGui::SheetView, Gui::MDIView)
/home/username/freecad-daily-source/src/Mod/Web/Gui/BrowserView.cpp:TYPESYSTEM_SOURCE_ABSTRACT(WebGui::BrowserView, Gui::MDIView)
/home/username/freecad-daily-source/src/Mod/TechDraw/Gui/MDIViewPage.cpp:TYPESYSTEM_SOURCE_ABSTRACT(TechDrawGui::MDIViewPage, Gui::MDIView)
```

xraf32, 2023-04-19:
I just found out how to search on github, but I am not going to pretend that I understand why this is the pattern I should search:
TYPESYSTEM_SOURCE_ABSTRACT( and Gui::MDIView)

It gives me 9 results. Your 7 plus these 2:
src/Gui/View3DInventor.cpp : TYPESYSTEM_SOURCE_ABSTRACT(Gui::View3DInventor,Gui::MDIView)
src/Gui/MDIView.cpp : TYPESYSTEM_SOURCE_ABSTRACT(Gui::MDIView,Gui::BaseView)

The first is the one mentioned in the recent Macro discussion. The second is from the steps 5 and 8 of the report from 2019. So, in total there are 8 specific view types which inherit (?) from Gui::MDIView? Is this somewhat correct?

Syres916, 2023-04-20:
8 specific types is correct, I've not looked into why the View3DInventor one wasn't picked up by my search. Now this issue is resolved and should be closed, if you need any further coding assistance I highly recommend opening a topic in the specific section of the forum https://forum.freecad.org/viewforum.php?f=22&sid=0f0e697a07221655ca393243ef5bc659

closed 2023-04-20 by xraf32