FreeCAD/FreeCAD #9321: [Issue] Add a CITATION file to FC repo

luzpaz, 2023-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77756

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Issue description

[Github instructions](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files) on how to go about doing this. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-15:
https://zenodo.org/search?page=1&size=20&q=freecad

luzpaz, 2023-08-15:
## How can researchers cite FreeCAD
Notes:
* Write a JOSS paper (contact https://twitter.com/KMMoerman at https://twitter.com/JOSS_TheOJ). JOSS is free to publish in and a week established journal for software. Both the short paper and the software are reviewed.
* Another option, which will not require peer review, is too archive a copy of FreeCAD on Zenodo. This will give you a DOI for citing too. However is is a data object DOI (not a paper) and is not peer reviewed so if you have academic contributors who want proper "academic credit" for their work then the paper route, e.g. with JOSS is best
* FYI JOSS papers can be 1 pagers and are super easy to write, it really is mostly about the software so we are different from other journals in that respect
* The blog also mentions the journal of open hardware (https://twitter.com/KMMoerman is also an editor). That is certainly also an option. However it will require a longer more traditional paper, will be slower to review/publish, and the journal is not free to publish in (around 200 pounds).

luzpaz, 2023-08-16:
@pieterhijma is this you, https://zenodo.org/record/7633508  ?

pieterhijma, 2023-08-21:
@luzpaz, yes that's me. This particular link is the Arch Linux package with our branch of FreeCAD. The link for the release of our branch is here: https://zenodo.org/record/7633414

yorikvanhavre, 2025-01-06:
Since we don't have a clear policy of how to cite FreeCAD (the dos and don'ts), and I can't think of anything that would be a "don't", or why we would want people to cite FreeCAD in a specific way (just saying "FreeCAD" is good enough I guess :smile: ), I suggest we close this, and reopen it when the need arises?