FreeCAD/FreeCAD #9324: [Issue]  Arch IFC export, wrong placement on nested compound objects

berndhahnebach, 2023-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77169&start=20#p671764

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32931 (Git)
Build type: Unknown
Branch: femtmp
Hash: 43db8a115caa56eba358e73a2e0e4190f37958dc
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 6.3.0, OCC 7.5.1
Locale: German/Switzerland (de_CH)
Installed mods: 
...
[/code]
```

### Subproject(s) affected?

Arch

### Issue description

On some nested compound objects the placement in ifc is wrong.  see link to forum post. There are example files as well. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berndhahnebach, 2023-04-19:
@yorikvanhavre  Would it be possible to have a look at this?

berndhahnebach, 2023-04-19:
- force brep is set to Tre
- serialier is set to False
- IFC2X3 was used in Arch prefs

yorikvanhavre, 2023-04-20:
That topic has become confused, there are several issues discussed there and I don't understand anymore :sweat_smile: 

Is the main problem the fact that a same object contains both solids and faces?

berndhahnebach, 2023-04-20:
Just use the files from the post of the forum link in the issue here. They show the problem described in this issue.

berndhahnebach, 2023-04-20:
I made a new topic and updated the link in this issue. This is the one to use https://forum.freecad.org/viewtopic.php?t=77785 Should be easy for you to confirm the problem.

yorikvanhavre, 2023-04-24:
There are several issues here:

- The inner shape is created as an assembly but it has no aggregated objects
- The placement of the outer assembly is not computed

yorikvanhavre, 2023-06-13:
Removed this from 0.21 target as it is not blocking

yorikvanhavre, 2025-01-06:
This is still a valid bug, but it's rather a corner case... I'm afraid it's a bit low priority :sweat_smile:

closed 2025-06-11 by github-actions[bot]

maxwxyz, 2026-11-07:
The fix from https://github.com/FreeCAD/FreeCAD/pull/31616 is also in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/26ff23ecd26a19605c2500efd5efb548c4f725ae

Part Compounds are exported as a single shape rather than an assembly of their links.