FreeCAD/FreeCAD #9330: [Problem] Offset on wacom pointer on production builds (0.20.x)

studiofuga, 2023-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When using a Wacom Intuos tablet, the pointer has an offset that prevents the application from being used

This seems to be due to a regression in Qt 5.12.4 (this is why I'm posting it as a problem and not a bug), as explained in [multiple post](https://bbs.archlinux.org/viewtopic.php?id=248540) [around the internet](https://obsproject.com/forum/threads/mouse-offset-on-obs-window.109957/). The following [Bug Report](https://bugreports.qt.io/browse/QTBUG-77826) in Qt explains a workaround.

Unfortunately the official build is linked to an affected version of Qt. Also Ubuntu 22.4 is affected.

The workaround is to set the following env variable: `QT_XCB_TABLET_LEGACY_COORDINATES=""`

### Anything else?

Qt Bug report: https://bugreports.qt.io/browse/QTBUG-77826

https://obsproject.com/forum/threads/mouse-offset-on-obs-window.109957/

https://bbs.archlinux.org/viewtopic.php?id=248540

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-21:
the qt bug report lists 5.15.0 as having this fixed, your freecad info says you are using 5.15.8 are you still affected by this? does the workaround work for you?

studiofuga, 2023-04-25:
Yes, I can confirm that the issue was still relevant and the workaround 100% fixes the issue.

adrianinsaval, 2023-04-27:
Is there any expected side effect from setting that env variable?

studiofuga, 2023-05-09:
No side effects so far.