FreeCAD/FreeCAD #9333: FreeCAD crashes when using chamfer tool in the Part Designer

gregreenwood, 2023-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

while in the Part Design work bench.  Selecting an edge and applying a chamfer causes the program to crash.  I believe the problem is due to the default size of the chamfer (1mm) is greater than what is able to be performed.  However, if that is true, the behavior should be to alert the user, not crash to the desktop.  The fillet tool does not crash and able to complete the action.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-21:
Most likely a bug in occt 7.5.1 used by your system. Please try the appimage, flatpak or snap version instead. If you are able to reproduce in one of those, please provide a file with clear indication on which edge causes the crash.

gregreenwood, 2023-04-21:
thank you for pointing me to app image.  I originally used the latest ppa and so received 20.1.  
Now Freecad does not crash to the desktop and if a chamfer that will not fit is tried the error dialog shows up.
I like the appimage concept.

gregreenwood, 2023-04-21:
when I open (start) the app image via a terminal I get:
User path has changed to /home/greg/.local/share/FreeCAD/. Please move user modules and macros
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory

do i need to move files from the previous install location to the path indicated?

adrianinsaval, 2023-04-21:
That talks about the data that used to be stored in ~/. FreeCAD since you where already using 0.21 I don't think you need to move anything

luzpaz, 2023-04-24:
This looks like downstream packaging issues. Should we close?

adrianinsaval, 2023-04-24:
Yes, this is just ubuntu having old libs, theoretically we could package our version of occt in the ppa, @kkremitzki is that something you would want to do?

gregreenwood, 2023-04-24:
Yes close.

gregreenwood, 2023-04-24:
Actually I like the appimage better.  Wish I had seen this early.  One
file.  Nice and simple.

closed 2023-04-24 by adrianinsaval