FreeCAD/FreeCAD #9364: [Issue] WM_INPUTLANGCHANGEREQUEST seems to freeze FreeCAD

HYBRID-BEING, 2023-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * dodo 1.0.0
  * Dracula 0.0.2
  * ExtremeProDark 2.6.4
  * freecad.gears 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

None

### Issue description

I'm using a little tool called [Mahou](https://github.com/BladeMight/Mahou) to control my keyboard language/layout. According to the [documented issue](https://github.com/BladeMight/Mahou/wiki/Problems-in-some-programs%3F#%EF%B8%8F-this-way-works-only-with-mahou-2510-and-above) (I cannot use provided solution in my setup, unfortunately), Mahou sends WM_INPUTLANGCHANGEREQUEST, which causes some applications to freeze.
It seems to be the case with FreeCAD as well - on switching layout via Mahou hotkeys FreeCAD freezes for good, forcing me to kill the process.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-04-23:
I use Caramba Switcher and have never encountered any trouble.

HYBRID-BEING, 2023-04-23:
Well, i don't see any mentions of it using WM_INPUTLANGCHANGEREQUEST, so it doesn't really prove or disprove anything. It also doesn't look like Caramba Switcher has features for my use case (two separate hotkeys, one to toggle between EN and RU, another to switch to JAP).

chennes, 2023-04-23:
Possibly related Stack Overflow response: https://stackoverflow.com/questions/55579777/sending-wm-inputlangchangerequest-to-some-applications-hangs-them

wwmayer, 2023-05-03:
The issue can easily reproduced with a small test application. However, FreeCAD must have opened the start page.

```
HWND globalFCMainWnd = 0;

// Determines the window handle to FreeCAD 0.21
BOOL CALLBACK enumFCProc(__in  HWND hWnd, __in  LPARAM lParam)
{
    DWORD lpdwProcessId;
    GetWindowThreadProcessId(hWnd, &lpdwProcessId);
    if (globalFCMainWnd == 0) {
        int length = ::GetWindowTextLength(hWnd);
        if (0 == length) return TRUE;

        TCHAR* buffer;
        buffer = new TCHAR[length + 1];
        memset(buffer, 0, (length + 1) * sizeof(TCHAR));

        GetWindowText(hWnd, buffer, length + 1);
        std::string windowTitle(buffer);
        delete[] buffer;
        if (windowTitle == "FreeCAD 0.21.0")
            globalFCMainWnd = hWnd;
    }
    return TRUE;
}

// Send the event directly to FreeCAD
      ::EnumWindows(enumFCProc, GetCurrentProcessId());
      if (globalFCMainWnd) {
          ::SendMessage(globalFCMainWnd, WM_INPUTLANGCHANGEREQUEST, 0, HKL_NEXT);
          globalFCMainWnd = 0;
      } 
```

The solution is to implement a native event filter that ignores all WM_INPUTLANGCHANGEREQUEST events

https://github.com/FreeCAD/FreeCAD/pull/9472

commit af12d4bf (2023-05-04): fixes #9364: WM_INPUTLANGCHANGEREQUEST seems to freeze FreeCAD
closed 2023-05-04 by wwmayer