FreeCAD/FreeCAD #9365: [Problem] Cannot input accented characters

xraf32, 2023-04-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * dodo 1.0.0
  * fasteners 0.4.56
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

1. Open a TechDraw page
2. Add an annotation
3. Type accented characters like "ã õ á ô ë", but get regular characters instead "a o a o e"
4. Copy and paste the accented characters, and it works fine
Same behaviour in the editable TechDraw Template fields

### Anything else?

Tried changing the language to Portuguese to check whether it was a locale problem, but the behaviour remained. This does not happen in 0.19, the version installed from the regular Debian repository:

```
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United Kingdom (en_GB)
```

Also, when I opened in 0.20.2 a file created in 0.19 with annotations containing accented characters, those characters were escaped, e.g. "\xe"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-04-23:
Is there another place in FreeCAD v0.20.2 where you **can** type in accented characters? 

I don't use accented characters much.  How are you typing the accented characters?  I tried the usual ctrl/shift/u + codepoint, but that didn't work for me.

xraf32, 2023-04-23:
Now that you mentioned it, I just tested also object labels, descriptions, spreadsheet cells, TD annotations and TD editable fields, and all have the same problem, which does not exist in 0.19. So the problem is not specific of TechDraw. It happens in any text input.

I did not know about ctrl/shift/u + code. This can only be usual in English, because it would be very cumbersome to remember codes and do that once at every couple of words for example in German or French. For a language which does have accented characters, people first type the accent key, release it, and then type the key to be accented. And the software must understand that it is a composite input of two keys. What is happening is that FreeCAD 0.20.2 is ignoring the first keystroke, so tilde followed by "a" gives plain "a" instead of "ã". This text box here in github is working as expected. In order to get a standalone "~", I must type the tilde key followed by space. In FreeCAD 0.20.2, tilde followed by space also gives plain space.

WandererFan, 2023-04-24:
ctrl/shift/u + codepoint is handy for odd characters like ° or ⌀, but not every application supports it.

Using a Canadian Bilingual key board setting in Linux Mint, I have no trouble entering accented characters from the keyboard.

[image]

Just in the interest of checking everything, is your OS keyboard map set to something other than US English? If I set my keyboard to US English I can't type accented characters at all.

xraf32, 2023-04-24:
My keyboard is not English and the layout is completely different. It has the cedilla key: "ç". So I guess the OS is set correctly, because all keys work as expected in all software except FreeCAD 0.20.2. I have the 0.20.2 AppImage in a folder and the binary 0.19 installed in another folder, and without changing any setting, just pointing the desktop file to one or the other I consistently toggle the problem. For completeness, I created a file in FreeCAD 0.19 with accented characters which are escaped when I open it in FreeCAD 0.20:

0.19:
[image]

0.20.2:
[image]

The file is attached (renamed with zip extension): 
[accented_chars_FC19.zip](https://github.com/FreeCAD/FreeCAD/files/11305750/accented_chars_FC19.zip)

WandererFan, 2023-04-26:
Patch to read escaped unicode annotation text from v019 files backported to v020 by #9389.

yahbluez, 2023-05-21:
I have a second issue i guess has something to do with this issue.
FreeCAD has lost the capability to handle unicode characters.
I have made some models with freecad a while ago using unicode in the draft WB ShapeString feature:

https://www.printables.com/model/370224-zodiac-astrology-hairsticks

All the zodiac signs are unicode characters.

Opening the expression input field show me the unicode character but drops a error message and forces to delete the unicode character to be able to close the expression input tab:

https://i.imgur.com/trH350F.png

This happens with the vanilla freecad and also with the realthunder fork, so i guess the bug is located in the common base.

https://github.com/realthunder/FreeCAD/issues/806

WandererFan, 2023-05-21:
I wouldn't expect the expression parser to understand this.  How should it evaluate a gamma?

No problem making a shapestring with accented characters here.  
[image]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33171 (Git)
Build type: Unknown
Branch: master
Hash: 029e20ea40f4cdd3c2ca4dfa74bd1abcd9fc505c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/Canada (en_CA)
Installed mods: 
  * dodo 1.0.0
  * BOLTSFC 2022.11.5
  * sheetmetal 0.2.63
  * ExplodedAssembly
  * FreeCAD_Electric
  * freecad.gears 1.0.0
  * Assembly4 0.50.2
  * A2plus 0.4.60n
[/code]

WandererFan, 2023-05-21:
Same result with v0.20.2.

[code]
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git) Snap 517
Build type: Release
Branch: tag: 0.20.2
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * A2plus 0.4.60k
[/code]

Roy-043, 2023-05-22:
This scenario:
https://github.com/realthunder/FreeCAD/issues/806#issuecomment-1556111917
works fine in the version listed below.

There are however fonts that do not have the mentioned unicode characters. I have tested with DejaVuSans.ttf.

Version:
```
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: Dutch/Netherlands (nl_NL)
```

Roy-043, 2023-05-22:
The original problem of typing accented characters does not occur in the Draft WB. If the right keyboard layout is specified in the OS, in Ubuntu I have selected: _English (US, alternative international)_, accented characters can be entered by first pressing the accent key and then the letter key.

Version:
```
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: Dutch/Netherlands (nl_NL)
```

adrianinsaval, 2023-06-06:
Can this be closed or are there still remaining problems?

Roy-043, 2023-06-22:
Unless @xraf32 or @yahbluez provide further information (i.e. a reproducible scenario) this can be closed IMO.

luzpaz, 2023-06-27:
Closing. If reproducible in any way...please re-open. TIA

closed 2023-06-27 by luzpaz