FreeCAD/FreeCAD #9372: [Issue] Raytracing WB is unmaintained and obsolete

chennes, 2023-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77702

### Version

0.21 (Development)

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

The Raytracing workbench uses the no-longer-supported (and therefore no-longer-distributed-by-default) LuxRender library. Keeping an unused WB around adds overhead to the project: Raytracing should be removed so it does not take up anyone's time. The Render WB addon provides a more modern take on the need for a rendering option.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-24:
Superseded by Render WB (CC @howetuft)

PaddleStroke, 2023-04-26:
If it is superseded by Render WB and if raytracing is discontinued, then we need to change the one that ships with FC no?

howetuft, 2023-04-26:
IMHO Render WB should _not_ be bundled with FreeCAD. To me, photorealistic rendering is just a side feature of a CAD software, not core. I would leave all those rendering solutions (Render, Povray-rendering...) as addons and let the user choose whether to activate them or not, especially since the plugin management is now remarkably efficient.
But Raytracing should be removed: not only it is an unnecessary burden for maintenance, but it may also be misleading for users, as it appears to be the standard, recommended solution for rendering, which it should not be.

luzpaz, 2023-04-26:
We can make a deprecated code sub-repo of FreeCAD org and add obsolete code in there?

chennes, 2023-04-26:
Git remembers. We can leave a note in the wiki about what the commit hash was just before it was removed.

closed 2023-05-02 by chennes