FreeCAD/FreeCAD #9377: [Issue] Helix precision

kotvkvante, 2023-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git) Snap 517
Build type: Release
Branch: tag: 0.20.2
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * Defeaturing 1.2.0
  * Assembly4 0.12.7
  * Dracula 0.0.4
  * A2plus 0.4.60k
  * Behave-Dark-Colors 0.1.1
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

There is some unexpected float precision error with Helix in Part Design Workbench. 
1) Pad(50mm) on gear curve.  Create sketch (shifted 50 mm) and pad it. Works well.

2) Helix (50 mm) on gear curve. Create sketch (shifted 50 mm) and pad it. => Two solids. 
expected => one solid 

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kotvkvante, 2023-04-24:
How I can attach file.FCStd example?

luzpaz, 2023-04-24:
@kotvkvante zip it and dragdrop zip file in to GitHub text field.

Also would you be open to testing this issue on development branch? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

kotvkvante, 2023-04-24:
Files for stable verision:
[Shaft-Gear_gh.zip](https://github.com/FreeCAD/FreeCAD/files/11314000/Shaft-Gear_gh.zip)

On dev branch everything works well (both new created file and file from stable version) .
File for dev:
[Shaft-Gear_gh_helix_daily.zip](https://github.com/FreeCAD/FreeCAD/files/11314328/Shaft-Gear_gh_helix_daily.zip)
[image]

luzpaz, 2023-04-24:
@alafr can you please suggest a better title to this bug  ?

grubertechnic, 2023-06-29:
That is indeed weird and confirm the issue, even with a simpler shape and as reference axis certain axis (z in that case), not the InvoluteGear.

```

OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

bgbsww, 2023-11-07:
So, this isn't insufficent precision, it's using estimated values.  The offending call generating the out of precision numbers is                 mkPS.Simulate(2, sim);   in FeatureHelix.cpp.  I have not yet figured out how to fix it; likely by not using the Simulate call, but I don't know how do to that.  Maybe yet.  Any OpenCascade experts out there with a suggested alternative?

luzpaz, 2023-11-07:
@bgbsww ig you click on the line number in the source code on github, and choose permalink; then you can paste the link and it will auto-display the line in question

FlachyJoe, 2023-11-07:
@bgbsww I've tested without `Simulate` but `BRepOffsetAPI_MakePipeShell::MakeSolid` and it give the same lake of precision. Even with `mkPS.SetTolerance(Precision::SquareConfusion());` the result is the same.

The error can came from the path generation:
https://github.com/FreeCAD/FreeCAD/blob/e235a1b795567a99d67e1887a1c7151e60ebe395/src/Mod/PartDesign/App/FeatureHelix.cpp#L397

bgbsww, 2023-11-07:
TL;DR  Assign me please!

Thanks @luzpaz.  In any case, I've replaced the simulate with FirstShape() and LastShape() and determined that the underlying OpenCascade model still shows the same symptom, as @FlachyJoe notes.

I've been down the path of the tolerance setting, including using all three parameters `mkPS.SetTolerance(Precision::SquareConfusion(),Precision::SquareConfusion(),Precision::Angular());`   Changing did not affect what mkPS came up with.  I also looked through generateHelixPath and checked its output, which IS within the precision goals, it was off by about 1e-14 in my test.

However, I think I've found the problem, which is:
 
https://github.com/FreeCAD/FreeCAD/blob/e235a1b795567a99d67e1887a1c7151e60ebe395/src/Mod/PartDesign/App/FeatureHelix.cpp#L244-L246

And the patch #11312 is to remove this case.

FlachyJoe, 2023-11-08:
@bgbsww couldn't we use [BRepOffsetAPI_MakePipeShell::MakeSolid](https://dev.opencascade.org/doc/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html#a35b96c256a7b1f1f04fc94335ceb7f90) instead of all the BRepBuilderAPI_MakeSolid stuff ?
Plus: I don't understand why `wires` are copied in `wiresections` I think we can remove one loop level.

bgbsww, 2023-11-08:
@FlachyJoe  I tried that call!  It threw an exception, so I assumed that's why the code takes the approach it does.  I agree with your hypothesis on the `wiresections`, but I considered it out of scope for this bug.  Worth another look if it doesn't confuse reviewers, since I'm already slipping in a little code cleaning.

bgbsww, 2023-11-08:
Thanks @FlachyJoe , added a version still without the failing makeSolid() call to the patch.

FlachyJoe, 2023-11-08:
@bgbsww Exception is due to lake of `mkPS.Build();`

bgbsww, 2023-11-08:
@FlachyJoe Nope, exception occurs either way.  I tried that.

bgbsww, 2023-11-08:
@FlachyJoe you were right, thanks for your patience.  I missed that the exception switched from on the makeSolid to an extraneous shell add later on.  Patch amended to greatly simplify the code

bgbsww, 2023-12-18:
This should be reopened.  Repairs to #11312  are in process in #11598 as it is wrong and breaks other things. 

The underlying issue is not the precision of the helix, but that the fuse in the featurePad of the DatumPlane creates more than one solid even though there is only one.  However, when I instrument this by adding the prism or the base to the model, before running the mkFuse the problem goes away and the fuse creates a single solid.

It comes down to the TopExp_Explorer in countSolids returning both
`14:20:19  "className": "TopoDS_Shape""TShape": {"className": "TopoDS_TShape", "this": "0xx55ece2433ea0", "ShapeType": 2, "NbChildren": 1, "Flags": 2, "Free": 0, "Free": 0, "Locked": 0, "Modified": 1, "Checked": 0, "Orientable": 0, "Closed": 0, "Infinite": 0, "Convex": 0}"Location": {"className": "TopLoc_Location", "Transformation": {"Location": [0, 0, 0], "Matrix": [1, 0, 0, 0, 1, 0, 0, 0, 1], "shape": 0, "scale": 1}, "IsIdentity": 1}"Orient": 0`
and
`14:20:19  "className": "TopoDS_Shape""TShape": {"className": "TopoDS_TShape", "this": "0xx55ece1aa17e0", "ShapeType": 2, "NbChildren": 1, "Flags": 2, "Free": 0, "Free": 0, "Locked": 0, "Modified": 1, "Checked": 0, "Orientable": 0, "Closed": 0, "Infinite": 0, "Convex": 0}"Location": {"className": "TopLoc_Location", "Transformation": {"Location": [0, 0, 0], "Matrix": [1, 0, 0, 0, 1, 0, 0, 0, 1], "shape": 7, "scale": 1}, "IsIdentity": 0}"Orient": 0`

This is readily fixed by SetFuzzyValue of that fuse to 1e05 instead of 1e07, but isn't clear yet if that is a reasonable thing to do.  

I'm actively digging to identify how to preprocess or postprocess the fuse to fix this bug.

bgbsww, 2024-05-16:
This issue appears to go away with TNP enabled, so there isn't any point in patching the older version.

If it does reappear after TNP is on, then changing the end of buildExtrusion in PartDesign/App/FeatureExtrude.cpp to have something like `prism.makeElementFuse(prism.getSubTopoShapes(TopAbs_SOLID),Part::OpCodes::Fuse,Precision::Confusion()/100);`
 instead of`prism.makeElementFuse(prism.getSubTopoShapes(TopAbs_SOLID));` should solve the problem.

Precision::Confusion()/100 should be placed into a hidden property per the comments on #11776.

maxwxyz, 2024-08-07:
@kotvkvante @FlachyJoe is this issue still reproducible with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
as it should be fixed with the TNP fixes.

MiqCG, 2024-08-26:
Hi,

Not tested with the very latest revision but looks to me that some problems are still present in revision 38459. 
No errors on recomputing the kotvkvante daily file but it doesn’t show the whole body with or without Part Design preference "Multiple solids" enabled.

I have used a simpler newly created file with and without the Datum plane and with Preferences > Part Design > Multiple solids disabled
... and seems to me that the imprecision is still present (don’t know what is the expected precision). The output of the python console is with a triangle as a base sketch and 50 mm Additive Helix.

doc = FreeCAD.activeDocument()
shp = doc.getObject("AdditiveHelix").Shape
[v.Point.z for v in shp.Vertexes]

[0.0, -10.0, -1.0658141036401503e-14, -1.0658141036401503e-14, -3.552713678800501e-15, 49.999994827677305, 49.99999464556699, 49.999994727155716]

See attached files.

On the other hand, in some occasions, when invoking the AdditiveHelix tool, FreeCAD gets busy for dozens of seconds (even with a triangular base sketch).
And don't know if they are relevant to this issue but in some tests, these errors appear in the Report view: 
18:22:58  <Part> ViewProviderExt.cpp(1299): Cannot compute Inventor representation for the shape of Unnamed#AdditiveHelix: Bnd_Box is void
18:22:58  <Part> ViewProviderExt.cpp(1299): Cannot compute Inventor representation for the shape of Unnamed#Body: Bnd_Box is void
18:24:33  <Part> ViewProviderExt.cpp(1299): Cannot compute Inventor representation for the shape of Unnamed#Body: Bnd_Box is void
18:24:33  <Part> ViewProviderExt.cpp(1299): Cannot compute Inventor representation for the shape of Unnamed#Body: Bnd_Box is void

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.21
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.24

[AdditiveHelix_rev38459.zip](https://github.com/user-attachments/files/16751281/AdditiveHelix_rev38459.zip)

maxwxyz, 2024-08-26:
please test on the latest version for the most recent fixes.

maxwxyz, 2024-08-27:
@FlachyJoe @bgbsww should this work now?

MiqCG, 2024-08-27:
> please test on the latest version for the most recent fixes.

Revision 38459 is just 14 days old but anyway I have tested the attached files as you requested with the last appimage 38553 and newly deleted .local .config .cache FreeCAD folders.

Opening the file "Helix_Datum_rev38459" >
1. gives a first error: Reading failed from embedded file: StringHasher.Table.txt (469 bytes, 174 bytes compressed)
2. It is visualized as one body (Part Design Multiple bodies option disabled)
3. executing the above Python code the answer is:
[0.0, -10.0, -1.1e-14, -1.1e-14, -4e-15, 49.999994827677305, 49.99999464556699, 49.999994727155716]

Opening the file "Helix_rev38459" >
1. Same error: Reading failed from embedded file: StringHasher.Table.txt (482 bytes, 175 bytes compressed)
2. ídem
3. Python answer is:
[0.0, -10.0, 4e-15, 7e-15, 4e-15, 49.99999486189137, 49.999994883608416, 49.999994559412066]

Making the model from scratch with this revision, the same occurs when the Python code is executed:
[0.0, -10.0, -7.105427357601002e-15, -7.105427357601002e-15, -7.105427357601002e-15, 49.99999482934381, 49.99999464729219, 49.99999472885464]

On the other hand, resource consumption issues remain too: when AdditeveHelix is clicked (with triangular base sketch) FreeCAD goes from eating 1% CPU to 70-98% for 30 seconds aprox, and it spends 20 more seconds to recover the 1% consumption.

[image]

[image]

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)

MiqCG, 2024-08-27:
Maybe different default values on the Tasks panel of Additve Helix can make a difference and be an easy patch.

I suggest:
- Axis: Normal sketch axis
- Mode: Height-Turns-Angle
- Height: 10,00 mm (same as Pad for coherence)
- Turns: 1,00 (or less)
- Cone angle: 0,00º (same value)

Default values:

[image]

bgbsww, 2024-08-27:
Reading failed from embedded file: StringHasher.Table.txt (482 bytes, 175
bytes compressed)

That implies that the files being read are broken and need recalculation
and resaving - that error appears when the StringHasher table was saved
with a broken version.
I don't know that it will have any effect on the helix question.

Message ID: ***@***.***>
>

AIRCAP, 2024-10-18:
I just realized that #17119 is basically a modern reincarnation of #11776 just with a much wider scope and more adaptiveness.

I wonder though why Helical shapes in particular are so susceptible  to this issue. Could it be that the helical path that is generated and used to generate the helix already has bad precision - which in turn causes the solid piped around it to be "bad" ?

if so, a possible solution (short of coming up with a more accurate 1d shape formulation for helical paths as a bspline) could be to set the OCCT tolerances of the helical path to something more accurate (accurate tolerance meaning tolerance as high/bad as the helical path actually is bad - maybe as bad as size*1e-3 or size*1e.4)  - that would then increase the vertex, edge and surface tolerances on the solid created and in turn have it "snap" to its neighbours with any boolean operation attempted.

of course there still will be issues if these parts are later exported and the shape is simply imprecise.

then again how much imprecision are we talking - the values reported above seems to be off by a factor of less than 10^-6 that should be acceptable for most applications

AIRCAP, 2024-10-18:
note: I tried higher tolerances on the helix path - that doesn't really fix anything, i don't think this is it

FlachyJoe, 2024-10-18:
@AIRCAP Have you try to increase the tolerance of the last face?

AIRCAP, 2024-10-18:
Not yet. Which face is the "last" ? the uppermost one?

FlachyJoe, 2024-10-18:
The face to merge with the cylinder .

AIRCAP, 2024-10-19:
tried it. set tolerances as high as 1.0mm on that face - doesn't fix the fusion (without fuzzyness) the result is self intersections on the edges touching the cylinder with the face of the cylinder

AIRCAP, 2024-10-19:
so just setting the tolerance of the surface parallel face higher alone doesn't fix anything. It might help if the tolerance of all surfaces, edges, vertices and faces that touch in any way is set higher, but in my test cases that would be all faces of the helix. here is the example. test with the branch in #17214 - it has a feature that can override part tolerances, which is used in this FCStd

(also with that, Part_CheckGeometry reports tolerances, which is useful for debugging this issue)

[Helixes_of_different_sizes.zip](https://github.com/user-attachments/files/17444766/Helixes_of_different_sizes.zip)

In these test cases I have to set a tolerance of exactle 1e-5 
if I increase to 5e-5 or higher then the helix becomes self intersecting because the "coils" touch after the first revolution
if I decrease to 1e-6 or lower then the boolen fusion with the cylinder produces self intersecting edges.

these are fusions without fuzzyness, so only using part tolrances

AIRCAP, 2024-10-19:
I tried to add a helix with 1km to see how scaling affects this, but PartDesign->AdditiveHelix cannot make a helix that large. It completely fails to generate that part.

AIRCAP, 2024-10-19:
[Helixes_of_different_sizes2.zip](https://github.com/user-attachments/files/17445149/Helixes_of_different_sizes2.zip)

AIRCAP, 2024-10-19:
here is a reproduction of the standalone issue - you can't make a helix with 1km radius
[Helix_does_not_work.zip](https://github.com/user-attachments/files/17445216/Helix_does_not_work.zip)

FlachyJoe, 2024-10-19:
@AIRCAP 
https://github.com/FreeCAD/FreeCAD/blob/ef2156d0f6d116b1642a5531a2d44fc6b7d17042/src/Mod/Part/App/TopoShape.cpp#L2338

should be

```cpp
    Standard_Real precision = (radiusbottom + radiustop + height) / 3 * Precision::Confusion();
    BRepLib::BuildCurves3d(wire, precision, GeomAbs_Shape::GeomAbs_C1, 14, 10000);
```

AIRCAP, 2024-10-20:
@FlachyJoe 
I changed that line earlier in tests - it didn't fix the boolean issues

> note: I tried higher tolerances on the helix path - that doesn't really fix anything, i don't think this is it

but it does fix this issue :-) it also fixes  #17113 - so that change is definitely needed!!!

AIRCAP, 2024-10-20:
I think better would be

`
double size=sqrt((radiusbottom+radiustop)*(radiusbottom+radiustop) + height*height);
    BRepLib::BuildCurves3d(wire, Precision::Confusion() * size, GeomAbs_Shape::GeomAbs_C1, 14, 10000);
`
that's sufficient to fix the crash, but not to get a valid helical shape in all cases. I also had to afterwards use ShapeFix_ShapeTolerance::limitTolerance() on it  - however I think that is better done within FeatureHelix.cpp - and it needs to be done again after the pipe, assuming that the sketch is tolerance 1e-7 - otherwise the subsequent boolean will fail or trigger weird OCC bugs again - this doesn't just need to compensate for the size but also for the helix imprecision itself

AIRCAP, 2024-10-20:
im heading to sleep, but  I can prepare a pull request tomorrow  -- will depend on both #17119 and #17349 though

FlachyJoe, 2024-10-20:
> why the division by 3 through?

To take the average.

> I think better would be

No matter the exact value only order of magnitude is important, no?

> get a valid helical shape in all cases

It can't be. If you sweep a 1mm circle along a 1km helix path it will break somewhere 🤪

AIRCAP, 2024-10-20:
I have just been playing with #11083

there seems to be shapes where OCCT behaviour is very very unstable - bugs can appear and disappear chaotically based on very small changes to tolerance, not just order of magnitude changes.

this is so bad, that - it seems when tolerance is calculated based on bounding box, the tiny changes in bounding box size depending on the tessellation used, can trigger the shape to break or succeed (because OCCT bug happens or doesn't)

#11083 is such a shape. the issue is that the rotated sketch inner boundary is pretty much exactly the same diameter as the cylinder the subtractive helix cuts from. but due to the "wonky" shape of the bspline, that cut line goes in and out of the cylinder, sometimes intersecting a little bit, sometimes leaving a little bit of material behind -- that causes nasty self-intersecting vertices and lines, which even large tolerance values cannot completely fix (meaning the resulting part fails BOP checks, but is at least not completely broken)

but the shape is prone to OCCT failure - although i am not even 100% sure if its a shape failure or just a tessellation failure/bug at this point.

since the new fuzzy tolerances are calculated based on boundingbox, technically all booleans could be affected by this, however I think the fix could be a "static tessellation" in data parts - and a view provider tessellation that has no effect on the TopoShape boundingbox... - at least that would remove the root cause.

do you know a way to get a TopoShape's dimensions without using boundingbox?

AIRCAP, 2024-10-20:
I'm somewhat relieved. I have this error:
[image]

but a subshapebinder on the same shape with default tessellation doesn't show this. so this seems to be a tessellation / display bug, not a shape bug

phew!

[image]

FlachyJoe, 2024-10-20:
> do you know a way to get a TopoShape's dimensions without using boundingbox?

What about taking the smallest edge as reference?

AIRCAP, 2024-10-20:
if, you'd need to take the largest edge as reference, that puts the lower limit on the tolerance
in theory the smallest edge puts an upper limit on the tolerance
if the lower limit gets larger than the upper limit, the part exceeds OCCT capabilities and you can't CAD it
(you can't have a 1mm profile on a 100km cylinder)

AIRCAP, 2024-10-20:
how do i determine the largest edge though?

AIRCAP, 2024-10-20:
@FlachyJoe 
if 
`BRepLib::BuildCurves3d(wire, Precision::Confusion() * size, GeomAbs_Shape::GeomAbs_C1, 14, 10000);`
is calculated in TopoShape.cpp with various formulas for size, the generated helixes fail volume checks. that decreases the accuracy of the generated helix further, so not a good idea. we need to fix that later when making the pipe only. looking into it

AIRCAP, 2024-10-20:
ok found a solution. the trick is to set the accuracy of the wire very very high, but size dependent. then it avoids the crash with very large helixes, but passes all tests with regular sized ones

PR #11312 (2023-11-30): Correct #9377 issue with accuracy in AdditiveHelix
After much instrumenting and trial and error, it appears we set the wrong mode when the angle is zero.  Not clear if there was some past version of the library that needed the special case removed here.  Also cleaned up a little unneeded code.

This should fix #9377
commit 94ca51b7 (2023-11-30): PartDesign: Fix #9377 - issue with accuracy in AdditiveHelix (#11312)

* Correct precision issue in PartDesign::AdditiveHelix

* Simplify getting the end wires

* Remove extraneous loop per @FlachyJoe

* Switch to using mkPS.Shape per @FlachyJoe

* Incorporate guarded efficiency improvement
PR #17356 (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry 
fix #17113
fix #9377
fix #11083

Helical shapes in PartDesign (and to a lesser degree helical paths) suffer from a precision flaw (see #9337) -- cylinders, cones, spheres, boxes, etc... are all "perfect" shapes in OCCT - ruled by exact mathematical formulas where their edges and surfaces are.

But OCCT has no primitive shape for helical shapes - be it 2d and 3d - so helical paths are approximated with bsplines and helicap solids are pipes along these bsplines.

now the issue is this approximation makes errors far in excess  [... cut]
commit 7680872e (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry  (#17356)

* increase tolerance on helical shapes to avoid crash

fix #17113
fix #9377
fix #11083

* added test cases for additive and subtractive helix

* fixed whitespace issue, reduced turncount for speed

* relaxed large scale testcases for OCCT 7.3 and below, work around windows CI error #17394

* Further reduce max size in test for buggy Ubuntu OCCT
closed 2024-11-11 by yorikvanhavre