FreeCAD/FreeCAD #9378: [Issue]

rmdug, 2023-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Revision 32382 on Linux Manjaro
```

### Subproject(s) affected?

PartDesign

### Issue description

In partdesign Workbench, first polar pattern on a pocket works well.
On the second one, on a second pocket, Freecad crashes.

After some tries : if the pocket is a simple rectangle then the pattern will work.
If I add replace an edge with a half circle in order to draw a machinable slot then Freecad crashes at update as I leave the sketch.

[image]

FreeCad will also crash while trying a mirror pattern.

I eventually managed to perform the pattern by, step by step, leaving the sketch after each minor change and as a precaution record the file. Then enter the sketch to go further until it is as wanted. This way FreeCad don’t crash : 

[image]

Thanks for your attention and thank you for this piece of software !

Rémi

### Anything else?

[2023-04-12-01.zip](https://github.com/FreeCAD/FreeCAD/files/11305049/2023-04-12-01.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-24:
please provide full version information (help->about->copy to clipboard), it is very important

rmdug, 2023-04-25:
Hello Adrianinsaval,

Here is the full information :
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32832 (Git)
Build type: None
Branch: master
Hash: 3432d8796c57ab51a1070c14a2a12069a31b9304
Python 3.10.10, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: French/Belgium (fr_BE)
[/code]

I will follow your instructions and come back.

Thank you for your help !

Remi

rmdug, 2023-04-25:
I just proceeded as you proposed and ran this appimage :
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-32985-2023-04-23-conda-Linux-x86_64-py310.AppImage

Which gives the following informations : 
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git) AppImage
[2023-04-12-01.zip](https://github.com/FreeCAD/FreeCAD/files/11326830/2023-04-12-01.zip)

Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/Belgium (fr_BE)
[/code]

And again I get the issue when trying to perform a circular pattern on a simple M4 threaded hole :

[image]

To reproduce : based on the here attached file :
[2023-04-12-01.zip](https://github.com/FreeCAD/FreeCAD/files/11326830/2023-04-12-01.zip)

In Part Design workbench, activate body « body » ;
Select Hole001 ;
[image]

Select the command « Partdesign_polarpattern
[image]

[image]

Change the number of repetitions to 3, then press OK to validate the command.
On press on « OK » button Freecad crashed after showing « segmentation fault » in the report view.

I tried a different approach by validating the pattern without changing the number of repetitions, leaving all options to default (2 repetitions). Freecad did not crash :

[image]

Then editing the pattern to update it to 3 holes lead the first time to a crash, but the second time Freecad didn’t crash :

[image]

[image]

At last I managed to get my patterns done. Still it seems to me a strange behavior…

Any suggestion is welcome !

Regards,

Remi

adrianinsaval, 2023-04-28:
No crash for me but I did get all these errors when opening the file:
```

19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Svg Hatch File: /usr/share/Mod/TechDraw/Patterns/simple.svg is not readable
19:54:10  Pat Hatch File: /usr/share/Mod/TechDraw/PAT/FCPAT.pat is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  Line Group File: /usr/share/Mod/TechDraw/LineGroup/LineGroup.csv is not readable
19:54:10  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
19:54:10  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.

```

adrianinsaval, 2023-04-28:
do you have a freecad package installed on your system? Can you try uninstalling it?

adrianinsaval, 2023-05-24:
are you still experiencing this problem? if no response I'm inclined to close this issue as it's not actionable

grubertechnic, 2023-06-29:
@adrianinsaval : works also flawless here (without any further editing)

Just two notes, why maybe some issues.

 - The first Sketch is bad constraint, also some unnecessary stuff in it, can lead to problems.

 - The views in the drawing are very little distorted and not complete straight. This issue is already solved by @WandererFan . This can be also harmful, since the sections are boolean operations.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

adrianinsaval, 2023-06-29:
thanks for testing, since there hasn't been response from OP in months and others can't reproduce I'm closing this.

closed 2023-06-29 by adrianinsaval