FreeCAD/FreeCAD #9382: [Issue] Crash when trying to move object from part to another body

nentibusarchitectura, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 1397818c9664d158281339a84be59d3e9887f6ad
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Manipulator 1.4.9
[/code]

Also crash happens in nightly:

[code]
OS: Ubuntu 20.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.
Build type: Release
Branch: unknown
Hash: 1282fdb7e8a721e9b411358d68daf8921b0bf50a
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

None

### Issue description

In the attached diagram I made a mistake, rather than creating a new part with multiple bodies, I created a body, and inside of it created a part, added a shape binder and created a pad using the shapebinder.

Tried to create another body to move the shape binder and the part to when I select the objects "MiniPC_sketch_ref" and "Pad" inside the "Mini_PC_Physical" right click and select "Move object to another body" FreeCAD segfaults with:

[see attached Segfault.zip file]

### Anything else?

[Mini_PC_Holder.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11320908/Mini_PC_Holder.FCStd.zip)
[Segfault.zip](https://github.com/FreeCAD/FreeCAD/files/11320954/Segfault.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-04-25:
Confirmed using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32990 (Git)
Build type: Release
Branch: master
Hash: 709a2cf23dff62c36b8cfc9a3576671dfa6f3d9d
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 1.0.0
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * Curves 0.6.8
  * toSketch 1.0.1
  * Plot 2022.4.17
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

The 'nightly' build quoted by the OP is actually from October 2022.

xtemp09, 2023-04-25:
I've found what causes the crash. Simply put, `isFeatureMovable` is invoked with `nullptr`. The backtrace: 

```c++
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
  * frame #0: 0x00007fff373187c3 PartDesignGui.so`PartDesignGui::isFeatureMovable(feat=0x0000000000000000) at Utils.cpp:441:15
    frame #1: 0x00007fff37318901 PartDesignGui.so`PartDesignGui::isFeatureMovable(feat=0x0000555555b2d2a0) at Utils.cpp:452:14
```

@usakhelo introduced the function `bool isFeatureMovable(App::DocumentObject* const feat)` in [the commit](https://github.com/FreeCAD/FreeCAD/commit/f49ddaea453fffcc5769961cd989a0dcee0bb746).

Should I make PR to fix it? It would be better if the author of the original commit will take a look at it, methinks.

adrianinsaval, 2023-04-25:
If you already have an idea how to fix it please go ahead and make a PR, indeed it would be best if the original author fix it themselves but we can't count on that, specially on something from 2016

xtemp09, 2023-04-25:
usakhelo still has recent activity on github. I think it would be better to postpone a little bit.

usakhelo, 2023-04-25:
> usakhelo still has recent activity on github. I think it would be better to postpone a little bit.

Hey @xtemp09, please don't hesitate to submit your PR. I'm not planning to contribute to FreeCAD at the moment.

PR #9398 (2023-05-04): Fixed segmentation fault in PartDesign workbench
Closes #9382. [usakhelo](https://github.com/usakhelo), the original author of the function `isFeatureMovable`, politely [refused](https://github.com/FreeCAD/FreeCAD/issues/9382#issuecomment-1522224360) to contribute to FreeCAD.

The cause of segmentation fault was attempted access to nonexistent `App::DocumentObject* const feat` (it was `nullptr`).

I concluded that if there is no object, so it cannot be moved. I can be wrong.

---

<details><summary>Standard form</summary>

Thank you for creating a pull request to contribute to FreeCAD! Place a [... cut]
closed 2023-05-04 by adrianinsaval