FreeCAD/FreeCAD #9386: [Issue] Crash when using box selection or box element selection

GeeBee2, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 11 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Select the Edit > Box Selection or Box Element Selection option from the menu.

Drag a rectangle. Nothing happens. If you try again it will crash.

Once the Tool is selected, you can't cancel it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GeeBee2, 2023-04-25:
How can I do that?

Syres916, 2023-04-25:
Using the exact same build as the OP, I cannot replicate it on Windows 7.

```
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60f
  * Alternate_OpenSCAD 1.0.0
  * Assembly3 0.11.4
  * Assembly4 0.12.6
  * CfdOF 1.20.4
  * Curves 0.6.6
  * fasteners 0.4.54
  * fcgear 1.0.0
  * freecad.xray 2022.4.17
  * FreeCAD_Turning_Addon 0.1.0
  * Help 1.0.3
  * Manipulator 1.5.0
  * PitchedRoof
  * Plot 2022.4.17
  * Render 2023.2.2
  * sheetmetal 0.2.63
  * Silk 1.0.0
  * timber
```

GeeBee2, 2023-04-26:
The crash only occurs with an empty file and it needs sometimes more than one click to crash.

Syres916, 2023-04-26:
For me, Box Selection, Box Element Selection and Select All shouldn't be active in the menu unless there's a shape in the 3D view. Whether that's classed as not fixing the root cause of the crash is another story. It doesn't look like a massive coding job but I know it involves C++ macros which is frowned upon here.

luzpaz, 2023-05-25:
Ping @GeeBee2

beaufrazer, 2023-06-08:
I just downloaded FreeCAD and have this issue. Mine presented when I was following along in the tutorials. Win 11 64 bit. When I try and use the box selection the application hangs.
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/11684771/FreeCAD.log)

beaufrazer, 2023-06-11:
Installed 0.21 and issue seems to not be present

luzpaz, 2023-06-11:
Do you mind adding the full [About](https://wiki.freecad.org/About) info for the version you couldn't reproduce it in ?

closed 2023-06-11 by luzpaz

impeeza, 2024-07-22:
Sorry to comment on a closed Issue, My PC is 

OS Name:                   Microsoft Windows 11 Enterprise
OS Version:                10.0.22621 N/A Build 22621
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free
System Type:               x64-based PC
Processor(s):              1 Processor(s) Installed.
                           [01]: Intel64 Family 6 Model 94 Stepping 3 GenuineIntel ~2701 Mhz
BIOS Version:              American Megatrends Inc. 4210, 5/28/2019
System Locale:             en-us;English (United States)
Input Locale:              en-us;English (United States)
Total Physical Memory:     16,254 MB
Available Physical Memory: 8,215 MB
Virtual Memory: Max Size:  18,430 MB
Virtual Memory: Available: 7,461 MB
Virtual Memory: In Use:    10,969 MB

My GPU is: Advanced Micro Devices, Inc; Radeon RX 580 Series; driver version: 31.0.21912.14

I found the issue following the (https://wiki.freecad.org/Creating_a_simple_part_with_Draft_and_Part_WB) tutorial.  Enabling software OpenGL allows me to continue, but looking on the issues they are closed without solution.

another thing I noted, Changing to Draft Workbench take lot of time before

There is any planed real fix to the issue?

if worth anything, this is my log file, but I see no errors or information on it.
[FreeCAD.log](https://github.com/user-attachments/files/16327455/FreeCAD.log)