FreeCAD/FreeCAD #9388: [Issue] Can't delete Spreadsheet link

YosypC, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
Installed mods: 
  * BOLTSFC
  * ExtMan
  * fasteners
  * fcgear
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

I'm trying to solve [this](https://forum.freecad.org/viewtopic.php?p=678127#p678127) problem by creating an empty link and linking it to another file's spreadsheet. Whenever I click on it, the file's tree and spreadsheet open automatically, not allowing me to do anything with the link in the tree. Pressing delete doesn't delete it.

Its deletion is only possible by moving it to the main file, and then attempting deletion.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-11:
can you reproduce this on 0.21? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

please provide example files or more clear reproduction steps

YosypC, 2023-05-15:
> Its deletion is only possible by moving it to the main file, and then attempting deletion.

Turns out I can right click it and delete it from there. Duh.

> can you reproduce this on 0.21?

https://github.com/FreeCAD/FreeCAD/assets/50027752/dc014fa0-b022-4b6a-af64-7c3fc7699bf5

You can see me clicking successfully the file once the Spreadsheet is closed from the opened documents, finally deleting it with the keyboard command. Clicking on its link while open takes you to the called document.

> please provide example files or more clear reproduction steps

Done in .20.2, pretty much the same as before:

https://github.com/FreeCAD/FreeCAD/assets/50027752/89bc0522-6e4a-4988-825a-0cef500017ec

luzpaz, 2023-05-16:
Can you please add the [About](https://wiki.freecad.org/About) info for v0.21.xxxxx you tested on. TIA

YosypC, 2023-05-16:
@luzpaz 

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

maxwxyz, 2024-06-29:
is this still an issue? I cannot reproduce it in the latest version

YosypC, 2024-07-13:
I can't test it indefinitely