FreeCAD/FreeCAD #9390: [Issue] Maximize button on Addon Manager does not work.

blacklightpy, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/default)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * offline-documentation 1.0.0-alpha
[/code]
```

### Subproject(s) affected?

None

### Issue description

Maximize button on addon manager does not work.

### Anything else?

I was trying to read the offline documentation from the addon manager. Then I figured out it couldn't be maximized.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-04-25:
Strange, I'll give it a look. What distro are you using?

blacklightpy, 2023-04-25:
openSUSE Tumbleweed

chennes, 2023-04-25:
To clarify -- is it the Addon Manager window that is failing to maximize, or the Offline Documentation addon's window?

blacklightpy, 2023-04-25:
It is the Addon Manager. The Offline Documentation does not have a seperate window, it is a frame within the window.

chennes, 2024-01-30:
I can't replicate this one with either the Flatpak or a self-compiled -- if it recurs for you please re-open and post your full OS details.

closed 2024-01-30 by chennes