FreeCAD/FreeCAD #9391: [Problem] PartDesign datum planes are too opaque by default and significantly obstruct visibility of objects

adrianinsaval, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33001 (Git)
Build type: None
Branch: master
Hash: 252c628f9024eff2752d9dacc47edfd99e2c14c1
Python 3.10.10, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
  * fasteners 0.4.56
  * Assembly3 0.11.4
  * CfdOF 1.21.7
  * Plot 2022.4.17
  * Curves 0.6.8
  * Defeaturing 1.2.0
  * DynamicData 2.46.0
  * Dracula 0.0.4
  * 3DfindIT 1.2.0
  * SelectorToolbar
  * FC_SU
  * A2plus 0.4.60k
  * Assembly4 0.50.1
  * CurvedShapes 1.0.4
  * sheetmetal 0.2.63
  * AirPlaneDesign
  * Glass
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

By default PartDesign datum objects have a transparency of 60% and can quickly clutter the view when several are visible at the same time.
[image]

Suggest increasing default transparency to 95%, still visible but not so obstructive anymore:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2023-05-05:
I feel like so much transparency make it very difficult to catch the relative location of planes vs. objects. :confused:

adrianinsaval, 2023-05-05:
can you check changing a datum plane's transparency to different values and tell me what's the highest transparency value where you still feel like you can understand relative positions? I'm really unhappy with the current 60% value, it obstructs too much and is specially annoying when you start a sketch on a plane and then have to go and hide the planes before being able to see the part you are working on. For visualizing the position of the plane the outline fells like enough to me.
Here's how it looks on #9393 (not the same as the picture I uploaded here on the issue):
[image]

0penBrain, 2023-05-05:
> can you check changing a datum plane's transparency to different values and tell me what's the highest transparency value where you still feel like you can understand relative positions? 

Did a very quick test with 2~3 datum planes. To me max transparency would be 75%. Over this I'm confused about locations.

> I'm really unhappy with the current 60% value, it obstructs too much and is specially annoying when you start a sketch on a plane and then have to go and hide the planes before being able to see the [...]

That's another rationale. Maybe the solution here is to enhance the visibility automation so datum objects are automatically hidden when Sketcher enters edit mode (and restored when leaving). ;)

adrianinsaval, 2023-05-05:
Another related problem is that we are using yellow as default color, this color is usually associated with warnings and it draws too much attention to itself, so it's even more distracting. I think we would benefit from using a cold color instead. Here a datum plane in master branch with 85% transparency and using color #00aa7f
[image]

0penBrain, 2023-05-05:
Still too much transparency for me (just my opinion). About colors and so on, I already proved I'm so bad at it that I won't comment. :)

JohnOCFII, 2023-05-05:
I think 80-85% transparency looks fine.  IMO, too many preference settings makes long-term maintenance more difficult.

luzpaz, 2023-11-11:
What is the verdict here ?

maxwxyz, 2024-03-10:
related: #12584

closed 2024-12-06 by chennes