FreeCAD/FreeCAD #9392: [Issue] Cannot move/drag objects (Part) to a folder or a Sketch to a Body

leoheck, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77898

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git) AppImage
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * Assembly4 0.50.2-dev
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

After the latest weekly build, I am not being able to drag objects in the Objects tree. I am trying to move a Part inside a Part folder. And also, tried to move a Sketch to inside a Body. When I grab the objects the pointer does not change to the one that indicates where I can drop the Object (it was a plus sign or something)

### Anything else?

Maybe the workflow just changed, but right now I cannot move objects as I used to do.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rbuehlma, 2023-04-26:
I'm having the same issue. Also after Upgrading to Ubuntu 23.04 and also updating the AppImage the same time.

leoheck, 2023-04-27:
Interesting. I am not sure if it is related but are you using X11 or
Wayland? I am using X11 but I have switched for a moment to check the
usability, and then reverted back to X11. But I didn't try to switch back
to Wayland to check if it improves the behavior of this issue. This is
actually an interesting test. I have to test it once I reach the computer.

leoheck, 2023-04-28:
Wayland/X11 does not make a difference at all. Choosing NVIDIA or Nouveau neither.

It is a bug in Freecad. 

No matter what I drag to any place, Freecad does not allow me to move the object. The cursor does not change either showing me that it is possible. this was working in the previous version of Freecad weekly.

I am attaching a video showing what is going on here. 
https://user-images.githubusercontent.com/1277920/235052157-637c4ca2-d1ac-48b4-a7da-f20f625a4c5b.mp4

rbuehlma, 2023-04-28:
I did download the latest stable version from here:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20.2

There the problem occurs as well. Therefore, it seems to be an issue with Ubuntu 23.04

leoheck, 2023-04-28:
Oh, thanks. I have to replicate your test too. 
I have another idea. Did you try to remove or reset the Freecad configs?
O have to try that too.

rbuehlma, 2023-04-28:
Other software which most probably has the same issue:

https://github.com/OpenShot/openshot-qt/issues/5221

May be its an issue in Qt?

rbuehlma, 2023-04-29:
If I install FreeCAD 0.20.2 through apt instead of AppImage, drag & drop works as expected. I though exactly such distro dependent issues should have been solved by using AppImage...

leoheck, 2023-04-29:
Awesome test. Indeed, those things should not be issues as AppImages lovers keep saying. 

@rbuehlma are you looking for a workaround to get this working or just reporting the issue to get this fixed asap, as myself?

If someone is looking for a workaround, files can be moved easily by:
- Selecting the destiny object, like for example the Group/Folder
- Then, in the properties view, there is a Group attribute, click the value column and the ellipsis. 
- And then select the object which you want to move in the Popup or directly in the Objects tree.

leoheck, 2023-04-29:
I tested with a fresh new install of Ubuntu 23.04, nothing has changed.
- AppImages still have the issue
- The one from the apt, does not.

rbuehlma, 2023-04-30:
I'm not looking deeper into this issue. I just try to provide hopefully useful information to help debugging the case. Using the apt version is a good workaround for me.

wandrewkeech, 2023-05-03:
https://gitlab.gnome.org/GNOME/mutter/-/issues/2715

adrianinsaval, 2023-05-04:
@leoheck can you confirm is this is the gnome issue mentioned above?

adrianinsaval, 2023-05-04:
Wandrewkeech mentioned above a bug reported on gnome, I think you are using gnome so that is likely the cause. Try a different window manager/desktop environment if you can or see if there's a workaround in that link

leoheck, 2023-05-04:
Ah, I see. Thanks, I cannot use another window manager. I use Ubuntu just to use Gnome. 

They are reporting a workaround, but it is for Wayland. 
[image]

I cannot use Wayland too, my mouse clicks get unresponsible with it.

But I already mentioned a working workaround, which is ok for me while this is not fixed:
[image]

adrianinsaval, 2023-05-04:
I looked more closely to the issue on gnome, it seems they fixed it already, there's also a linked downstream Ubuntu bug report that says they commited a fix, but I don't know if that has reached the repos yet, can you check if there is an update for mutter available with apt and if it fixed the problem?

leoheck, 2023-05-04:
Ah, that's awesome. Thank you. It may take some time. But I will check it right now. 

Also, regarding my workaround. I tested moving a Part to a Folder, it is not working. 
The Popup Window to change the Group property does not show Parts, too.

leoheck, 2023-05-04:
Where did you see this fix? They are saying they fixed for Wayland only, right?

leoheck, 2023-05-04:
Nevermind, found it.

<br class="Apple-interchange-newline">

leoheck, 2023-05-04:
They are saying it was fixed on 44.2, 

I still have the 44.0 for now.
➜ mutter --version    
mutter 44.0

adrianinsaval, 2023-05-04:
you should contact ubuntu then, this is a critical bug that they should at least backport if they are not updating to 44.2

adrianinsaval, 2023-05-04:
Eitherway, this is a gnome/ubuntu bug so I don't think there's anything for us to do so I'll close this issue, if you find there is something we can do on our side you may reopen this.

> Also, regarding my workaround. I tested moving a Part to a Folder, it is not working.
The Popup Window to change the Group property does not show Parts, too.

yes I've seen the other issue you created, but let's keep the discussion for that on the corresponding issue

closed 2023-05-04 by adrianinsaval

leoheck, 2023-05-05:
Here is an improved workaround that works with Folders/Parts too.
https://forum.freecad.org/viewtopic.php?t=78118

Selected object/objects and the destiny folder/part, and then paste the code from the url above.

leoheck, 2023-05-05:
Hey, @adrianinsaval this comment here https://github.com/FreeCAD/FreeCAD/issues/9392#issuecomment-1528258657 says that the 0.20 donwloaded with the package manager works just fine. It looks like it is a bug with the AppImage.

adrianinsaval, 2023-05-05:
@rbuehlma are you running wayland? the appimage runs on xwayland when in a wayland session, while teh apt package would run natively on wayland. The issue seems to be exclusive to X11 so that would explain it.

rbuehlma, 2023-05-05:
Yes, you are right, my system is running on wayland.

luzpaz, 2023-05-05:
Heads up @PLChris (there is an upstream Linux based issue with Gnome that is causing UI/UX instability)

alexbegoon, 2023-06-16:
I've upgraded to `mutter 44.2` on Ubuntu 23.04 (it's currently in the proposed repo), but the issue persists. It's still present on X11.

adrianinsaval, 2023-06-17:
can you reproduce on some other window manager? if not consider reporting this to gnome and ubuntu please. Does it also happen with any other qt application? what version of freecad are you using? (full version info please)

whatevergit, 2025-05-20:
I have the same problem, with NixOS 24.11, Wayland/Hyprland, FreeCAD 1.0.0

I have a very strange solution, that sometimes works, sometimes not:
right click on the body, den left click and hold, pulling it up a little, then release. After that maneuver the body sticks to my mouse and I can move it where I want to, as long as I don't have to scroll.