FreeCAD/FreeCAD #9394: [Issue] Snap version does not display template edit handles

WandererFan, 2023-04-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77905

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33004 (Git) Snap 686
Build type: Release
Branch: master
Hash: 8c382af744f9a5cddd06f9195daa5629dbcf7536
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * A2plus 0.4.60k
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Local build and AppImage display OP's custom template correctly, Snap does not.  Snap version does display templates from the distribution correctly. 

Possibly related to recent changes to Svg handling for Qt6??

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-26:
Let me know what needs to happen to the configs on the Snap side of things.

wwmayer, 2023-04-26:
> Possibly related to recent changes to Svg handling for Qt6??

This can be ruled out. As a start point you can use 96bf29304

wwmayer, 2023-04-27:
The change to blame is PR #9352

luzpaz, 2023-04-27:
@wwmayer i made that PR :face_with_spiral_eyes:  
What did it break ?

wwmayer, 2023-04-27:
> What did it break ?

The changes made to DrawViewArch::getSVGHead(), DrawViewDraft::getSVGHead(), DrawViewSpreadsheet::getSVGHead()

adrianinsaval, 2023-04-27:
can it be made to match either freecad or freecadweb variations or do we revert to freecadweb here? If we need to go back to freecadweb it probably needs to be reverted on all built-in templates too

WandererFan, 2023-04-27:
Thanks @wwmayer.  I was still in mid-bisect.

Notice to custom template users:
https://forum.freecad.org/viewtopic.php?t=77955

WandererFan, 2023-04-27:
"can it be made to match either freecad or freecadweb variations or do we revert to freecadweb here?"

Going forward, we can change the code to use either url, but the v0.20 and earlier standard templates will still point to freecadweb and the old code will still look for freecadweb.

As long as the freecadweb url is still valid the old version will still work.  Is freecadweb going to be removed any time soon?

Syres916, 2023-04-27:
Could the references to Snap please be removed as this applies to all, does it not?

adrianinsaval, 2023-04-27:
Keeping the freecadweb domain redirect shouldn't be expensive so I don't think there's any reason to get rid of it

luzpaz, 2023-04-27:
>Keeping the freecadweb domain redirect shouldn't be expensive so I don't think there's any reason to get rid of it

We have to keep ownership of the freecadweb domain for 2 main reasons: it's an important legacy URL for the project and also has many many years of search engine history (people buy domain names just for that these days because of traffic and high search ranking

luzpaz, 2023-04-27:
My bad, fat fingers

wwmayer, 2023-04-28:
@WandererFan 
I guess the ticket can be closed as 1b2857c30 is supposed to fix it.

WandererFan, 2023-05-01:
completed.

closed 2023-05-01 by WandererFan