FreeCAD/FreeCAD #9396: [Issue] Expression syntax error with single quote

ConnorWorrell, 2023-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * dodo
[/code]
```

### Subproject(s) affected?

Core

### Issue description

Expressions using single quotes throw error when clicking OK.

Steps to reproduce:
1. Open any formula editor
2. Input 1'
3. Click ok
4. Error

`<class 'SyntaxError'>: ('unterminated string literal (detected at line 1)', ('<string>', 1, 75, "App.getDocument('Unnamed').Sketch.setExpression('.Placement.Base.x', u'1 '')", 1, 75))`

It appears these single quotes are not being properly escaped.

Potential for injection:  1') or print('jfdkls

Putting this here since this link seems dead:
https://github.com/FreeCAD/FreeCAD/security/advisories/new

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-26:
@ConnorWorrell can you test on dev version please ?

ConnorWorrell, 2023-06-29:
Tested on master branch shows same error with same steps to reproduce.

```
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Unknown
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * dodo
```

`<class 'SyntaxError'>: ('unterminated string literal (detected at line 1)', ('<string>', 1, 72, "App.getDocument('Unnamed').Sketch.setExpression('Constraints[0]', u'1 '')", 1, 72))`

Roy-043, 2023-07-15:
Confirmed. You can however enter `1ft` instead.

Roy-043, 2025-01-02:
This has been fixed.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

closed 2025-01-02 by Roy-043