FreeCAD/FreeCAD #9403: [Issue] Memory leak in Path

xtemp09, 2023-04-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77927

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32960 (Git)
Build type: Debug
Branch: path-fix
Hash: 59a208e81bfaeba560ab6939e66e3b89b413f040
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Path

### Issue description

I compiled FreeCAD using `clang 16.0.3` and `mold 1.11.0` with leak sanitizer with the following configuration:

<details><summary>The configuration</summary>

```python
cmake .. -G Ninja \
-D CMAKE_BUILD_TYPE=Debug \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/path-test/ \
\
-D CMAKE_C_COMPILER=clang -D CMAKE_CXX_COMPILER=clang++                   \
-D CMAKE_C_FLAGS_DEBUG="-glldb -fsanitize=leak -fno-omit-frame-pointer"   \
-D CMAKE_CXX_FLAGS_DEBUG="-glldb -fsanitize=leak -fno-omit-frame-pointer" \
-D LINK_FLAGS_DEBUG="-fsanitize=leak -fuse-ld=mold"                       \
\
-D PYTHON_LIBRARY_DEBUG=/usr/lib/x86_64-linux-gnu/libpython3.10d.so       \
\
-D CMAKE_AR=/usr/lib/llvm-16/bin/llvm-ar               \
-D CMAKE_ADDR2LINE=/usr/lib/llvm-16/bin/llvm-addr2line \
-D CMAKE_MT=/usr/lib/llvm-16/bin/llvm-mt               \
-D CMAKE_NM=/usr/lib/llvm-16/bin/llvm-nm               \
-D CMAKE_OBJCOPY=/usr/lib/llvm-16/bin/llvm-objcopy     \
-D CMAKE_OBJDUMP=/usr/lib/llvm-16/bin/llvm-objdump     \
-D CMAKE_RANLIB=/usr/lib/llvm-16/bin/llvm-ranlib       \
-D CMAKE_READELF=/usr/lib/llvm-16/bin/llvm-readelf     \
-D CMAKE_RC_COMPILER=/usr/lib/llvm-16/bin/llvm-rc      \
-D CMAKE_STRIP=/usr/lib/llvm-16/bin/llvm-strip
```

</details>

Then I executed FreeCAD with the command `./FreeCAD > leak-report.txt 2>&1`

I ran simulation within the file [drill_sheet_jig.FCStd](https://forum.freecad.org/download/file.php?id=169503) attached to the first post of [this topic](https://forum.freecad.org/viewtopic.php?f=15&t=63656) twice. This file contains a malformed tool, but the simulation works with my patch sent at #9339. I think this is irrelevant.

Attached is the [leak-report.txt](https://github.com/FreeCAD/FreeCAD/files/11334486/leak-report.txt). FreeCAD leaked 56 Mb. Sure, this is incomparable to 11-20 Gb observed in Draft, but still a lot.

Major leaks come from the file `src/Mod/Path/PathSimulator/App/PathSimPy.cpp`.

### Anything else?

[leak-report.txt](https://github.com/FreeCAD/FreeCAD/files/11334534/leak-report.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-05-10:
I can confirm that PathSim doesn't correctly implement the memory management. The methods `PathSim::BeginSimulation` or `PathSim::SetToolShape` simply create new instances of `cStock` or `cSimTool` without deleting already existing instances.

The raw pointers should be replaced with `std::unique_ptr` to make sure memory is deallocated automatically.

wwmayer, 2023-05-11:
In my test I got a leak of 10 MB and after fixing the PathSim class it went down to 5 MB.

@xtemp09 if you apply the attached patch can you conform that it fixes this issue?
[9403.patch.txt](https://github.com/FreeCAD/FreeCAD/files/11450746/9403.patch.txt)

xtemp09, 2023-05-12:
Yes, I can confirm that the memory leaks are gone. The leaks before your patch: 32 Mb, after: 5 Mb.

Here are the reports. [before.txt](https://github.com/FreeCAD/FreeCAD/files/11459818/before.txt), [after.txt](https://github.com/FreeCAD/FreeCAD/files/11459821/after.txt)

wwmayer, 2023-05-12:
OK, thanks. Then I will prepare a PR.

xtemp09, 2023-11-18:
I compiled FreeCAD using `gcc 12.3.0` and `mold 2.1.0` with _leak sanitizer_ with the following configuration:

<details><summary>cmake configuration</summary>

```
cmake .. -G Ninja                                 \
-D CMAKE_BUILD_TYPE=Debug                         \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/path-test/  \
\
-D CMAKE_CXX_STANDARD=20                          \
-D BUILD_ENABLE_CXX_STD=C++20                     \
\
-D CMAKE_C_FLAGS_DEBUG="-g -Og -fno-omit-frame-pointer -fvar-tracking-assignments -fsanitize=leak -Wno-deprecated"   \
-D CMAKE_CXX_FLAGS_DEBUG="-g -Og -fno-omit-frame-pointer -fvar-tracking-assignments -fsanitize=leak -Wno-deprecated" \
\
-D LINK_FLAGS_DEBUG="-fuse-ld=mold"
```

----

FreeCAD version:

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35068 (Git)
Build type: Debug
Branch: main
Hash: cbc3d81bcad67b99ef39a9b4125a9b8acef656f4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

I ran the same simulation.

```
SUMMARY: LeakSanitizer: 28598205 byte(s) leaked in 15118 allocation(s).
```

The memory leak has returned. Perhaps, it depends on Python, Boost.Python, Shiboken or some options.

Attached is the complete [leak-report.txt](https://github.com/FreeCAD/FreeCAD/files/13399394/leak-report.txt).

wwmayer, 2023-11-25:
> The memory leak has returned. Perhaps, it depends on Python, Boost.Python, Shiboken or some options.

I still only have a leak of ~5MB but instead of boost.python I use pybind11

xtemp09, 2023-11-26:
I can't find anything on usage of `pybind11` on Ubuntu, only on Windows. [^1]

So, what is the conclusion? To use `pybind11` instead of `libboost-python-dev` to avoid the memory leak?

[^1]: https://wiki.freecad.org/Compile_on_Windows

wwmayer, 2023-11-26:
In Ubuntu there is a package pybind11-dev. After installing it run cmake-gui and set the option FREECAD_USE_PYBIND11. Now re-run the build process.

xtemp09, 2023-11-26:
I've compiled the same source code of FreeCAD with the same cmake configuration but with pybind11 on.

```
SUMMARY: LeakSanitizer: 4874685 byte(s) leaked in 6511 allocation(s).
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35068 (Git)
Build type: Debug
Branch: main
Hash: cbc3d81bcad67b99ef39a9b4125a9b8acef656f4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

xtemp09, 2025-01-23:
I like this autoclosing idea. No one has done anything to fix the problem, while the issue is closed and moved to done.

luzpaz, 2025-01-28:
@maxwxyz auto-closing CAM tickets isn't a good idea since there's been so much backlog.

luzpaz, 2025-01-28:
@shaise any chance you could check this out ?

xtemp09, 2025-01-28:
Usage of _pybind11_ gives much smaller leaks in comparison to _Boost.Python_. If the support for _Boost.Python_ is going to be removed (#17779), the issue should be reconsidered.

shaise, 2025-01-28:
@luzpaz , thanks for letting me know, I will take a look.

luzpaz, 2025-01-28:
@shaise some more relevant info at https://github.com/FreeCAD/FreeCAD/issues/19200

luzpaz, 2025-07-08:
Is this still relevant ?

xtemp09, 2026-01-17:
Transition to _pybind11_ has made the leaks invisible or eliminated them. Here are the leak reports for the new and the legacy CAM simulators.

[leaks-new-CAM-sim.txt](https://github.com/user-attachments/files/24687211/leaks-new-CAM-sim.txt)

[leaks-leg-CAM-sim.txt](https://github.com/user-attachments/files/24687212/leaks-leg-CAM-sim.txt)

Though, I ought to add that now I'm using _gcc_ 13.3, I can't compile _FreeCAD_ with _clang_ anymore. If compilation with Qt 5 is impossible, then I won't be able to compile _FreeCAD_ with a debug version of Qt.

commit 50056462 (2023-05-15): Merge pull request #9554 from wwmayer/issue_9403

Path: fixes #9403: Memory leak in Path
commit c242b5de (2023-05-15): Path: fixes #9403: Memory leak in Path
closed 2026-01-17 by xtemp09