FreeCAD/FreeCAD #9405: Path: Op/Drilling.py has superfluous translations of gcode

luzpaz, 2023-04-27:
https://github.com/FreeCAD/FreeCAD/blob/25157275f07075ccd40873837f9b0f48aef28c5a/src/Mod/Path/Path/Op/Drilling.py#L69-L70

@sliptonic Should this gcode even need to be translated ?

luzpaz, 2023-04-30:
@sliptonic when you have a moment, please advise. TIA

sliptonic, 2023-04-30:
Exposing the Gcode commands to the user isn't very friendly. It's fine to store the G98/99 value but the displayed value should be something more user friendly and that should be translated.  The desired behavior is described  
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g98-g99

Maybe something like "Safe Height" and "retract height"

mlampert, 2023-05-12:
Tried to fix this but it's a bit contrived. One can change the `RetractMode` property - however it gets its forced value on each `circularHoleExecute()` evaluation from derived from `KeepToolDown`.
The best approach would be to remove the property `RetractMode`, but it was only recently refactored so I'm not sure what the plan is here. Maybe @J-Dunn can weigh in.

J-Dunn, 2023-05-12:
The GCODE G98/99 behaviour is quite specific, as accurately reflected in the link sliptonic provided,  and can not be summed up in some "user friendly" two word snippet. 

Trying to dumb this down will just cause confusion. 

If the user is going to use canned cycles he needs to know what they are and how they work. If FC is going to do that on his behalf, it should be clearly indicated what is being done so the user can find out what G98 and G99 do. 

The plan of introducing this new property to get away from the previous hacked bahavoir which was neither G98 no G99 but someone's person trick to simulate all canned cycles as G99 behavious while issuing G98 instructions. 

This was frankly a mess and caused me enough tool crashes and trashed clamps I decided to fix it. 

The RetractMode is effectively a means of selecting whether you use G98 or G99 behaviour between holes.  If anyone thinks it is not doing that please point out the failure. 

If anyone thinks G98 was a mistake by those writing the standards and all holes should be rigged to do G99 using G98 commands maybe they could make there case for reverting.

luzpaz, 2023-06-10:
@sliptonic do you think this will get addressed pre-release ?

luzpaz, 2025-11-19:
@J-Dunn any chance you can revisit this ?

J-Dunn, 2025-11-21:
> [@J-Dunn](https://github.com/J-Dunn) any chance you can revisit this ?

Nope. I've had enough of being insulted and having my time wasted. 
Thanks for asking.