FreeCAD/FreeCAD #9406: Draft: Clarify specific translated string in `draftguitools/gui_selectplane.py` so that it's less ambiguous

luzpaz, 2023-04-27:
https://github.com/FreeCAD/FreeCAD/blob/25157275f07075ccd40873837f9b0f48aef28c5a/src/Mod/Draft/draftguitools/gui_selectplane.py#L523

Most likely in context meant to be 'Direction'. But some translators might confuse with 'Directory'.

yorikvanhavre, 2023-04-28:
This text is written on the toolbar button. It needs to be kept as short as possible to not make the button too long or have truncated text. However maybe there is an unicode icon that could represent a direction better? Maybe ↗ ?

d1nuc0m, 2023-04-30:
Is it possible to add a comment visible to translators? Something like "Dir here means Direction, not Directory. Also shorten the translation because of available space in GUI"

luzpaz, 2023-04-30:
We can use the 3rd field: 
`vdir = " " + translate("draft", "Dir", "Dir here means Direction, not Directory. Also shorten the translation because of available space in GUI") + ": " + vdir`

d1nuc0m, 2023-05-01:
> We can use the 3rd field: `vdir = " " + translate("draft", "Dir", "Dir here means Direction, not Directory. Also shorten the translation because of available space in GUI") + ": " + vdir`

I don't know the details of the translation system, but if it works... why not?

chennes, 2023-05-02:
Yeah, using the comment string is a good idea here. Whether translators actually notice the comment is an open question, of course :).

PR #9464 (2023-05-05): Draft: Add l10n context for ambiguous string
Also tweaked code formatting  
Closes #9406

---

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit c6b2b3fa (2023-05-05): Draft: Add l10n context for ambiguous string
Also tweaked code formatting
Closes #9406