FreeCAD/FreeCAD #9407: Plot: advise user how they are impacted and what to do if matplotlib isn't installed

luzpaz, 2023-04-27:
https://github.com/FreeCAD/FreeCAD/blob/0a23f9b42c02ee2609dd63b99f9b0e8bde4a237f/src/Mod/Plot/Plot.py#L45-L47

* Can this output a more helpful error that explains how the user's workflow is impacted if Plot workbench isn't available.
* And also some orientation on how to solve the issue of installing matplotlib (or noting that they may need to let their package manager of their OS/distro to make sure to include matplotlib as a 3rd party dependency of FC)

adrianinsaval, 2023-04-28:
what more should it say? We can't know how their workflow is impacted because we don't know their workflow. Installation instruction vary widely across systems so I don't see how or why we should cram that into the error message.

I don't think this is a big issue, most people will have it installed, AFAIK all our packages include it and most serious package managers should list it as a dependency. Have you encountered this error yourself or people confused by it?

luzpaz, 2023-04-29:
The installed plot workbench is bare bones. Most of the complexity was moved in to [freecad.plot](https://github.com/FreeCAD/freecad.plot) external addon. So say a new user has downloaded FC without it being compiled with matplotlib (and therefore no internal plot wb), to me it makes sense to say (not verbatim) 

'Looks like FC was compiled without 3rd party matplotlib library. This shouldn't impact your workflow unless you are planning to generate plots (see https://wiki.freecad.org/Plot_Workbench for more details). If this is a necessary step in your workflow, download the official FC from freecad.org' 

@oliveroxtoby pardon. question: what would you let a user know (in a succinct way) if Plot workbench wasn't available?

adrianinsaval, 2023-04-29:
this is python code, you don't need matplotlib when compiling it's just a matter of having it available at runtime.

oliveroxtoby, 2023-04-29:
> @oliveroxtoby pardon. question: what would you let a user know (in a succinct way) if Plot workbench wasn't available?

To me, that error message already seems quite specific in telling you that missing matplotlib is the cause of the issue; including further instructions would be overkill and likely to age poorly IMHO. One could perhaps just change "matplotlib not found" to "The matplotlib python package was not found" to give a bit more context?

luzpaz, 2023-04-30:
The fix didn't address the original issue. But at least the error message is a slightly friendlier?

PR #9442 (2023-05-05): Plot: tweak missing matplotlib warning
Close #9407

---

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit b4834dc5 (2023-05-05): Plot: tweak missing matplotlib warning (#9442)

Close #9407
closed 2023-05-05 by adrianinsaval