FreeCAD/FreeCAD #9409: Segfault loading BIM file

sundtek, 2023-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
https://www.soil-parts.at/downloads/
https://www.soil-parts.at/app/download/9495163882/Beispiel+IFC-Modell.ifc?t=1637736812

0, debug map time is 2022-10-02 05:59:04.000000000) since this executable was linked, file will be ignored
* thread #1, name = 'CrBrowserMain', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x00007fff6dceae52 libsystem_platform.dylib`_platform_strlen + 18
    frame #1: 0x00000001000101e5 FreeCAD`std::__1::char_traits<char>::length(__s=0x0000000000000000) at __string:253:53
    frame #2: 0x00000001052eb149 libFreeCADApp.dylib`std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(this="%\x9es\x01\0\0\0 >???\0\0*", __s=0x0000000000000000) at string:819:19
    frame #3: 0x00000001052d681d libFreeCADApp.dylib`std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(this="%\x9es\x01\0\0\0 >???\0\0*", __s=0x0000000000000000) at string:817:88
    frame #4: 0x00000001052e5f43 libFreeCADApp.dylib`App::ObjectIdentifier::access(this=0x00007ffeefbf4070, obj=0x0000000132d07800, prop=0x0000000000000000, propName=0x0000000000000000) const::$_0::operator()(App::DocumentObject*, App::Property*, char const*) const at ObjectIdentifier.cpp:1686:28 <<<<------

        if(propset.size()!=1 || propset.begin()->size()) {
            if(!propName)
                propset.clear();
----
>>            if (propName == nullptr)
>>                propName = strdup("");
----
            propset.insert(propName);
        }

that's what I did for now, somewhere else maybe it expects a nullptr in that propset however it points out that there's an issue here.
```

### Subproject(s) affected?

None

### Problem description

https://www.soil-parts.at/downloads/
https://www.soil-parts.at/app/download/9495163882/Beispiel+IFC-Modell.ifc?t=1637736812

loading this model crashes freecad

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sundtek, 2023-04-27:
issue is fixed in latest sourcecode

closed 2023-04-27 by sundtek

luzpaz, 2023-04-27:
Please, in the future, always add your full [About](https://wiki.freecad.org) info. Thanks