FreeCAD/FreeCAD #9411: SIGSEGV error when trying to use filled

monkiki, 2023-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * fasteners 0.4.6
[/code]
```

### Subproject(s) affected?

None

### Issue description

This error is raised when trying to make this part rounded:

[image]

[code]
10:48:54  SIGSEGV signal raised: 11
10:48:54  Recompute failed! Please check report view.
[/code]

### Anything else?

This is the sample design which causes the error:

[fillet error.zip](https://github.com/FreeCAD/FreeCAD/files/11341544/fillet.error.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-04-27:
The Fillet issue has been reported many times and it's due to a 3rd Party product called OCCT (OpenCascade). There are ways to mitigate this by improving your modelling practices. Please see attached file for a comparison.
[fillet error_Not.zip](https://github.com/FreeCAD/FreeCAD/files/11342069/fillet.error_Not.zip)

monkiki, 2023-04-27:
I see: you have used an sketch to generate the model avoiding the boolean operations with basic forms. Great! :)

closed 2023-04-27 by monkiki