FreeCAD/FreeCAD #9414: [Feature Request] Change guessing algorithm of Rotation::makeRotationByAxes()

Jolbas, 2023-04-27:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77092

### Subproject(s) affected?

Core

### Idea description

The wiki for [OXY attachment](https://wiki.freecadweb.org/O-X-Y_type_attachment_modes) describes how the second direction is guessed when omitted. I don't think the guessing process is very good and have room for improvement but I want your opinion before changing.
I started to look into this after reading a [post](https://forum.freecad.org/viewtopic.php?p=656069#p656069) by @ppemawm and the wiki by @edwilliams16.
I like to suggest that the "hint" direction instead uses the local axis corresponding to the last character. This is true now for OXZ, OYZ and OZY attachments. But not true for OXY, OYX and OZX. 

To achieve this I planned to target the rotation constructor Rotation(vX, vY, vZ, orderString) in Rotation.cpp which is used by the Attachment engine. I realize the change in some cases may brake models but I think it will make it more intuitive and beneficial for future works.

If the orderString is for example "YZX" the current implementation works like this to find a rotated CS:
- if vY (first character in "YZX") is not a zero vector, lock the target Y axis to this direction
- else try the same for vZ and then vX
- look for a hint direction in remaining vectors
- if no hint direction is found, use a hard coded local axis as hint depending on the main axis as described in the wiki link above

I would like to change the behavior to this:
- if vY is not a zero vector, lock the target Y axis to this direction
- else lock the target Y axis to the local Y
- if vZ (second character in "YZX") is not a zero vector and not equal to main direction, use as a hint for the new Z axis
- else use the local Z as hint direction
- the third vector is never used

I suggest that a `orderString` with only two characters is accepted too.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2023-04-28:
I agree that the current behavior of the attachment modes is inconsistent - but I don't think the best way to go is to change the behavior of App.Rotation(v1,v2,v3, 'XYZ') modes when the input vector(s) are zero. Zero vectors are of course invalid, as they are supposed to represent directions, so we are indeed free to do what we want with them. 

let

```
V = App.Vector
R = App.Rotation
vx = V(1,0,0)
vy = V(0,1,0)
vz = V(0,0,1)
def T(rot):
    print(f'vx->{rot*vx}\nvy->{rot*vy}\nvz->{rot*vz}\n')

```

Consider the rotation  R(vx, -f*vz, vy) where f>=0 is a float. Currently:

```
>>> T(R(vx, -f*vz, vy, 'YZX'))
vx->Vector (1.0, 0.0, 0.0)
vy->Vector (0.0, 2.220446049250313e-16, -1.0)
vz->Vector (0.0, 1.0, 2.220446049250313e-16)

>>> f =0
>>> T(R(vx, -f*vz, vy, 'YZX'))
vx->Vector (1.0, 0.0, 0.0)
vy->Vector (0.0, 2.220446049250313e-16, -1.0)
vz->Vector (0.0, 1.0, 2.220446049250313e-16)

```

So when f=0 the algorithm continuously reverts to the same rotation. However, with the proposed algorithm, 
 R(vx, 0, vy, 'YZX') would evaluate as  R(vx, vy, vy, 'YZX')

 ```
>>> T(R(vx, vy, vy, 'YZX'))
vx->Vector (1.0, 0.0, 0.0)
vy->Vector (0.0, 1.0, 0.0)
vz->Vector (0.0, 0.0, 1.0)
```

which would be discontinuous at f=0

I agree with your proposal

"I like to suggest that the "hint" direction instead uses the local axis corresponding to the last character."

but I think the 'fix' should be directed at the attachment modes - in the situation where insufficient references are provided to pin down the axial rotation.
 So, for instance, for 'OYZ', if no reference is provided for the Z direction hint, the local Z direction is used, as proposed. 

This still leaves the question of what to do if the Y-direction selected is the same as the local z-axis, thus providing no hint direction.  One possibility might be to do the rotation in the (y, newy) plane. App.Rotation(y, newy) - another might be to make the OYZ unavailable until a further selection is made.

Jolbas, 2023-04-28:
@edwilliams16  Yes, I can see a point having the function being continuous but the current implementation just happen to be continuous in your example. With e.g. ` T(R(vx, f*(vz+vy), vy, 'YZX'))`
you get

```
>>> T(R(vx, 1*(vz+vy), vy, 'YZX'))
vx->(-1.00,  0.00,  0.00)
vy->( 0.00,  0.71,  0.71)
vz->( 0.00,  0.71, -0.71)

>>> T(R(vx, 0*(vz+vy), vy, 'YZX'))
vx->( 1.00,  0.00,  0.00)
vy->( 0.00,  0.00, -1.00)
vz->( 0.00,  1.00,  0.00)

>>> T(R(vx, -1*(vz+vy), vy, 'YZX'))
vx->( 1.00,  0.00,  0.00)
vy->( 0.00, -0.71, -0.71)
vz->( 0.00,  0.71, -0.71)

```
I can't see a way to guarantee it to be continuous at zero and how it should guess the intended series of inputs.

edwilliams16, 2023-04-29:
Yes.  Continuity is not a particularly good argument.  
    It makes me uneasy to rely on undocumented behavior of App.Rotation(a,b,c, 'XYZ') for invalid inputs  (which includes not only zero vectors, but equal ones for the priority ones.)  to fix a higher level function.  What it does now is not unreasonable and who knows what code might rely on it corner cases.
   
 As  I said before, I would prefer the logic of the Attachment modes to be where ambiguities are resolved. But if we change the logic there, it could still break previously constructed models.   If we had an empty canvas, it could be done better, but it isn't clear to me that changing the behavior now, just to make things a little less arbitrary is a good idea.

 I think the typical user of the OXY-type Attachments makes the first two selections and if the result is as desired, stops right there. If not he/she adds a third reference.  That workflow is fine whatever we make the defaults, but changing those defaults breaks previous models.  Is it worth that price?

Jolbas, 2023-04-29:
> Is it worth that price?

I would say yes. I think the current behaviour is hard to explain and that it's just good luck if the outcome is the expected. If I put the first vertex on origo and the second vertex moves around it on the XZ plane. Then if I choose OXY attachment, the Y axis flips two times per turn because the attacher tries to align Z to the local Z. Z is not mentioned in 'OXY' which makes the attachment unpredictable to me.

edwilliams16, 2023-04-29:
I agree it is a matter of luck if the code-chosen default determining the axial orientation is correct. But that is true whatever default is chosen.

A user wants to attach an object to a vertex with some specific orientation.

1.   He chooses the attachment vertex

2.  He chooses the direction to align one of the axes (X/Y/Z) of his object to.

At this point the axial orientation is not specified, so the code chooses one for the sake of definiteness.  If the object is axisymmetric, or he is lucky, he's done.  If this is not the case a third reference is required.

So what is the point of changing the default? Will a user be lucky more often?  Why would a user care exactly what the default is, in advance?  He can inspect to see if he needs a third reference or not.

Jolbas, 2023-05-01:
- I think the default would be easier to guess.
- There would be a meaningful difference between OXY and OXZ when only two reference points is selected.
- There may not be a third reference point available to represent the local axis
- As you pointed out at the end of [this post](https://forum.freecad.org/viewtopic.php?p=670858#p670858) it would be possible to select the same point for the first two references and achieve a lock to a local axis and then use the third reference as hint without an available reference to a local axis in relation to the selected origin.

Jolbas, 2023-08-19:
I now realized that if only 1 vector is passed, the sequence argument is actually completely ignored.