FreeCAD/FreeCAD #9415: [Problem] Navigation animations don't always work

Rexbas, 2023-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) (and also 0.21)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Some navigation animations don't work with animations enabled. The animations don't work when using the two arrows above the NaviCube. Also, alternating between the corner and edge of the NaviCube results in maybe one or two frames of animation and feels like it immediately snaps to the selected view. I attached a video that demonstrates this.

I have been working on something to deal with different kinds of navigation animations. A first version, that also fixes the animations, is almost finished so expect a PR soon.

### Anything else?

https://user-images.githubusercontent.com/14298143/234894433-039e478c-4b04-48e6-bebf-c49a26bea4d4.mp4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-30:
Have you tested this on development branch ? Navigation cube code just got refactored in #9356

Rexbas, 2023-05-01:
Yes I have tested it on the development branch and I was aware that the navigation cube got refactored. The navigation cube is, however, not directly related to the animations. The problem is clearly visible when using the navigation cube as described previously.

PR #9446 (2023-09-25): Gui: Refactor navigation animations
Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or changed once this PR is  [... cut]
PR #10881 (2023-10-25): Gui: Refactor navigation animations (attempt 2)
- Animations are played through an Animator
- Standard animations have a fixed duration and inherit from QVariantAnimation
- The animation duration can be changed in the preferences
- Fix animations when using the two arrows above the NaviCube
- Start an animation or start and wait for an animation
- Replaces standard camera animations (e.g. by selecting standard views or NaviCube) with FixedTimeAnimation
- Replace View3DInventorViewer's CameraAnimation with FixedTimeAnimation
- Replace OpenInventor navigation style infinite spin animation with S [... cut]
closed 2023-09-25 by yorikvanhavre