FreeCAD/FreeCAD #9422: [Feature Request] Dependency Graph: export to .dot and send to external tools

d1nuc0m, 2023-04-27:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=61355

### Subproject(s) affected?

Other (specify in description)

### Idea description

The Dependency Graph of complex projects can become quite messy, it would be nice to be able to export it through the GUI as a .dot Graphviz file, to use external tools. 

It is already possible to export it but it's not easy, as it requires two instructions in the Python console
```python
import contextlib
with contextlib.redirect_stdout(open('/path/to/my/file.dot', 'w')): App.ActiveDocument.DependencyGraph
``` 
and some file cleanup to remove first/last characters (as there are unwanted ' due to Python console) and replace `\n` with actual newlines ([kudos to SO](https://stackoverflow.com/questions/24574489/replace-n-with-newline-in-awk))
```bash
awk -F: '{print substr ($0,2,length($0)-2)}' /path/to/my/file.dot | awk '{gsub(/\\n/,"\n")}1' >> /path/to/my/file.dot
```

So it would be useful to have a GUI to save the .dot file of the Dependancy Graph.

A nice addition would be another separate button/choice/whatever that saves a copy of the file in the system temp directory and feeds it to a tool that the user configures in FC Settings (as there are a lot of external tools for graphviz, see links).

### Anything else?

**Forum**
[[56579] Graphviz to show Part of Model, be editable etc.](https://forum.freecad.org/viewtopic.php?t=56579)
[[56725] Graph to show Part of Model, editable etc. - by Graphviz ?](https://forum.freecad.org/viewtopic.php?t=56725)
[[61355] Do we have any way to export the dependency graph in a textual/script way?](https://forum.freecad.org/viewtopic.php?t=61355)

**External**
[Graphviz Tools (awesome graphviz)](https://stackoverflow.com/questions/24574489/replace-n-with-newline-in-awk)
[Graphviz graphical interfaces (graphviz.org)](https://stackoverflow.com/questions/24574489/replace-n-with-newline-in-awk)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

d1nuc0m, 2023-04-30:
Add: it might be useful (when exporting to a .dot/.gv file for external use) to use the `-Tcanon` switch for a clean format ([Graphviz forum](https://forum.graphviz.org/t/linter-for-the-dot-language/1594/8)).

If I'm not wrong, the graph is managed in:

- /src/App/DocumentPyImp.cpp
- /src/App/Graphviz.cpp
- /src/Gui/GraphvizView.cpp

and the CLI options are in [GraphvizView.cpp#L395-L411](https://github.com/FreeCAD/FreeCAD/blob/b5a548e3fd931745667dd482aa39305f4db49ecf/src/Gui/GraphvizView.cpp#L395-L411)

Is this a good (and feasible) idea?

d1nuc0m, 2023-05-01:
Ping @luzpaz please could you label this issue?

luzpaz, 2023-09-07:
@wwmayer when you have a moment, can you weigh in?

luzpaz, 2023-09-07:
Thanks wwmayer :heart:

closed 2024-07-05 by adrianinsaval