FreeCAD/FreeCAD #9423: [Problem]

amreo, 2023-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Problem description

Freecad shows the error "Can't open input file '/var/tmp/fc-00002-510987-000001.scad'!" after I click to add in the add openscad object section. I expect that a cube is added in the 3d view.

Steps to reproduce the bug:
1) open freecad (I have removed /var/config/* before this test)
2) select openscad environment
3) click to "add openscad object" in the toolbar (the button with openscad logo)
4) in the left panel there should be the "add openscad object" section
5) click to add

In the error text should show a text similar to " Can't open input file '/var/tmp/fc-00002-551901-000002.scad'!". I expected to view a cube in the right 3d view

[image]

### Anything else?

I use the flatpak version of freecad. I have cleaned the /var/app/config files before the test.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-28:
I assume you do have openscad installed through flatpak too? Are you using the built-in openscad wb or the one available in addon manager?

chennes, 2023-04-28:
I think with the Flatpak you might have to switch to using the "named pipe" transfer mechanism -- it's in the OpenSCAD WB preferences. Give that a shot and see if it fixes the issue for you.

amreo, 2023-04-28:
> I assume you do have openscad installed through flatpak too? Are you using the built-in openscad wb or the one available in addon manager?

Yes, openscad is installed inside the flatpak (but I also have installed openscad using 'pacman`) 
```
[amreo@amreo-manjaro-pc ~]$ flatpak run --branch=stable --arch=x86_64 --command=/app/bin/openscad org.freecadweb.FreeCAD  --version
OpenSCAD version 2021.01
```

Yes, I am using the built-in and default openscad wb, with no customization

amreo, 2023-04-28:
> I think with the Flatpak you might have to switch to using the "named pipe" transfer mechanism -- it's in the OpenSCAD WB preferences. Give that a shot and see if it fixes the issue for you.

I don't find "named pipe" transfer mechanism in `OpenSCAD WB preferences`. Allowed options for "Send to OpenSCAD via:" are:
* Standard temp directory
* User specified directory
* stdout pipe

[image]

Anyway using the "stdout pipe" transfer mechanism it works

chennes, 2023-04-28:
Yeah, sorry, I wrote the code too long ago, I forgot the name! Yes, the stdout pipe is the one I meant. Glad to hear it works! I have autodetection code for some other packaging types, but I didn't know how to detect running within a Flatpak.

adrianinsaval, 2023-04-28:
Could you do me the favor of providing a patch to make that the default @chennes ? I can then apply it while packaging for flatpak

chennes, 2023-04-28:
I'm a little hesitant to make it the default since it requires the most recent version of OpenSCAD. Is that available in all of our target distros?

chennes, 2023-04-28:
Oh, I see, you meant a literal patch -- what format? git patch or patch?

chennes, 2023-04-28:
Here's a git patch
[OpenSCAD_transfermechanism.patch](https://github.com/FreeCAD/FreeCAD/files/11357784/OpenSCAD_transfermechanism.patch)

adrianinsaval, 2023-04-29:
Thank you, a flatpak update that makes this the default option will go live shortly however I notice that when entering the openscad preference page it still shows the first option as default on the dropdown, how may I make it so that the preference page also uses the pipe method as default? Otherwise after entering the preference page and clicking ok it changed back to standard temp directory

adrianinsaval, 2023-05-03:
ok we may not actually need to change the default for flatpak, I just checked how we are launching openscad from flatpak and we are giving it permissions to the wrong dir I think: https://github.com/flathub/org.freecadweb.FreeCAD/blob/00927053088318eaf152949971adb53d99728878/openscad.sh#L8
`BASEDIR="$XDG_CACHE_HOME/tmp"`

instead FreeCAD tries to write to /var/tmp, could you give some pointer on how is freecad determining what directory to use?

chennes, 2023-05-03:
It's the "transferdirectory" preference in the OpenSCAD Mod. I don't know where the default comes from, probably something like Python's default temp directory.

luzpaz, 2023-06-11:
Any updates on this ?

Knochi, 2024-07-28:
Facing the same problem when trying to import certain .csg files. Not all fail but some.
Seeing it on 0.21.0 and 0.21.2
Setting the "send to openSCAD via" option to "stdout pipe" gives me a different error. I have openSCAD 2021.1 installed via flatpak.

new error:

> 13:55:45  Recompute failed : difference
> 13:55:45  obj [<Part::PartFeature>] has no Name & Shape
> 13:55:45  opening /var/tmp/fc-00002-774206-000066.dxf...
> 13:55:45  There has been an error:
> 13:55:45  Failed to find any sections!
> 13:55:45  Traceback (most recent call last):
>   File "/app/freecad/Mod/OpenSCAD/OpenSCADFeatures.py", line 592, in execute
>     shape = OpenSCAD.OpenSCADUtils.process_ObjectsViaOpenSCADShape(fp.Document,fp.Children,\
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 648, in process_ObjectsViaOpenSCADShape
>     return process2D_ObjectsViaOpenSCADShape(children,name,doc)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/freecad/Mod/OpenSCAD/OpenSCADUtils.py", line 590, in process2D_ObjectsViaOpenSCADShape
>     face = importDXFface(tmpfilename,None,None)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/freecad/Mod/OpenSCAD/OpenSCAD2Dgeom.py", line 486, in importDXFface
>     layers = importDXF.processdxf(doc,filename,False,False) or importDXF.layers
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/freecad/Mod/Draft/importDXF.py", line 2147, in processdxf
>     drawing = dxfReader.readDXF(filename)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/deck/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Macro/dxfReader.py", line 399, in readDXF
>     (infile, drawing) = sm.run((infile, None))
>     ^^^^^^^^^^^^^^^^^
> <class 'TypeError'>: cannot unpack non-iterable bool object
> 13:55:45  Recompute failed!

luzpaz, 2024-08-30:
flatpaks still have an issue with openscad see related:
* https://github.com/FreeCAD/FreeCAD/issues/15926#issuecomment-2295373765

amreo, 2025-01-03:
I have tried to reproduce the issue, and even it doesn't show errors (neither in the report panel), it doesn't show no cube.

```
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

amreo, 2025-05-29:
Same error in freecad 1.0.1, no cube is shown

```
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

closed 2025-08-13 by github-actions[bot]