FreeCAD/FreeCAD #9424: [BUG] Flatpakked freecad fails to save openscad files

amreo, 2023-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Problem description

Freecad shows the error "AttributeError: PySide2.QtWidgets.QFileDialog.getSaveFileName(): unsupported keyword 'option'" after I click to add in the add openscad object section. I don't know what I should expect.

Steps to reproduce the bug:
1)  open freecad (I have removed /var/config/* before this test)
2) elect openscad environment
3) click to "add openscad object" in the toolbar (the button with openscad logo)
4) in the left panel there should be the "add openscad object" section
5) click to save

The report view should log something similar to this:
```
00:51:50  Traceback (most recent call last):
00:51:50    File "/app/freecad/Mod/OpenSCAD/OpenSCADCommands.py", line 432, in saveelement
00:51:50      filename, filter = QtGui.QFileDialog.getSaveFileName(parent=self.form, caption='Open file', dir='.', filter='OpenSCAD Files (*.scad)',selectedFilter='',option=0)
00:51:50  AttributeError: PySide2.QtWidgets.QFileDialog.getSaveFileName(): unsupported keyword 'option'
```

[image]

### Anything else?

I use the flatpak version of freecad. I have cleaned the /var/app/config files before the test.
Maybe it is related to https://github.com/FreeCAD/FreeCAD/issues/9423

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-28:
should that not be `options` instead? https://doc.qt.io/qtforpython-5/PySide2/QtWidgets/QFileDialog.html#PySide2.QtWidgets.PySide2.QtWidgets.QFileDialog.getSaveFileName @chennes

chennes, 2023-04-28:
This is fixed in 0.21_dev, I think. I can't find the exact commit right now -- can someone please re-test in latest dev?

amreo, 2023-04-28:
> This is fixed in 0.21_dev, I think. I can't find the exact commit right now -- can someone please re-test in latest dev?

After a bit of research I found that this (https://github.com/FreeCAD/FreeCAD/commit/a79b397e85ed6031f7faf4659720b253787ddcf4) is the commit what are you referring so

So, this issue is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/8577

closed 2023-04-28 by chennes

adrianinsaval, 2023-04-29:
should we backport this to 0.20 branch?

chennes, 2023-04-30:
No objection from me.