FreeCAD/FreeCAD #9426: [Issue] Move Tool - Continue mode appears to have no effect

pastcompute, 2023-04-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

None

### Issue description

I am attempting to follow the tutorial at https://wiki.freecad.org/Manual:Traditional_2D_drafting
I am also referencing the manual for the Move tool at https://wiki.freecad.org/Draft_Move
According to that Draft_move page, if I turn on the 'Continue (T)' option, along with Copy, when I complete the copy, the tool should restart, which to me means it should stay open and ready to perform another copy. However, the tool is being dismissed and I need to start it again.
To repeat:
- select an object
- M V
- select the checkboxes: C (copy) T (continue) R (Relative)
- 0 ENTER 0 ENTER 0 ENTER 1 ENTER 0 ENTER 0 ENTER     these can be anything of course
- ENTER

Expected result: 

object is copied, and the tool stays open ready to make another copy

Actual result:

object is copied, but tool closes. I need to then press M V to bring it back
I also noticed that neither Continue (T) or Copy (P) stay clicked and I need to click them again

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-04-28:
can you also reproduce this on the latest dev version?

Roy-043, 2023-05-17:
Although the related code has been updated (#7611), the problem also occurs in the latest dev.
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33175 (Git)
Build type: Release
Branch: master
Hash: c4a6b3a017469f7bd2c20dce44ac1264155a8488
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

PR #9597 (2023-05-17): Draft: Draft_Move continue mode failed with numeric input
Fixes #9426.

---